litex-hub/linux-on-litex-vexriscv #5: finish ULX3S support

enjoy-digital, 2019-05-07:
ULX3S is alsmot booting, but https://github.com/enjoy-digital/linux-on-litex-vexriscv/issues/4 prevents it to go to user space. When programming the board with OpenOCD, the serial is no longer detected and then prevents using lxterm to load the linux images. When loading the ujprog, the serial is still detected correctly after bitstream is loaded.

````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital
 (c) Copyright 2012-2015 M-Labs Ltd

 BIOS built on May  7 2019 09:40:21
 BIOS CRC passed (c64754ab)

--============ SoC info ================--
CPU:       VexRiscv @ 50MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========= Peripherals init ===========--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading binaries/Image to 0xc0000000 (2726132 bytes)...
[LXTERM] Upload complete (15.3KB/s).
[LXTERM] Uploading binaries/rootfs.cpio to 0xc0800000 (4054528 bytes)...
[LXTERM] Upload complete (15.3KB/s).
[LXTERM] Uploading binaries/rv32.dtb to 0xc1000000 (1618 bytes)...
[LXTERM] Upload complete (14.1KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (9128 bytes)...
[LXTERM] Upload complete (15.0KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x20000000
--============= Liftoff! ===============--
VexRiscv Machine Mode software built May  7 2019 10:01:09
--========== Booting Linux =============--
[    0.000000] No DTB passed to the kernel
[    0.000000] Linux version 5.0.9 (florent@lab) (gcc version 8.3.0 (Buildroot 2019.05-git-00938-g75f9fcd0c9)) #1 Thu May 2 17:43:30 CEST 2019
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] elf_hwcap is 0x1100
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: mem=32M@0x40000000 rootwait console=hvc0 root=/dev/ram0 init=/sbin/init swiotlb=32
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 119052K/131072K available (1957K kernel code, 92K rwdata, 317K rodata, 104K init, 184K bss, 12020K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x114c1bade8, max_idle_ns: 440795203839 ns
[    0.000699] sched_clock: 64 bits at 75MHz, resolution 13ns, wraps every 2199023255546ns
[    0.005987] Console: colour dummy device 80x25
[    0.026123] printk: console [hvc0] enabled
[    0.030423] Calibrating delay loop (skipped), value calculated using timer frequency.. 150.00 BogoMIPS (lpj=300000)
[    0.032373] pid_max: default: 32768 minimum: 301
[    0.067479] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.069751] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.266660] devtmpfs: initialized
[    0.424107] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.426071] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.964934] clocksource: Switched to clocksource riscv_clocksource
[    2.118148] Unpacking initramfs...
[    8.935616] Initramfs unpacking failed: junk in compressed archive
[    9.015373] workingset: timestamp_bits=30 max_order=15 bucket_order=0
[   10.362975] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
[   10.364303] io scheduler mq-deadline registered
[   10.365359] io scheduler kyber registered
[   17.700662] random: get_random_bytes called from init_oops_id+0x4c/0x60 with crng_init=0
[   42.084334] OF: fdt: not creating '/sys/firmware/fdt': CRC check failed
[   42.144834] Freeing unused kernel memory: 104K
[   42.147410] This architecture does not have kernel memory protection.
[   42.148354] Run /init as init process
````

enjoy-digital, 2019-05-07:
With #4 fixed, it's now booting correctly:
````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital
 (c) Copyright 2012-2015 M-Labs Ltd

 BIOS built on May  7 2019 11:34:11
 BIOS CRC passed (94cbc412)

--============ SoC info ================--
CPU:       VexRiscv @ 50MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========= Peripherals init ===========--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading binaries/Image to 0xc0000000 (2726132 bytes)...
[LXTERM] Upload complete (15.3KB/s).
[LXTERM] Uploading binaries/rootfs.cpio to 0xc0800000 (4054528 bytes)...
[LXTERM] Upload complete (15.3KB/s).
[LXTERM] Uploading binaries/rv32.dtb to 0xc1000000 (1866 bytes)...
[LXTERM] Upload complete (14.2KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (9128 bytes)...
[LXTERM] Upload complete (15.1KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x20000000
--============= Liftoff! ===============--
VexRiscv Machine Mode software built May  7 2019 11:15:54
--========== Booting Linux =============--
[    0.000000] No DTB passed to the kernel
[    0.000000] Linux version 5.0.9 (florent@lab) (gcc version 8.3.0 (Buildroot 2019.05-git-00938-g75f9fcd0c9)) #1 Thu May 2 17:43:30 CEST 2019
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] elf_hwcap is 0x1101
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M@0x40000000 rootwait console=hvc0 root=/dev/ram0 init=/sbin/init swiotlb=32
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 21588K/32768K available (1957K kernel code, 92K rwdata, 317K rodata, 104K init, 184K bss, 11180K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x114c1bade8, max_idle_ns: 440795203839 ns
[    0.000694] sched_clock: 64 bits at 75MHz, resolution 13ns, wraps every 2199023255546ns
[    0.006007] Console: colour dummy device 80x25
[    0.026125] printk: console [hvc0] enabled
[    0.030447] Calibrating delay loop (skipped), value calculated using timer frequency.. 150.00 BogoMIPS (lpj=300000)
[    0.032399] pid_max: default: 32768 minimum: 301
[    0.068752] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.073240] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.269172] devtmpfs: initialized
[    0.436467] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.438471] futex hash table entries: 256 (order: -1, 3072 bytes)
[    1.002666] clocksource: Switched to clocksource riscv_clocksource
[    2.151693] Unpacking initramfs...
[    9.037214] Initramfs unpacking failed: junk in compressed archive
[    9.117003] workingset: timestamp_bits=30 max_order=13 bucket_order=0
[   10.469731] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
[   10.471119] io scheduler mq-deadline registered
[   10.472172] io scheduler kyber registered
[   17.818164] random: get_random_bytes called from init_oops_id+0x4c/0x60 with crng_init=0
[   17.942082] Freeing unused kernel memory: 104K
[   17.944665] This architecture does not have kernel memory protection.
[   17.945622] Run /init as init process
mount: mounting tmpfs on /dev/shm failed: Invalid argument
mount: mounting tmpfs on /tmp failed: Invalid argument
mount: mounting tmpfs on /run failed: Invalid argument
Starting syslogd: OK
Starting klogd: OK
Initializing random number generator... [   27.893920] random: dd: uninitialized urandom read (512 bytes read)
done.
Starting network: ip: socket: Function not implemented
ip: socket: Function not implemented
FAIL

Welcome to Buildroot
buildroot login: root
login[48]: root login on 'hvc0'
````

kamejoko80, 2019-06-01:
I got an error when building ulx3s

ERROR: Max frequency for clock '$glbnet$main_soclinux_clkout0': 45.65 MHz (FAIL at 50.00 MHz)

Info: Max delay <async>                               -> posedge $glbnet$main_soclinux_clkout0: 4.53 ns
Info: Max delay posedge $glbnet$main_soclinux_clkout0 -> <async>                              : 4.24 ns
 
Is there any problem with the synthesizer tool nextpnr-ecp5?

Full log message is below:

# ./make.py --board=ulx3s --build
/home/phuong/Workspace/Projects/LITEX/linux-on-litex-vexriscv/litex/litex/soc/cores/cpu/vexriscv/verilog/VexRiscv_Linux.v:898: Warning: System task `$display' outside initial block is unsupported.
Warning: Conflicting init values for signal \main_soclinux_sdram_trrdcon_ready (\main_soclinux_sdram_trrdcon_ready = 1'1, \main_soclinux_sdram_choose_req_want_activates = 1'0).
Warning: Wire top.builder_array_muxed0 has an unprocessed 'init' attribute.
Warning: Wire top.builder_array_muxed1 has an unprocessed 'init' attribute.
Warning: Wire top.builder_array_muxed5 has an unprocessed 'init' attribute.
Warning: Wire top.builder_array_muxed6 has an unprocessed 'init' attribute.
Warning: Wire top.builder_bankmachine0_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_bankmachine1_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_bankmachine2_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_bankmachine3_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_cache_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_litedramwishbone2native_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_multiplexer_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_refresher_next_state has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed12 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed13 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed15 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed16 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed18 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed19 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed21 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed22 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed24 has an unprocessed 'init' attribute.
Warning: Wire top.builder_rhs_array_muxed32 has an unprocessed 'init' attribute.
Warning: Wire top.builder_soclinux_error has an unprocessed 'init' attribute.
Warning: Wire top.builder_soclinux_slave_sel has an unprocessed 'init' attribute.
Warning: Wire top.main_emulator_ram_we has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_cache_data_port_dat_w has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_cache_data_port_we has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_cache_tag_di_dirty has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_cache_tag_port_we has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_port_cmd_payload_addr has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_port_cmd_payload_we has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_choose_req_want_activates has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_inti_p0_rddata has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_inti_p0_rddata_valid has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_address has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_bank has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_cas_n has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_cke has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_cs_n has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_ras_n has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_rddata_en has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_we_n has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_wrdata has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_master_p0_wrdata_en has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sdram_trrdcon_ready has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_sram_we has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_uart_eventmanager_status_w has an unprocessed 'init' attribute.
Warning: Wire top.main_soclinux_vexriscv_interrupt0 has an unprocessed 'init' attribute.
Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x02a4b064

Info: constraining clock net 'clk25' to 25.00 MHz
Info: constraining clock net 'clk25' to 25.00 MHz

Info: Packing IOs..
[... 390 lines cut ...]
Info: [ 52432,  56612) |
Info: [ 56612,  60792) |
Info: [ 60792,  64972) |
Info: [ 64972,  69152) |
Info: [ 69152,  73332) |
Info: [ 73332,  77512) |
Info: [ 77512,  81692) |*+
0 warnings, 1 error
Traceback (most recent call last):
  File "./make.py", line 230, in <module>
[... cut ...]
  File "/home/phuong/Workspace/Projects/LITEX/linux-on-litex-vexriscv/litex/litex/build/lattice/platform.py", line 29, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/phuong/Workspace/Projects/LITEX/linux-on-litex-vexriscv/litex/litex/build/lattice/trellis.py", line 193, in build
    _run_script(script)
  File "/home/phuong/Workspace/Projects/LITEX/linux-on-litex-vexriscv/litex/litex/build/lattice/trellis.py", line 106, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
phuong@ubuntu1604lts:~/Workspace/Projects/LITEX/linux-on-litex-vexriscv$

Dolu1990, 2019-06-05:
@kamejoko80 that timing issue is weird.
Did you tried to synthetise on the vendor tool too get its FMax ? Just to figure out if that's a nextpnr-ecp5 issue or a RTL design issue ?

kamejoko80, 2019-06-07:
@Dolu1990 
Thanks for your reply.

For board ulx3s, there is a hardcode defined toolchain trellis in make.py @line 196
I changed to:
soc_kwargs["toolchain"] = "diamond"
However, my Linux version of diamond tool didn't work, I used the windows version and compiled the generated Verilog files.

Timing error report seems to be persisted below is the log:

PAR: Place And Route Diamond (64-bit) 3.11.0.396.4.
Copyright (c) 1991-1994 by NeoCAD Inc. All rights reserved.
Copyright (c) 1995 AT&T Corp.   All rights reserved.
Copyright (c) 1995-2001 Lucent Technologies Inc.  All rights reserved.
Copyright (c) 2001 Agere Systems   All rights reserved.
Copyright (c) 2002-2019 Lattice Semiconductor Corporation,  All rights reserved.
Fri Jun 07 23:00:07 2019

Z:/FOSSIL/ProgramFiles/lscc/diamond/3.11_x64/ispfpga\bin\nt64\par -f
ulx3s_impl1.p2t ulx3s_impl1_map.ncd ulx3s_impl1.dir ulx3s_impl1.prf -gui

Preference file: ulx3s_impl1.prf.

Cost Table Summary
Level/       Number       Worst        Timing       Worst        Timing       Run          NCD
Cost [ncd]   Unrouted     Slack        Score        Slack(hold)  Score(hold)  Time         Status
----------   --------     -----        ------       -----------  -----------  ----         ------
5_1   *      0            -2.233       4218904      0.160        0            02:36        Completed
* : Design saved.

Total (real) run time for 1-seed: 2 mins 37 secs 

par done!

Note: user must run 'Trace' for timing closure signoff.

Lattice Place and Route Report for Design "ulx3s_impl1_map.ncd"
Fri Jun 07 23:00:07 2019

Best Par Run
PAR: Place And Route Diamond (64-bit) 3.11.0.396.4.
Command Line: par -w -l 5 -i 6 -t 1 -c 0 -e 0 -gui -exp parUseNBR=1:parCDP=auto:parCDR=1:parPathBased=OFF ulx3s_impl1_map.ncd ulx3s_impl1.dir/5_1.ncd ulx3s_impl1.prf
Preference file: ulx3s_impl1.prf.
Placement level-cost: 5-1.
Routing Iterations: 6

Loading design for application par from file ulx3s_impl1_map.ncd.
Design name: top
NCD version: 3.3
Vendor:      LATTICE
Device:      LFE5U-45F
Package:     CABGA381
Performance: 6
Loading device for application par from file 'sa5p45.nph' in environment: Z:/FOSSIL/ProgramFiles/lscc/diamond/3.11_x64/ispfpga.
Package Status:                     Final          Version 1.43.
Performance Hardware Data Status:   Final          Version 55.1.
License checked out.

Ignore Preference Error(s):  True

Device utilization summary:

   PIO (prelim)      44/245          17% used
                     44/203          21% bonded
   IOLOGIC           38/245          15% used

   SLICE           4267/21924        19% used

   GSR                1/1           100% used
   EBR               29/108          26% used
[... 309 lines cut ...]

Dumping design to file ulx3s_impl1.dir/5_1.ncd.

All signals are completely routed.

PAR_SUMMARY::Run status = Completed
PAR_SUMMARY::Number of unrouted conns = 0
PAR_SUMMARY::Worst  slack<setup/<ns>> = -2.233
PAR_SUMMARY::Timing score<setup/<ns>> = 4218.904
PAR_SUMMARY::Worst  slack<hold /<ns>> = 0.160
PAR_SUMMARY::Timing score<hold /<ns>> = 0.000
PAR_SUMMARY::Number of errors = 0

Total CPU  time to completion: 2 mins 23 secs 
Total REAL time to completion: 2 mins 37 secs 

par done!

Note: user must run 'Trace' for timing closure signoff.

Copyright (c) 1991-1994 by NeoCAD Inc. All rights reserved.
Copyright (c) 1995 AT&T Corp.   All rights reserved.
Copyright (c) 1995-2001 Lucent Technologies Inc.  All rights reserved.
Copyright (c) 2001 Agere Systems   All rights reserved.
Copyright (c) 2002-2019 Lattice Semiconductor Corporation,  All rights reserved.

Dolu1990, 2019-06-07:
Hoooo. Maybe that's due to one change i made in the way the verilog is generated.
Basicaly, now all combinatorial always blocks generated from SpinalHDL will only drive one signal. By the past, the generation was allowed to merge combinatorial always blocks which had in common some if statements.
I tested on both Vivado, Quartus, and ice40 yosys, that seem to have no impact.  But maybe i'm wrong. 
I will test on my setup.

enjoy-digital, 2019-06-07:
I just tried rebuilding with upstream LiteX and last version of vexriscv-verilog and build is working fine:
https://hastebin.com/iguxatibal.http

I'm going to update Yosys/NextPnr and see if it's also successful.

enjoy-digital, 2019-06-07:
With upstream Yosys/Trellis/NextPnr it's also building fine: https://hastebin.com/cuhumiyuwa.http
@kamejoko80 is everything up to date on your side?

kamejoko80, 2019-06-07:
@enjoy-digital 

I installed the toolchains as below:

Yosys:

https://github.com/YosysHQ/yosys

Prjtrellis:

https://github.com/SymbiFlow/prjtrellis

Then nextpnr:

https://github.com/YosysHQ/nextpnr

Are there any special configurations before compiling and installing the toolchains?

enjoy-digital, 2019-06-07:
@kamejoko80: just following the respective READMEs should be fine.
Can you try executing build_top.sh in the attached archive and see if
you have the same result (remove the .txt extension)?

[gateware.tar.xz.txt](https://github.com/enjoy-digital/linux-on-litex-vexriscv/files/3267386/gateware.tar.xz.txt)

daveshah1, 2019-06-07:
10-15% could well be within inter-seed variation. So if there is any nondeterminism (slight tool version or database differences, differences in the Verilog even nonfunctional or library differences) then this might take the design from passing to failing (you can test this by passing a value other than 1 to the nextpnr --seed parameter). I have observed this behaviour with Diamond too.

Dolu1990, 2019-06-08:
What i could do to improve that specific critical path, is to pipeline the interrupt delegation / agregation.
this would only cost something like 8 register + 1 cycle latency on interrupts

kamejoko80, 2019-06-09:
@enjoy-digital 

I got the same error after running your build_top.sh. I've compared your gateware folder and mine, they are identical.

I've tried with @daveshah1's suggestion however it doesn't help.

Dolu1990, 2019-06-09:
What is the critical path for you ? 
I just submited a MTIP relaxation fix

Dolu1990, 2019-06-09:
ahh but it look like litex isn't using the updated version (https://github.com/enjoy-digital/VexRiscv-verilog/blob/master/VexRiscv_Linux.v)

kamejoko80, 2019-06-09:
@Dolu1990 

I've updated the repo in local and built again, but the same error happened.
Have you already tried the command ./make.py --board=ulx3s --build ?

Dolu1990, 2019-06-09:
No no, don't worry, i have the same issue. Now i can reproduce it.
I'm now looking how to relax that exact path.

Dolu1990, 2019-06-09:
I mean, one easy way would be to disable the static branch prediction in the VexRiscv config. for sure then the timings would be ok. But i'm trying to find a way to not having to do that

Dolu1990, 2019-06-09:
I think i have a good fix, but i need some time to check it work on all configs.
Basicaly, avoiding that the branch prediction having to check that the instruction in the decode stage is allowed to go in the execute stage, which was adding much combinatorial path on the nextPC calculation

Actualy got 
Info: Max frequency for clock '$glbnet$main_soclinux_clkout0': 55.39 MHz (PASS at 50.00 MHz)
I hope that's not only because of a lucky seed XD

I will let's you know when things are ok.

Dolu1990, 2019-06-15:
At i reworked the whole CPU flush/pc calculation logic, and now it pass all the regressions tests.
https://github.com/SpinalHDL/VexRiscv/commit/7c3c4e8c8126afe3b957a7779223fa8cdfcd660e#diff-04c6e90faac2675aa89e2176d2eec7d8R98

Will release it soon, Just need to release SpinalHDL with it to bringe jtag fixes

kamejoko80, 2019-06-16:
@Dolu1990

Sounds good
Thanks

Dolu1990, 2019-06-16:
So, i updated VexRiscv verilog with a specific config : 

```
VexRiscv_LinuxNoDspFmax.v: $(SRC)
	sbt compile "runMain vexriscv.GenCoreDefault --csrPluginConfig linux-minimal --singleCycleMulDiv=false --relaxedPcCalculation=true --prediction=none --outputFile VexRiscv_LinuxNoDspFmax"
```

Mainly, relaxed PC calculation, and iterative mul div as currently, there is no DSP inferation in the open source flow for the ECP5.

@enjoy-digital  This need to be integrated futher in the litex toolchain. 
- Updating the vexriscv verilator git module in litex,
- Setting the VexRiscv_LinuxNoDspFmax.v as the one to use for the ULX3S

Does this seem good ? 

I tried synthesis, and i get a good margin : 
Info: Max frequency for clock '$glbnet$main_soclinux_clkout0': 61.97 MHz (PASS at 50.00 MHz)

enjoy-digital, 2019-06-17:
Thanks @Dolu1990, i integrated that and now also get good margins, @kamejoko80 if it's also fine for you, we can close this issue.

kamejoko80, 2019-06-17:
Thanks @Dolu1990

Confirmed that I'm able to synthesize and generate the bitstream for board ulx3s now.

I got the maximum clock frequency ~ 58.7MHz

Info: Max frequency for clock '$glbnet$main_soclinux_clkout0': 58.70 MHz (PASS at 50.00 MHz)
Info: Max delay <async>                               -> posedge $glbnet$main_soclinux_clkout0: 4.20 ns
Info: Max delay posedge $glbnet$main_soclinux_clkout0 -> <async>                              : 4.63 ns

enjoy-digital, 2019-06-17:
Good, thanks.

closed 2019-06-17 by enjoy-digital