litex-hub/linux-on-litex-vexriscv #10: Auto-generate Linux Device Tree in LiteX?

mithro, 2019-05-08:
It would be good for LiteX to generate suitable device tree for Linux / Zephyr (potentially micropython in future).

I started (but never finished) a tool to take a `csr.cvs` file and generate a [Linux Device Tree file](https://github.com/mithro/litex-devicetree) - see https://github.com/mithro/litex-devicetree

It was targeting producing suitable dts for or1k as part of the LCA2018 Miniconf tutorial (see https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf). It seems to be used here;

https://github.com/timvideos/litex-buildenv/blob/005f7463bd5e022ee3e3d8cc90efde57e8465b86/scripts/build-linux.sh#L97-L131

A much better solution would be to allow modules to provide devicetree fragments in the same way that they can provide CSRs and stuff.

enjoy-digital, 2019-05-08:
Thanks, i also want to go in that direction. I'll experiment with your previous work.

mithro, 2019-05-08:
FYI - There are some useful links at https://github.com/timvideos/litex-buildenv/wiki/DeviceTree

Some [links to devicetree fragments which might be useful in this old Google Doc too](https://docs.google.com/document/d/1GfWKjrcfA27qbF8F9EtvRP5mzsaBjRzADfZTb5H8irw/edit#heading=h.3112hunryxys).

futaris, 2019-06-08:
https://github.com/riscv/meta-riscv/issues/143

gsomlo, 2019-07-08:
Just to thrown in my $0.02, from the 64bit RocketChip perspective: I'm starting with the Chisel-generated dts file (see e.g., https://github.com/enjoy-digital/rocket-litex-verilog/blob/master/generated-src/freechips.rocketchip.system.LitexFullConfig.dts) which for now needs to have `cpu@0`'s `timebase-frequency` moved one level up under `cpus`, and then throw out `mmio-port-axi4` and adjust `memory` accordingly, then paste in the various LiteX devices as appropriate (`uart0` and `mac0` for now).

Not sure how I/we could automate the hand-editing of this file from the LiteX builder. Maybe the 32-bit CPU options are less complex as far as DTS is concerned?

enjoy-digital, 2019-08-15:
With https://github.com/enjoy-digital/litex/commit/2899928aba932d5fa93902d9a2ef6a169f1c76b7 and https://github.com/litex-hub/linux-on-litex-vexriscv/commit/acffa47e621e11f8eed9eefea950323755838198 dts files are not auto-generated. A json file is exported with LiteX and then used by the json2dts.py script. We could improve that in the future but at least it's already enough for linux-on-litex-vexriscv.

mithro, 2019-08-15:
We also had https://github.com/litex-hub/litex-devicetree

Would be good to merge these two approaches...