litex-hub/linux-on-litex-vexriscv #12: Integrate LiteSDCard

enjoy-digital, 2019-05-12:
- [ ] 1) Simplify LiteSDCard to run with a divided clock (will reduce resource usage) and with a vendor agnostic PHY.
- [ ] 2) Allow booting from SD Card in LiteX BIOS. (copy Linux images to RAM and jump to it).
- [ ] 3) Allow having the file system directly on the SD Card.

mateusz-holenko, 2019-08-27:
We are currently working on supporting LiteSDCard in Linux focusing on filesystem (point 3).

enjoy-digital, 2019-08-27:
Great thanks, i can the two first points.

sajattack, 2020-02-23:
Any progress on this?

mateusz-holenko, 2020-02-24:
We introduced some changes to LiteSDCard core in order to handle longer responses (currently available on a branch: https://github.com/antmicro/litesdcard/tree/sdcard-fixes; will clean it up soon and create a PR out of it) and have a working Linux driver, also on a branch: https://github.com/antmicro/litex-linux-riscv/tree/sd_card_driver.

We need to get rid of some small hacks before pushing it to master, but the linked version is already usable if you want to experiment with it.

enjoy-digital, 2020-02-24:
Great @mateusz-holenko. Just by curiosity, do you have an ideas of the write/read speeds?

mateusz-holenko, 2020-02-26:
This is the output of `hdparm`:
```
root@buildroot:~# hdparm -Tt /dev/mmcblk0
/dev/mmcblk0:
Timing buffer-cache reads:     2 MB in 0.75 seconds = 2701 kB/s
Timing buffered disk reads:    1 MB in 10.90 seconds = 93 kB/s
```

enjoy-digital, 2020-02-26:
Thanks for the results.

mateusz-holenko, 2020-02-26:
And for writing (don't sure if this is a correct way of testing due to caches etc.):
```
root@buildroot:~# time dd if=/dev/urandom of=test bs=512 count=2048; sync
2048+0 records in
2048+0 records out
real    0m 32.78s
user    0m 0.41s
sys     0m 9.01s
```

enjoy-digital, 2020-02-26:
@mateusz-holenko: it was really for my curiosity. The more important for now is to have something reliable, we could identify the bottleneck later and improve performance. Once you have something you think is reliable, we could merge.

Dolu1990, 2020-02-26:
@mateusz-holenko on which dev kit, at which frequancy you got those results ?also, the driver do active wait or use interrupt instead ? For reference on ulx3s, using a spi driver not using interrupts, at 50 mhz on ulx3s on another soc, that’s 6 time faster to do buffered disc accesses.

mateusz-holenko, 2020-02-26:
@Dolu1990 dev kit: arty 7, cpu freq: 100MHz, sd card freq: 25Mhz

`sdcard_wait_cmd_done` is based on polling of `cmdevt`, so I guess it counts as active wait

enjoy-digital, 2020-02-26:
@Dolu1990: thanks for the comparison, we should have some margin to improve things then :) The core has been mostly tested on block transfers (and should be able to do 10MB/s writes/reads at this frequency) and there is probably a bottleneck with this usecase that we need to remove.

enjoy-digital, 2020-04-13:
@Dolu1990: would you mind sharing a link to the Spinal HDL gateware and Linux driver you are using in SaxonSoC for MMC over SPI? @gsomlo is trying to get MMC working over SPI and we would like to see how you are handling the CS pin.

pdp7, 2020-04-14:
> And for writing (not sure if this is a correct way of testing due to caches etc.):
> 
> ```
> root@buildroot:~# time dd if=/dev/urandom of=test bs=512 count=2048; sync
> 2048+0 records in
> 2048+0 records out
> real    0m 32.78s
> user    0m 0.41s
> sys     0m 9.01s
> ```

Which FPGA dev board are you using?

I got the Arty and the Antmicro expansion board at Embedded World but I'm not sure what branch of LiteX to try on it.  Thanks!

enjoy-digital, 2020-12-16:
We can now have the file system on the SDCard in SPI-Mode. SD-Mode should also be working (since validated by @mateusz-holenko and @gsomlo) we just need to finish the integration on Linux-on-LiteX-VeRiscv: probably only make sure the `.dts` is valid.

enjoy-digital, 2021-01-14:
This was indeed related to the `.dts`, `config-frequency` was missing in the CPUs properties. (https://github.com/enjoy-digital/litex/commit/b1cad93e62ecf57f1cd98489ca592dcf6d1cf3a2). We can probably finally close this :)

closed 2021-01-14 by enjoy-digital