litex-hub/linux-on-litex-vexriscv #13: Integrate LiteEth

enjoy-digital, 2019-05-12:
- [ ] 1) Regenerate VexRiscv to have similar interrupts for User/Supervisor mode.
- [ ] 2) Add interrupt support to software.
- [ ] 3) Get actual LiteEth Linux driver working.
- [ ] 4) Review driver, do eventual optimizations, understand limitations.
- [ ] 5) Add DMA support if needed to improve performance.

Dolu1990, 2019-05-20:
Hi,

How things are going on it ? What is blocking ?
Regenerate VexRiscv to have similar interrupts for User/Supervisor mode can be checked right ?

enjoy-digital, 2019-05-20:
Hi @Dolu1990,

yes sorry, nothing blocking, just that i haven't spent time on it :)

Dolu1990, 2019-05-20:
No worries, i'm just asking to be sure there is nothing dramatic happening ^^

frantony, 2019-06-06:
Please see https://github.com/enjoy-digital/linux-on-litex-vexriscv/pull/18

enjoy-digital, 2019-06-10:
@frantony i just regenerated everything from upstream and tested successfully ethernet in polling mode, thakns.

enjoy-digital, 2020-12-16:
LiteEth is now working with decent performance. We could add a DMA in the future but we can consider the integration done.

closed 2020-12-16 by enjoy-digital