litex-hub/linux-on-litex-vexriscv #15: Issue booting linux on ArtyA7

Dolu1990, 2019-05-15:
Hi,

I tested the flow on the ArtyA7 and got an issue at the moment it liftoff in linux.

Commands used :

```
git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv.git

git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv-prebuilt
cp -r linux-on-litex-vexriscv-prebuilt/* linux-on-litex-vexriscv

wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
sudo ./litex_setup.py init install --user

cd linux-on-litex-vexriscv
./make.py --board=arty --build
sudo ./make.py --board=arty --flash
```

ttyUSB => 
```        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs Ltd

 BIOS built on May 15 2019 14:36:07
 BIOS CRC passed (6772ad38)

--============ SoC info ================--
CPU:       VexRiscv @ 100MHz
ROM:       32KB
SRAM:      32KB
L2:        8KB
MAIN-RAM:  262144KB

--========= Peripherals init ===========--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111110000000000000000000000000| delays: 03+-03
m0, b1: |00000000000000111111111000000000| delays: 18+-04
m0, b2: |00000000000000000000000000000011| delays: 31+-01
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 18+-04
m1, b0: |11111111111000000000000000000000| delays: 05+-05
m1, b1: |00000000000000111111111111000000| delays: 20+-06
m1, b2: |00000000000000000000000000000011| delays: 31+-01
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b1 delays: 20+-05
SDRAM now under hardware control
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Loading Image from flash...
Loading rootfs.cpio from flash...
Loading rv32.dtb from flash...
Loading emulator.bin from flash...
Executing booted program at 0x20000000
--============= Liftoff! ===============--
``` 

And that's all.
The ./sim.py work and boot the linux properly.

Currently, i'm not sure where the problem is.
Maybe somebody else can try on their own the same procedure to cross check ?
Or I did something wrong ?

Dolu1990, 2019-05-15:
In the bios, the memtest command pass.

enjoy-digital, 2019-05-15:
Hi @Dolu1990,

i can test it tomorrow. Your issue is probably specific to the boot from the SPI Flash.
Can you test booting from serial?: https://github.com/enjoy-digital/linux-on-litex-vexriscv#load-the-linux-images-over-serial

Dolu1990, 2019-05-15:
Hi @enjoy-digital 
I tested both flash and serial and got the same issue.

Let's me know how it go :)

futaris, 2019-05-15:
I suspect it is because of the changes to the memory / device-tree. I will try booting this again soon.

https://github.com/enjoy-digital/linux-on-litex-vexriscv/commit/6b6f12660e23c217c7f9790f2f023addffc52213 was the last build that I tested working.

Dolu1990, 2019-05-16:
Unfortunatly, testing with 6b6f12660e23c217c7f9790f2f023addffc52213 endup into a flow error, as some stuff changed in the repo dependancies :)

enjoy-digital, 2019-05-16:
@Dolu1990: i just did the test with the prebuild binaries and rebuilding and both are working here:
````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs Ltd

 BIOS built on May 16 2019 21:22:33
 BIOS CRC passed (685c0ba9)

--============ SoC info ================--
CPU:       VexRiscv @ 100MHz
ROM:       32KB
SRAM:      32KB
L2:        8KB
MAIN-RAM:  262144KB

--========= Peripherals init ===========--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111000000000000000000000000000| delays: 02+-02
m0, b1: |00000000000011111111000000000000| delays: 16+-04
m0, b2: |00000000000000000000000000001111| delays: 30+-02
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 16+-04
m1, b0: |11111111100000000000000000000000| delays: 04+-04
m1, b1: |00000000000011111111111000000000| delays: 17+-05
m1, b2: |00000000000000000000000000001111| delays: 30+-02
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b1 delays: 17+-05
SDRAM now under hardware control
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Loading Image from flash...
Loading rootfs.cpio from flash...
Loading rv32.dtb from flash...
Loading emulator.bin from flash...
Executing booted program at 0x20000000
--============= Liftoff! ===============--
VexRiscv Machine Mode software built May 10 2019 10:54:08
--========== Booting Linux =============--
[    0.000000] No DTB passed to the kernel
[    0.000000] Linux version 5.0.9 (florent@lab) (gcc version 8.3.0 (Buildroot 2019.05-git-00938-g75f9fcd0c9-dirty)) #2 Fri May 10 15:14:31 CEST 2019
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000c0000000-0x00000000c7ffffff]
[    0.000000] elf_hwcap is 0x1101
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: mem=128M@0x40000000 rootwait console=hvc0 root=/dev/ram0 init=/sbin/init swiotlb=32
[... 98 lines cut ...]
0x50f80290  13 06 f0 02 93 76 f4 07 93 57 c4 00 63 8e c6 00  .....v...W..c...
0x50f802a0  13 06 30 07 63 82 c6 18 b7 27 00 f0 13 07 10 00  ..0.c....'......
0x50f802b0  23 a0 e7 00 6f 00 00 00 93 f7 77 00 63 9e e7 14  #...o.....w.c...
0x50f802c0  93 57 d4 00 37 36 00 20 13 06 46 c5 93 f7 c7 07  .W..76. ..F.....
0x50f802d0  b3 87 c7 00 03 a5 07 00 93 57 24 01 93 f7 c7 07  .........W$.....
0x50f802e0  93 56 b4 01 b3 87 c7 00 13 54 74 00 83 a7 07 00  .V.......Tt.....
0x50f802f0  13 74 f4 01 b7 08 02 00 73 a0 08 30 97 08 00 00  .t......s..0....
0x50f80300  93 88 88 01 73 90 58 30 13 08 10 00 03 27 05 00  ....s.X0.....'..
0x50f80310  13 08 00 00 b7 08 02 00 73 b0 08 30 63 00 08 02  ........s..0c...
0x50f80320  03 24 81 00 83 20 c1 00 83 24 41 00 13 05 09 00  .$... ...$A.....
0x50f80330  03 29 01 00 13 01 01 01 6f 00 40 37 13 08 c0 01  .)......o.@7....
0x50f80340  63 60 d8 0c 37 28 00 20 93 96 26 00 13 08 c8 b7  c`..7(. ..&.....
0x50f80350  b3 86 06 01 83 a6 06 00 67 80 06 00 b3 87 e7 00  ........g.......
0x50f80360  37 08 02 00 73 20 08 30 17 08 00 00 13 08 88 01  7...s .0........
0x50f80370  73 10 58 30 93 06 10 00 23 20 f5 00 93 06 00 00  s.X0....# ......
0x50f80380  37 08 02 00 73 30 08 30 e3 9c 06 f8 13 14 24 00  7...s0.0......$.
0x50f80390  33 04 c4 00 23 20 e4 00 13 09 49 00 73 10 19 34  3...# ....I.s..4
0x50f803a0  b7 07 00 20 93 87 47 11 73 90 57 30 83 20 c1 00  ... ..G.s.W0. ..
0x50f803b0  03 24 81 00 83 24 41 00 03 29 01 00 13 01 01 01  .$...$A..)......
0x50f803c0  67 80 00 00 b3 c7 e7 00 6f f0 9f f9 b3 f7 e7 00  g.......o.......
0x50f803d0  6f f0 1f f9 b3 e7 e7 00 6f f0 9f f8 e3 52 f7 f8  o.......o....R..
0x50f803e0  93 07 07 00 6f f0 df f7 e3 dc e7 f6 6f f0 5f ff  ....o.......o._.
0x50f803f0  e3 78 f7 f6 6f f0 df fe e3 f4 e7 f6 6f f0 5f fe  .x..o.......o._.
BIOS> 
````

Dolu1990, 2019-05-16:
I tried again with the --flash, and got the same issue, but here we got something interresting : 

```
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Cancelled

--============= Console ================--
BIOS> mr 0x50f80000 1024   
Memory dump:
0x50f80000  00 0b 13 00 00 00 13 00 00 00 13 00 00 00 13 00  ................
0x50f80010  00 00 13 00 00 00 13 00 00 00 13 00 00 00 23 2e  ..............#.
0x50f80020  11 fe 23 2c 51 fe 23 2a 61 fe 23 28 71 fe 23 26  ..#,Q.#*a.#(q.#&
0x50f80030  a1 fe 23 24 b1 fe 23 22 c1 fe 23 20 d1 fe 23 2e  ..#$..#"..# ..#.
0x50f80040  e1 fc 23 2c f1 fc 23 2a 01 fd 23 28 11 fd 23 26  ..#,..#*..#(..#&
0x50f80050  c1 fd 23 24 d1 fd 23 22 e1 fd 23 20 f1 fd 13 01  ..#$..#"..# ....
0x50f80060  01 fc ef 00 40 09 83 20 c1 03 83 22 81 03 03 23  ....@.. ..."...#
0x50f80070  41 03 83 23 01 03 03 25 c1 02 83 25 81 02 03 26  A..#...%...%...&

```

All bytes are two bytes off XD

Then i tried with the serial boot, and it worked :D

enjoy-digital, 2019-05-17:
@thanks for the test, can you give me the reference of the SPI Flash mounted on your board?

You should get it to work by changing https://github.com/enjoy-digital/linux-on-litex-vexriscv/blob/master/soc_linux.py#L59 to  11 +-4.

Dolu1990, 2019-05-17:
That's it, thanks :D
Your fix did it.

It is an Arty A7 revision E (On the bottom of the PCB, on the right of the big Xilinx logo)

The mounted flash is a Spansion FL128SAIF00.

Thanks for the help ^^

closed 2019-05-17 by Dolu1990

enjoy-digital, 2019-05-17:
Thanks for the feedback. In the future, we should make this dummy parameter dynamically configurable.