litex-hub/linux-on-litex-vexriscv #21: use BR2_EXTERNAL buildroot feature

frantony, 2019-06-08:
Buildroot already has mechanism for keeping project-specific customizations outside of the Buildroot tree, so we can drop out the ```cp -r ../linux-on-litex-vexriscv/buildroot/* ./``` step from 
[README.md](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/1c01088bb6706d8dbf877780fc1c392f8bc5f7fb/README.md).

See [buildroot/docs/manual/customize-outside-br.txt](https://github.com/buildroot/buildroot/blob/master/docs/manual/customize-outside-br.txt) for details.

futaris, 2019-06-08:
I agree we should use this.

enjoy-digital, 2019-06-10:
Thanks @frantony, @futaris for the suggestion .https://github.com/litex-hub/linux-on-litex-vexriscv/commit/54bb980b8f3712c8522280b63dce156b0005d191 implements it.

closed 2019-06-10 by enjoy-digital

futaris, 2019-06-22:
We need to update the README.md for this.

6:08 AM <mnr> Hello, I'm a newbie to litex and I've started experimenting around with https://github.com/litex-hub/linux-on-litex-vexriscv. When using the prebuilt binaries from https://github.com/enjoy-digital/linux-on-litex-vexriscv-prebuilt, I can successfully run the simulation with ./sim.py. When trying to build everything myself, I'm somewhat stuck at how emulator/emulator.bin is intended to be built. 
6:08 AM <mnr> emulator/Makefile refers to $(SOC_DIRECTORY), but I see nothing that would define this variable and AFAICS the README.md doesn't provide any information about that. Pointers welcome :).
6:17 AM <mnr> Hm, looks like the last line was too long and got truncated, therefore the second part again:
6:17 AM <mnr> emulator/Makefile refers to $(SOC_DIRECTORY), but I see nothing that would define this variable and AFAICS the README.md doesn't provide any information about that. Pointers welcome :).