litex-hub/linux-on-litex-vexriscv #24: Support the NOMMU port

mithro, 2019-06-12:
According to Palmer;
> The NOMMU RISC-V port was posted to the Linux Kernel mailing list just this week

I think he is talking about the stuff at https://github.com/chmmn/riscv-uclinux

It would be good to support NOMMU Linux on LiteX for even lower resource usage (and more predictable performance).

mithro, 2019-06-12:
@Dolu1990 - FYI How many resources would not having a MMU save?

Dolu1990, 2019-06-12:
@mithro I will check.
But also, wasn't it also about only having a machine mode only + no mmu linux port ?
That would probabably save much :D

Dolu1990, 2019-06-12:
I would say, it would mainly make it usable in a ice40 8K
Currently, i have a Linux SoC with SDRAM which can hold in it, but that's realy full

mithro, 2019-06-12:
@Dolu1990 That would be awesome!

futaris, 2019-06-14:
https://lwn.net/Articles/790888/

futaris, 2019-06-14:
NB, this doesn't support shared libraries. Static executables only.

AnttiLukats, 2019-08-04:
it would also make it able to run in MAX10 10M08 device, and maybe maybe maybe in 8Mbyte HyperRAM,,, for many years ago running nommu in 4Mbyte was not a problem

enjoy-digital, 2021-01-08:
Except doing it for the challenge, I'm not sure to see a real use case for this: iCE40 are good for pure logic gateware or to embeds small CPUs running bare-metal  apps or small RT OSes, not to run Linux. Medium size FPGAs (ECP5, Cyclone4/5, Artix7, etc...) easily integrate the current Linux VexRiscv SoC so I'm not sure it's worth putting much efforts in a NOMMU port (but if anyone do it, that would be nice to share this work with us).

closed 2021-01-08 by enjoy-digital

disdi, 2024-10-15:
Now buildroot root supports [NoMMU riscv32 qemu](https://github.com/buildroot/buildroot/blob/master/configs/qemu_riscv32_nommu_virt_defconfig).
[RP2350 support ](https://github.com/Mr-Bossman/pi-pico2-linux)also exists.

I think it will be very useful to have NoMMU linux running on [Vexriscv Linux](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/demo/Linux.scala) generated without MMU plugin.