litex-hub/linux-on-litex-vexriscv #27: cpuinfo processor	: 0

CarlFK, 2019-07-09:
```
                32-bits VexRiscv CPU with MMU integrated in a LiteX SoC
login[48]: root login on 'hvc0'
root@buildroot:~# cat /proc/cpuinfo
processor	: 0
hart		: 0
isa		: rv32ima
mmu		: sv32
uarch		: spinalhdl,vexriscv
```

mithro, 2019-07-09:
Actually this looks fine?
```
$ cat /proc/cpuinfo
processor       : 0 
vendor_id       : GenuineIntel
cpu family      : 6
stepping        : 1                                                                                                                     
cache size      : 15360 KB                                                                                                              
physical id     : 0      
siblings        : 6 
core id         : 0 
cpu cores       : 6                                
apicid          : 0
initial apicid  : 0
<snip>
```

CarlFK, 2019-07-09:
hmm  I was on:
https://github.com/litex-hub/linux-on-litex-vexriscv#running-the-litex-simulation

typing and echo was very sluggish - could something have been eating chars?

mithro, 2019-07-09:
That doesn't have anything to do with `cat /proc/cpuinfo` ?

enjoy-digital, 2019-07-09:
cpu info returns the infos for each processor, here we only have one processor with index 0, so this is fine.
The simulation is indeed slower than running on hardware (50 to 100x), so there can be a bit a latency between typing and the echo... :)

closed 2019-07-09 by enjoy-digital