litex-hub/linux-on-litex-vexriscv #29: readme litex_term.py needs more

CarlFK, 2019-07-09:
"use the lxterm terminal/tool"  could use more

I figured it out, but it seemed messy:
$ wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex/tools/litex_term.py
Saving to: ‘litex_term.py’

$ chmod u+x litex_term.py 
$ ./litex_term.py --images=images.json /dev/ttyUSB1 

can't bootstrap.sh or something set this up?

enjoy-digital, 2019-07-09:
lxterm should be installed when installing LiteX, was it not the case?

CarlFK, 2019-07-09:
distributing python apps is hard ;)

it isn't installed to $PATH or anything that will make this work:

```
To load the Linux images over Serial, use the lxterm terminal/tool provided by LiteX and run:
$ lxterm --images=images.json /dev/ttyUSBX
```
- https://github.com/litex-hub/linux-on-litex-vexriscv#load-the-linux-images-over-serial

It is installed with:
python setup.py init install --user
which puts something in ~/.local with is on sys.path

carl@twist:~/temp/lhub$ python3 -c "import sys;print(sys.path)"
 '/home/carl/.local/lib/python3.5/site-packages',

so this might work: 
python3 -m lxterm  --images=images.json /dev/ttyUSBX

CarlFK, 2019-07-09:
didn't work: 
```
juser@cnt5:~/lhub$ python3 -m lxterm
/usr/bin/python3: No module named lxterm
juser@cnt5:~/lhub$ python3 -m litex_term
/usr/bin/python3: No module named litex_term
```

closed 2019-07-11 by enjoy-digital

CarlFK, 2019-07-14:
sudo all that stuff is a very heavy hammer just to get lxterm added to the path.

How about "add (path to lxterm) to your path" ?

and lets hope anyone doing this knows how to do that.

commit 83b4ee1c (2019-07-11): README.md: install litex for all user, fixes #29