litex-hub/linux-on-litex-vexriscv #34: litex vexriscv cpu-variant linux not starting

AnttiLukats, 2019-08-04:
LiteX build seems working properly, 4 different new platforms did all come alive instantly, but in all cases as soon as the cpu-variant is changed from "full" to "linux" then the BIOS prompt does not come any more:
cyclone10LP target (quartus 18.1) - no uart output with "linux" variant
artix-7 target (vivado 2018.3 or 2019.1) - about 10 first characters come out, then stop

any suggestions what can be the issue? I will try to get some more boards and do more testing, but seems weird that until "full" all variants always work with no issues, and then dead comes

enjoy-digital, 2019-08-04:
Can you do a test with the linux variant using this mem_map:
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/soc_linux.py#L25-L33
instead of:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L142-L147
?

AnttiLukats, 2019-08-05:
not sure if I did it right:

class BaseSoC(SoCCore):

    mem_map = {
       "rom":          0x00000000,
       "sram":         0x10000000,
       "emulator_ram": 0x20000000,
       "main_ram":     0xc0000000,
       "csr":          0xf0000000,
    }
    mem_map.update(SoCCore.mem_map)

    def __init__(self, sys_clk_freq=int(100e6), **kwargs):
        platform = te0711.Platform()

this at least made no change, it is all working with variant=full and with variant=linux there comes out about 10 characters from the LiteX logo

"        __"
first 10 chars of greeting are printed then no further output..

enjoy-digital, 2019-08-05:
OK thanks for testing, i'll look at that.

AnttiLukats, 2019-08-05:
e, I am using riscv-embed-none as toolchain prefix, but I bet this can not cause the uart printout to stop after 10 characters ?  weird..

AnttiLukats, 2019-08-05:
when compiliny arty from litex-boards with "linux" variant the uart output is similar, maybe 12 characters come out from uart..

enjoy-digital, 2019-08-05:
Can you test with the attached patch?
[0001-soc_core-use-same-mem_map-than-soc_linux.patch.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/3467154/0001-soc_core-use-same-mem_map-than-soc_linux.patch.txt)

AnttiLukats, 2019-08-05:
with this patch IT WORKS. hope it helps to find some guidance how to deal with this issue

AnttiLukats, 2019-08-05:
update. it works on Artix FPGA's on Cyclone 10LP same result as before, no uart at all, there it may be different issue

AnttiLukats, 2019-08-06:
working, C10LP RefKit booted to linux prompt.

closed 2019-08-06 by AnttiLukats

enjoy-digital, 2019-08-06:
Good, what was the issue?

AnttiLukats, 2019-08-07:
missing "soc_reset" module