litex-hub/linux-on-litex-vexriscv #35: ethernet testing

AnttiLukats, 2019-08-07:
[   38.440914] libphy: Fixed MDIO Bus: probed
[   38.464502] liteeth f0003800.mac: Failed to get IRQ, using polling
[   38.610547] liteeth f0003800.mac eth0: irq 0, mapped at a0002800

I guess this is normal bootlog, and gives 0 information if the ethernet MAC/PHY is actually working? there are no packets sent out at all..

enjoy-digital, 2019-08-07:
Ethernet needs first to be validated with LiteX bios (at least getting netboot over TFTP working).

When validated, to test it in Linux, you can do:
`ifconfig -a` to see if eth0 is listed and then just configure it with
`ifconfig eth0 ip_address`

You can then try to ping the FPGA board from another computer, or ping another computer from the FPGA board.

AnttiLukats, 2019-08-08:
yes, BIOS just freezes, so I escaped to serial boot in the hope linux gives more information

closed 2019-08-08 by AnttiLukats