litex-hub/linux-on-litex-vexriscv #41: Print console messages as early as possible

enjoy-digital, 2019-08-26:
With the switch to real LiteX UART driver, first consoles messages seem to be buffered until the driver is loaded, so first console message can take up to a few seconds to appear (on HW) and up to a few minutes (on Simulation). This is not convenient, especially for simulation basic SoC verification (just see if Linux starts booting) now take a few minutes instead of a few seconds.

mithro, 2019-08-26:
FYI - @mateusz-holenko

mithro, 2019-08-26:
@enjoy-digital - You might want to look at the `earlycon` kernel parameter?

Links
 * https://github.com/torvalds/linux/blob/master/drivers/tty/serial/earlycon.c
 * https://www.kernel.org/doc/html/v5.2/admin-guide/kernel-parameters.html
 * https://github.com/torvalds/linux/blob/146c3d3220e039b5d61bf810e0b42218eb020f39/Documentation/admin-guide/kernel-parameters.txt#L978
 * https://github.com/torvalds/linux/blob/04ce9318898b294001459b5d705795085a9eac64/drivers/tty/serial/Kconfig#L11-L16

enjoy-digital, 2019-08-27:
Thanks @mithro,fixed by https://github.com/litex-hub/linux-on-litex-vexriscv/commit/2823bd9d3868d100f8f0cc70cf6e32decbf687fe.

closed 2019-08-27 by enjoy-digital