litex-hub/linux-on-litex-vexriscv #42: Integrate LiteVideo Framebuffer and use for console

enjoy-digital, 2019-08-26:
@daveshah1 demonstrated it on the Trellisboard with commits:
https://github.com/daveshah1/linux-on-litex-vexriscv/commit/7a4ccd70de92aa004b27ad714114cb8b89928612
https://github.com/daveshah1/linux-on-litex-vexriscv/commit/44fe4a89be308138823a027c1d1d958d2eef6070

Allow boards with native VGA/HDMI output to enable it.

mithro, 2019-08-26:
FYI - @mateusz-holenko

enjoy-digital, 2019-08-27:
https://github.com/litex-hub/linux-on-litex-vexriscv/commit/6d686fcea34915e325862cb1f55611b832155727 adds initial framebuffer support on Nexys Video.

mateusz-holenko, 2019-08-27:
We are currently working on LiteVideo framebuffer driver for Linux. I'll create a PR once we have some working code.

enjoy-digital, 2019-08-27:
@mateusz-holenko: ok nice. Are you going to do the clocks/timings/DMA initialization in the driver?

mateusz-holenko, 2019-08-28:
@enjoy-digital Yes, we plan to handle it.

enjoy-digital, 2019-08-30:
@mateusz-holenko: ok nice, then we will be able to remove the initialization code in the emulator. For now the code in the emulator is doing a minimal initialization of the framebuffer (640x480 @ 75Hz), this should show you how to control the registers in the FPGA. The timings/DMA part will be similar for all FPGAs but the clock part will be specific to FPGA families.

mateusz-holenko, 2020-01-22:
There is a PR #82 pending that adds the LiteVideo driver (ad removes the framebuffer-related code from the emulator). We are now working on extending the driver by integrating it with the MMCM to allow for dynamic reconfiguration.

enjoy-digital, 2021-03-30:
A new simplified Video Framebuffer has been added to LiteX and is now directly use here. The Video Framebuffer comes with the gateware pre-initialized and just requires the BIOS to enable it. For Linux-on-LiteX-VexRiscv it means that nothing specific has to be done except declaring the Video Framebuffer in the `.dts`.  A specific driver can still be used if resolution re-configuration is required but this is no longer mandatory. The Video Framebuffer has been already enabled on a few boards: NexysVideo/Nexys4DDR/Minispartan6/ULX3S and will be extended to more boards in the future.

closed 2021-03-30 by enjoy-digital