litex-hub/linux-on-litex-vexriscv #44: Document how to use/test Ethernet/SPI/I2C/Framebuffer

enjoy-digital, 2019-08-30:
Ethernet/SPI/I2C/Framebuffer have been integrated recently, but we haven't added information on how to use/test it. Add that to the README or another documentation file.

enjoy-digital, 2019-09-25:
**Ethernet**
Verify `eth0` is present: ` ifconfig -a`:
````
eth0      Link encap:Ethernet  HWaddr C6:6A:FB:04:6A:B9  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````
Configure `eth0`: `ifconfig eth0 192.168.1.50`
Verify you can ping another machine on the network:` ping 192.168.1.100`:
````
PING 192.168.1.100 (192.168.1.100): 56 data bytes
64 bytes from 192.168.1.100: seq=0 ttl=64 time=19.839 ms
64 bytes from 192.168.1.100: seq=1 ttl=64 time=4.585 ms
64 bytes from 192.168.1.100: seq=2 ttl=64 time=8.510 ms
64 bytes from 192.168.1.100: seq=3 ttl=64 time=12.522 ms
^C
--- 192.168.1.100 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 4.585/11.364/19.839 ms
````

Leds:
Configure GPIOs as output:
````
echo 508 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpiochip508/direction
````
Set value:
````
echo 0 > /sys/class/gpio/gpio508/value
echo 1 > /sys/class/gpio/gpio508/value
...
````

enjoy-digital, 2019-10-08:
**PWM/RGB Led**
from https://github.com/litex-hub/linux-on-litex-vexriscv/pull/52#issuecomment-539368711
<pre>
cd /sys/class/pwm/pwmchip0
echo 0 > export
cd pwm0
echo 100 > period
echo 50 > duty_cycle
echo 1 > enable
</pre>

This should turn the LED on at 50% of brightness - you can control it by changing `duty_cycle` value from 0 to `period`.

enjoy-digital, 2019-11-14:
**SPI Flash:**
from https://github.com/litex-hub/linux-on-litex-vexriscv/pull/55#issuecomment-553937357

There should be a `/dev/mtd0` that you can read from/write to directly from bash, i.e.,:

```
$ cat /dev/mtd0
```

Or even better, to see the data clearly:

```
$ dd if=/dev/mtd0 count=6 bs=1 status=none | hexdump
```

Before writing you should erase the flash first.
This requires `BR2_PACKAGE_MTD` and `BR2_PACKAGE_MTD_JFFS_UTILS` to be enabled in the buildroot config.

```
$ flash_erase /dev/mtd0 0 1
$ echo -ne "\x01\x01" > /dev/mtd0
```

mithro, 2019-11-14:
FYI - @kgugala @mateusz-holenko

enjoy-digital, 2019-11-25:
ICAPBitstream:
https://github.com/antmicro/litex-icapbitstream-module-example

enjoy-digital, 2020-03-19:
**Boot from the SDCard:**
 
Use a FAT16 partition on SDCard and copy the images to it:
- Image from buildroot/Image.
- rootfs.cpio fom buildroot/rootfs.cpio.
- rv32.dtb from buildroot/rv32.dtb.
- emulator.bin emulator/emulator.bin.
    
Then use the `spisdcardboot` command of the LiteX BIOS.

enjoy-digital, 2020-05-20:
**SDCard:**

Testing instructions:

1. During boot you should see the following lines in the log:
```
[    3.907468] litex-mmc f000a800.mmc: Setting clk freq to: 0
[    3.926727] litex-mmc f000a800.mmc: Setting clk freq to: 10000000
[    3.968631] litex-mmc f000a800.mmc: Setting clk freq to: 25000000
[    3.982844] mmc0: new SDHC card at address 1234
[    4.003202] blk_queue_max_hw_sectors: set to minimum 8
[    4.003936] blk_queue_max_segment_size: set to minimum 4096
[    4.004877] mmcblk0: mmc0:1234 SA16G 14.4 GiB 
[    4.216082]  mmcblk0: p1
```

2. Simple read test:
```
root@buildroot:~# hdparm -tT /dev/mmcblk0
/dev/mmcblk0:
Timing buffer-cache reads:    14 MB in 0.53 seconds = 26762 kB/s
Timing buffered disk reads:    2 MB in 4.31 seconds = 474 kB/s
```

3. Mounting the SD card in the filesystem:
```
root@buildroot:~# mkdir sdcard
root@buildroot:~# mount /dev/mmcblk0p1 sdcard
[  163.419996] EXT4-fs (mmcblk0p1): recovery complete
[  163.451384] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
```

4. Write test:
```
root@buildroot:~# yes "Testing sdcard write speed" | dd of=inputtestfile.txt bs=
512 count=1000; sync
1000+0 records in
1000+0 records out
root@buildroot:~# time dd if=inputtestfile.txt of=sdcard/outputtestfile.txt bs=5
12 count=1000; sync
1000+0 records in
1000+0 records out
real	0m 1.19s
user	0m 0.07s
sys	0m 1.10s
root@buildroot:~# ls -la sdcard/outputtestfile.txt 
-rw-r--r--    1 root     root        512000 Jan  1 00:09 sdcard/outputtestfile.txt
root@buildroot:~# head sdcard/outputtestfile.txt
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
root@buildroot:~# tail sdcard/outputtestfile.txt
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
Testing sdcard write speed
```

5. Unmounting / mounting again and listing the content of the card:
```
root@buildroot:~# umount sdcard
root@buildroot:~# mount /dev/mmcblk0p1 sdcard
[  667.107812] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
root@buildroot:~# ls -la sdcard/outputtestfile.txt
-rw-r--r--    1 root     root        512000 Jan  1 00:09 sdcard/outputtestfile.txt
```

enjoy-digital, 2020-07-22:
An initial [HOWTO](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/HOWTO.md) document has been created with the content if this issue. The SDCard is still in TODO since we'll have to update it when updating the driver. We can continue filling information there for un-documented/new peripherals.

closed 2020-07-22 by enjoy-digital