litex-hub/linux-on-litex-vexriscv #45: Add reboot support

enjoy-digital, 2019-08-30:
We could support reboot by writing 1 to `CSR_CTRL_RESET_ADDR` register at the end of the Linux reboot sequence to reset the CPU.

geertu, 2020-12-22:
I have added added a restart handler in https://github.com/geertu/linux/commit/7d09dc0797a8208a11eb7c0c2156c1a4c120180f

As a different version of drivers/soc/litex/litex_soc_ctrl.c was merged upstream, this will need to be rebased after the release of v5.11-rc1 (as is most of the LiteX Linux drivers?).

enjoy-digital, 2020-12-22:
Great @geertu, that will be very useful, thanks.

enjoy-digital, 2022-03-18:
This is now working, so I suppose you changes have landed @geertu :), thanks!

closed 2022-03-18 by enjoy-digital

geertu, 2022-03-18:
> This is now working, so I suppose you changes have landed @geertu :), thanks!

Yes, as of Linux v5.12.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3706f9f76a4f79f8e7b2eb8b99877e89fe9ad732