litex-hub/linux-on-litex-vexriscv #47: Support mutlicore CPU configuration

mithro, 2019-09-15:
We should extend this repository to support multicore configurations. To support this, there is a bunch of work that would need to be done, specially to VexRISCV.

@SpinalHDL / @Dolu1990 - Could you enumerate what would be needed?

Dolu1990, 2019-09-15:
Long answer there : https://github.com/SpinalHDL/VexRiscv/issues/85#issuecomment-531548389
But to resume : 
- Coherent D$
- Coherent interconnect 
- But first i would like to boost the I$ D$ memory bus data width first. Would that be fine in litex ?

enjoy-digital, 2019-09-15:
@Dolu1990: thanks. Yes increasing the memory bus data width will be possible (and we'll probably work on that soon since this is also useful for Rocket that is currently down-converted to 32-bit (just to have a first working version) but should be 64-bit).

enjoy-digital, 2020-07-20:
#47 and #126 are duplicate, let's continue the discussion in https://github.com/litex-hub/linux-on-litex-vexriscv/issues/126

closed 2020-07-20 by enjoy-digital