litex-hub/linux-on-litex-vexriscv #48: run in NEXYS4DDR stalled, and Simulation stalled in boot Linux

dhm2013724, 2019-09-18:
Hi everyone, I want to test this project in my Nexys4 DDR. I followed the README, there are no problems during make and install phase. But I met two problems in the last phase. 
First, the simulation stopped at booting Linux for too long times during sim.py. Should I still be waiting for it? 
[image]
Second, when I tested it in Nexys4DDR, it stopped at [LXTERM] Starting....
[image]
I have no idea, can anyone give me some advices? Thanks!

dhm2013724, 2019-09-18:
After I modified two speed values(the speed in shell from 2e6 to 115200, the baudrate in sco_linux.py from 1e6 to 115200), some better things turned out.
[image]
But it still thrown out other errors....
[image]

dhm2013724, 2019-09-18:
I tried to re-build the bitstream, and now it shows like [issue#15](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/15)
[image]

dhm2013724, 2019-09-18:
I also tried to boot from TFTP, but it returned the similar result.
[image]

mithro, 2019-09-18:
I believe @shenki just ran into the same issue?

dhm2013724, 2019-09-19:
Now, the simulation and real hardware test stalled at the same point......Booting Linux.

enjoy-digital, 2019-09-25:
This is now working correctly (i just re-tested it with updated prebuilt images). There was an issue in the .dtb generation that i introduced when testing the framebuffer with the NeTV2. That's now fixed.

closed 2019-09-25 by enjoy-digital