litex-hub/linux-on-litex-vexriscv #50: Running sim.py fails

scanakci, 2019-09-19:
Sim.py fails during compilation. By looking at the error, I can see that the libraries are specified in the middle of command instead of the end of the command. This causes a bunch of undefined reference to library errors. 

```
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=1 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -Wall -O3 -ggdb  -I/home/scanakci/Research_sado/Litex/litex/litex/build/sim/core   -c -o Vdut__ALLsup.o Vdut__ALLsup.cpp
      Archiving Vdut__ALL.a ...
ar r Vdut__ALL.a Vdut__ALLcls.o Vdut__ALLsup.o
ar: creating Vdut__ALL.a
ranlib Vdut__ALL.a
**g++    veril.o dut_init.o verilated.o verilated_dpi.o verilated_vcd_c.o Vdut__ALL.a   -lpthread -ljson-c -lm -lstdc++ -ldl -levent libdylib.o modules.o pads.o parse.o sim.o  -o Vdut -lm -lstdc++  2>&1 | c++filt**
libdylib.o: In function `platform_raw_open':
/home/scanakci/Research_sado/Litex/litex/litex/build/sim/core/libdylib.c:102: undefined reference to `dlopen'
libdylib.o: In function `platform_set_last_error':
```

enjoy-digital, 2019-09-24:
@scanakci sorry for the delay, have you tested with the prebuilt Linux images or with images you generated? (the prebuilt images should probably be updated).

scanakci, 2019-09-24:
I tested the prebuild Linux images. When I use a commit from July, I could boot up. Probably, the prebuilt images should be updated as you said.

enjoy-digital, 2019-09-24:
Ah ok thanks for the feedback, i'll update the bistreams/images soon.

enjoy-digital, 2019-09-25:
The prebuilt images have been updated and simulation re-tested with it.

closed 2019-09-25 by enjoy-digital