litex-hub/linux-on-litex-vexriscv #57: Missing `dtc` dependency

xobs, 2019-11-17:
```
user@Pazuzu:~/Code/HaD/linux-on-litex-vexriscv$ python3 ./make.py --board ulx3s
**sh: 1: dtc: not found**
 CC       isr.o
 CC       framebuffer.o
 CC       main.o
 LD       emulator.elf
riscv64-unknown-elf-ld: cannot find ../build/ulx3s/software/libbase/crt0-vexriscv-ctr.o: No such file or directory
riscv64-unknown-elf-ld: cannot find -lbase-nofloat
riscv64-unknown-elf-ld: cannot find -lcompiler_rt
Makefile:19: recipe for target 'emulator.elf' failed
make: *** [emulator.elf] Error 1
```

enjoy-digital, 2019-11-26:
Thanks for the feedback, this is fixed with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/07840a154e0ed1e94191641486f1fbb2887484d5.

mithro, 2019-11-26:
07840a1 fixes the emulator, but the build instructions still need to be updated to include installing dtc?

enjoy-digital, 2019-11-26:
Ah ok, sorry i missread the the issue. dtc is be installed with `apt install device-tree-compiler` which is part of the prerequisites: https://github.com/litex-hub/linux-on-litex-vexriscv#prerequisites

closed 2019-11-26 by enjoy-digital