litex-hub/linux-on-litex-vexriscv #65: Documents/guides about (net)booting Linux ?

Ruinland, 2019-12-27:
Hi.
Sorry for making noises here.

I've been working around Alpine Linux RV32 port during my day job and I am willing to bring it to VexRiscv  (kinda related to #54 ).

As I'm trying to side-load Linux/rootfs/dtb ...... to `versa_ecp5` board via network, I'm confused by the instructions in README.

I've setup a tftp server with needed files placed on my laptop, which is hooked up to versa_ecp5 directly. The laptop has been configured the ip address `192.168.1.100` on that NIC properly.

I typed `./make --board=versa_ecp5 --remote-ip=192.168.1.100 --local-ip=192.168.1.50`
Judging by the output of command, it seems that the board is correctly loaded with gateware. (Sparkling LEDs went out.)
Yet it seems to fail to fetch kernel and stuffs from my laptop.

Also, I don't understand the meaning of "local-ip" option. Would it be configured by Litex SOC, which has its own BIOS ? If not, then I'm kinda lost about how to set it up.

I tried to find the more detailed documents but in vain.

enjoy-digital, 2019-12-27:
Hi @Ruinland,

your procedure seems correct. Can you provide the logs of the BIOS on the UART? Do you see any ARP/TFTP transactions happening between `192.168.1.50` and `192.168.1.100`?

mithro, 2019-12-27:
Kind of related -- while there **no** support for the ECP5 Versa board in LiteX-BuildEnv (which also supports Linux) there is a comprehensive tutorial for the Digilent Arty at https://github.com/timvideos/litex-buildenv/wiki/HowTo-LCA2018-FPGA-Miniconf-VexRiscv-Renode

Ruinland, 2020-05-20:
Sorry for digging up the dirt.
Currently I have my Alpine Linux on booting successfully with ECP Versa 5G dev board successfully.

Here's a [short note](https://gist.github.com/Ruinland/f07eaf8a651b441fd4ac4c2341616af2) I created.

However, I failed to boot my freshly generated Litex/RocketChip bitstream on a ULX3S board (equipped with ECP5-85F) through serial boot. The command I used is pretty much the same - -

```
ujprog top.svf && \
lxterm --no-crc --serial-boot --kernel=~/boot.bin ---kernel-adr 0x40000000 /dev/ttyUSB0
```

After `boot.bin` being uploaded successfully, the LiteX bootloader refuse to boot.
The console spills character 'C' and 'K' before dropping to shell.

With some investigation (by inserting printf() code into bios/boot.c),
I found out it's caused by a CRC check fail and command abort (sent by lxterm ?) handling,
which makes little sense to me since I specified `--no-crc` with lxterm.

The `boot.bin` I'm using is scavenged from [gsomlo's webpage](http://www.contrib.andrew.cmu.edu/~somlo/BTCP/). 
I know that LiteX has changed memory map from time to time, so it might not boot correctly.
Yet if I understand correctly, lxterm shall not give `SFL_CMD_ABORT` to bootloader under this circumstances?

= = = = = =

By the way, I really appreciate this great project and all the hard works have been done.

If nothing goes wrong (e.g. COVID-19 pandemic erupts again),
I'm going to present [my own experience on porting Alpine Linux to LiteX/VexRiscv](https://pretalx.com/coscup2020/talk/review/3KU77FFCHWLC9KN3NSYL8C8NGZHGV8RK) this August in a Taiwan-based conference, [COSCUP](https://coscup.org/2020/landing/en.html).

If there's anything I should be aware of, e.g. properly referencing LiteX/VexRiscv with DOI serial number, please tell me :-)

enjoy-digital, 2020-07-22:
@Ruinland: there is now a page on the LiteX Wiki explaining the possible boot options: https://github.com/enjoy-digital/litex/wiki/Load-Application-Code-To-CPU

Sorry for the late answer, but the issue with Rocket is that the `main_ram` is mapped differently: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/rocket/core.py#L86 so you have to change the l`xterm` command to:
`lxterm --no-crc --serial-boot --kernel=~/boot.bin ---kernel-adr 0x80000000 /dev/ttyUSB0`

Otherwise, thanks for your feedback, we'll be happy to hear about your future work/presentation and could add a link to it in the wiki.

I'm closing since this should be resolved, but please re-open is you still have troubles loading the binary for Rocket/Linux.

closed 2020-07-22 by enjoy-digital