litex-hub/linux-on-litex-vexriscv #67: SPI Boot fails on Arty7

ric96, 2019-12-31:
I have a REVC Arty735T with a 25q125a Flash...

The flash process seems to proceed fine:
```
[sahaj@fedora-work linux-on-litex-vexriscv]$ ./make.py --board=arty --flash
make: Entering directory '/data/linux-on-litex-vexriscv/build/arty/software/libcompiler_rt'
make: Nothing to be done for 'all'.
make: Leaving directory '/data/linux-on-litex-vexriscv/build/arty/software/libcompiler_rt'
make: Entering directory '/data/linux-on-litex-vexriscv/build/arty/software/libbase'
 CC       exception.o
 CC       system.o
 CC       id.o
 CC       uart.o
 CC       time.o
 CC       spiflash.o
 CC       mdio.o
 AR       libbase.a
 AR       libbase-nofloat.a
make: Leaving directory '/data/linux-on-litex-vexriscv/build/arty/software/libbase'
make: Entering directory '/data/linux-on-litex-vexriscv/build/arty/software/libnet'
 CC       microudp.o
 AR       libnet.a
make: Leaving directory '/data/linux-on-litex-vexriscv/build/arty/software/libnet'
make: Entering directory '/data/linux-on-litex-vexriscv/build/arty/software/bios'
 CC       isr.o
 CC       sdram.o
 CC       main.o
 CC       boot.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
make: Leaving directory '/data/linux-on-litex-vexriscv/build/arty/software/bios'
build/arty/arty.dts:142.5-35: Warning (reg_format): /soc/spiflash@f0005000/flash@0:reg: property has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
buildroot/rv32.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
buildroot/rv32.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
buildroot/rv32.dtb: Warning (simple_bus_reg): Failed prerequisite 'reg_format'
buildroot/rv32.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
build/arty/arty.dts:116.29-134.8: Warning (spi_bus_bridge): /soc/spi@f0008000: incorrect #size-cells for SPI bus
buildroot/rv32.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
buildroot/rv32.dtb: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'
build/arty/arty.dts:140.19-143.9: Warning (avoid_default_addr_size): /soc/spiflash@f0005000/flash@0: Relying on default #address-cells value
build/arty/arty.dts:140.19-143.9: Warning (avoid_default_addr_size): /soc/spiflash@f0005000/flash@0: Relying on default #size-cells value
buildroot/rv32.dtb: Warning (avoid_unnecessary_addr_size): Failed prerequisite 'avoid_default_addr_size'
buildroot/rv32.dtb: Warning (unique_unit_address): Failed prerequisite 'avoid_default_addr_size'
 CC       isr.o
 CC       framebuffer.o
 CC       main.o
 LD       emulator.elf
chmod -x emulator.elf
 OBJCOPY  emulator.bin
chmod -x emulator.bin
Flashing build/arty/gateware/top.bin at 0x00000000
Open On-Chip Debugger 0.10.0+dev-00985-g2dc88e14 (2019-12-31-14:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
fpga_program
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 25000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
Info : sector 0 took 242 ms
Info : sector 1 took 238 ms
[... 345 lines cut ...]

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Loading emulator.bin from flash...
Error: Invalid image length 0xffffffff
Booting from flash...
Error: Invalid image length 0xffffffff
Booting from network...
Local IP : 192.168.1.50
Remote IP: 192.168.1.100
Fetching from: UDP/69
Unable to download Image over TFTP
Network boot failed
Unable to download Linux images, falling back to boot.bin
Unable to download boot.bin over TFTP
Network boot failed
No boot medium found

--============= Console ================--
litex> 

```

enjoy-digital, 2019-12-31:
Booting the kernel images from the SPI flash is indeed currently broken due to changes in flashboot in LiteX and needs to be fixed. Can you try booting using from Ethernet or UART while things are broken? I'll try to fix that soon.

ric96, 2019-12-31:
Serial works fine.

enjoy-digital, 2020-02-18:
This has been fixed.

closed 2020-02-18 by enjoy-digital