litex-hub/linux-on-litex-vexriscv #74: Conflict between csr region and main_ram region

mithro, 2020-01-17:
A number of the builds are failing with;
```
$ ./make.py --board=Genesys2
Traceback (most recent call last):
  File "./make.py", line 332, in <module>
    main()
  File "./make.py", line 283, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 162, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 55, in __init__
    soc_cls.__init__(self, cpu_type="vexriscv", cpu_variant=cpu_variant, uart_baudrate=1e6, **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/genesys2.py", line 74, in __init__
    BaseSoC.__init__(self, **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/genesys2.py", line 63, in __init__
    timing_settings     = sdram_module.timing_settings)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_sdram.py", line 100, in register_sdram
    self.register_mem("main_ram", self.mem_map["main_ram"], wb_sdram, main_ram_size)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_core.py", line 426, in register_mem
    self.add_memory_region(name, address, size)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_core.py", line 421, in add_memory_region
    o1, self.mem_regions[o1],
ValueError: Memory region conflict between csr (<SoCMemRegion 0xf0000000 0x10000 cached>) and main_ram (<SoCMemRegion 0xc0000000 0x40000000 cached>)
The command "./make.py --board=$BOARD" exited with 1.
```

```
$ ./make.py --board=KC705
Traceback (most recent call last):
  File "./make.py", line 332, in <module>
    main()
  File "./make.py", line 283, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 162, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 55, in __init__
    soc_cls.__init__(self, cpu_type="vexriscv", cpu_variant=cpu_variant, uart_baudrate=1e6, **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/kc705.py", line 65, in __init__
    timing_settings = sdram_module.timing_settings)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_sdram.py", line 100, in register_sdram
    self.register_mem("main_ram", self.mem_map["main_ram"], wb_sdram, main_ram_size)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_core.py", line 426, in register_mem
    self.add_memory_region(name, address, size)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_core.py", line 421, in add_memory_region
    o1, self.mem_regions[o1],
ValueError: Memory region conflict between csr (<SoCMemRegion 0xf0000000 0x10000 cached>) and main_ram (<SoCMemRegion 0xc0000000 0x40000000 cached>)
The command "./make.py --board=$BOARD" exited with 1.
```

See https://travis-ci.com/mithro/linux-on-litex-vexriscv/builds/144890173

enjoy-digital, 2020-01-17:
This happened for boards with large SDRAM and where we are not able to map the entire SDRAM without creating conflict with other peripherals. https://github.com/enjoy-digital/litex/commit/5913c91caaaf29b33a7b498093395ff47e596651 and https://github.com/litex-hub/linux-on-litex-vexriscv/commit/76fe0db4028d5ce1a8eb33ed02d8fd3a7348c421 limit  the mapped SDRAM size to 256MB and removes the conflict.

In the future, we could increase this when reorganizing the memory map of SoCLinux, but for now i didn't wanted to touch it.

mithro, 2020-01-17:
Now I'm getting `KCU105` board;
```
Traceback (most recent call last):
  File "./make.py", line 332, in <module>
    main()
  File "./make.py", line 283, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 168, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 61, in __init__
    **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/kcu105.py", line 111, in __init__
    BaseSoC.__init__(self, **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/kcu105.py", line 100, in __init__
    main_ram_size_limit = 0x40000000)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex/litex/soc/integration/soc_sdram.py", line 58, in register_sdram
    **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litedram/litedram/core/__init__.py", line 26, in __init__
    **kwargs)
TypeError: __init__() got an unexpected keyword argument 'main_ram_size_limit'
The command "./make.py --board="$BOARD"" exited with 1.
```

enjoy-digital, 2020-01-17:
Ah sorry, forgot to push to litex-boards. It should be good now.

mithro, 2020-01-17:
My suggestion for changing the mem_map is in https://github.com/litex-hub/linux-on-litex-vexriscv/pull/75

enjoy-digital, 2020-01-17:
Thanks, i just saw that, i'm testing.

closed 2020-01-20 by enjoy-digital