litex-hub/linux-on-litex-vexriscv #79: Error linker when executing simulator

kamejoko80, 2020-01-20:
Hello,

I not sure this is really an issue or not when invoking the command "python3 sim.py" I got the following error: 

``
g++    veril.o dut_init.o verilated.o verilated_dpi.o verilated_vcd_c.o Vdut__ALL.a   -lpthread -ljson-c -lm -lstdc++ -ldl -levent libdylib.o modules.o pads.o parse.o sim.o  -o Vdut -lm -lstdc++  2>&1 | c++filt
libdylib.o: In function `platform_raw_open':
/home/phuong/Workspace/FPGA/LITEX/mainline/linux-on-litex-vexriscv/litex/litex/build/sim/core/libdylib.c:102: undefined reference to `dlopen'
libdylib.o: In function `platform_set_last_error':
/home/phuong/Workspace/FPGA/LITEX/mainline/linux-on-litex-vexriscv/litex/litex/build/sim/core/libdylib.c:97: undefined reference to `dlerror'
libdylib.o: In function `platform_raw_close':
/home/phuong/Workspace/FPGA/LITEX/mainline/linux-on-litex-vexriscv/litex/litex/build/sim/core/libdylib.c:113: undefined reference to `dlclose'
libdylib.o: In function `platform_set_last_error':
/home/phuong/Workspace/FPGA/LITEX/mainline/linux-on-litex-vexriscv/litex/litex/build/sim/core/libdylib.c:97: undefined reference to `dlerror
``

Temporally I could fix the compilation error by editing the makefile in a folder:
\linux-on-litex-vexriscv\litex\litex\build\sim\core\Makefile

Replace the bellow line:

```diff
- LDFLAGS = -lpthread -ljson-c -lm -lstdc++ -ldl -levent
+ LDFLAGS = -lpthread -Wl,--no-as-needed -ljson-c -lm -lstdc++ -Wl,--no-as-needed -ldl -levent
```

I'm using ubuntu 16.04 LTS, does anyone run to the same problem with me?

enjoy-digital, 2020-01-20:
Hello @kamejoko80,

thanks for reporting. Most of use are probably using a more recent Ubuntu distribution (18.04 on my side). Since your patch is not causing issue on 18.04, i integrated it in directly in LiteX: https://github.com/enjoy-digital/litex/commit/a0d95766acc33dcbd870dca343d94da153ac72df. You should now be able to simulate without patching.

kamejoko80, 2020-01-20:
Hello enjoy-digital,

It works now.
Thanks for the update.

closed 2020-01-20 by kamejoko80