litex-hub/linux-on-litex-vexriscv #85: Build Error: Nexys Video Ethernet Reset multi-driven net

mescoba1, 2020-01-25:
```
CRITICAL WARNING: [Synth 8-3352] multi-driven net eth_tx_rst with 1st driver pin 'FDPE_11/Q' [/home/miguel/hdl-playground/litex/linux-on-litex-vexriscv/build/nexys_video/gateware/top.v:18687]
CRITICAL WARNING: [Synth 8-3352] multi-driven net eth_tx_rst with 2nd driver pin 'FDPE_15/Q' [/home/miguel/hdl-playground/litex/linux-on-litex-vexriscv/build/nexys_video/gateware/top.v:18727]
...
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net eth_tx_rst has multiple drivers: FDPE_15/Q, and FDPE_11/Q.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.
```

Thought maybe this was a copy-pasta error, but `eth_rx_rst` get driven in `FDPE_17`

mescoba1, 2020-01-26:
The problem looks to be with liteeth (or the `EthernetSoC` instantiation of `LiteEthPHYRGMII`), which looks like is not supported on nexys_video based on the README. The bug with `nexys_video.EthernetSoC` is that something is generating TWO double-flop synchronizers for the async reset of `eth_tx_rst`

Error occurs when running `./make.py --board=nexys_video --build`, looks like it is trying to build SoC with ethernet by default. I am able to get a bitfile with using `nexys_video.BaseSoC`
```
Board.__init__(self, nexys_video.EthernetSoC, {"serial", "framebuffer"})
```

enjoy-digital, 2020-01-28:
Thanks for reporting, this has been fixed with https://github.com/enjoy-digital/liteeth/pull/28.

closed 2020-01-28 by enjoy-digital