litex-hub/linux-on-litex-vexriscv #87: pre-built bitstream (genesys and kc705)

winner96, 2020-01-28:
Hi. I'm newbie to these works.

I wants to build softcore and linux on it. (Maybe KC705 will be used)

Looking at supported boards, I can see the numbers of FPGA device of KC705 and Genesys2  are same. So why are they have same numbers.

In addition, there is pre-built bitstream only for Genesys2 not for KC705. Can I use that for my KC705 board?

Thanks :D

enjoy-digital, 2020-01-28:
Hi @winner96,

the KC705 and Genesys2 are using the same FPGA, but since FPGA have very flexible IOs, a different pinout is used on these two boards, so we need two different platform support and thus different bitsteams. So you can't use the pre-built bitstream of the Genesys2 on the KC705.

The pre-built bitstream of the KC705 is indeed not available, i will try to generate it in the next days.

winner96, 2020-01-30:
Thanks @enjoy-digital  :)

It helped me a lot.

closed 2020-02-26 by winner96