litex-hub/linux-on-litex-vexriscv #88: Linux doesn't boot on KC705 with the generated emulator.bin file

atommann, 2020-01-28:
Hi,

The board I'm testing is KC705.

At beginning I followed the steps according the steps in repo's README.

I did the "Pre-built Bitstreams/Linux images" step. i.e.

```sh
$ git clone https://github.com/enjoy-digital/linux-on-litex-vexriscv-prebuilt
$ cp -r linux-on-litex-vexriscv-prebuilt/* linux-on-litex-vexriscv
```
Then I built the bitstream:

```sh
$ ./make.py --board=kc705 --build
```
Then loaded the bitstream to the board.

```sh
$ ./make.py --board=kc705 --load
```
It works fine. Now load Linux Image.

```sh
$ lxterm --images=images.json /dev/ttyUSB0 --speed=921600 --no-crc
```
> **Note**: I changed the baudrate in soc_linux.py (line 97) from 1e6 to 921600 (The maximum baudrate CP2103 listed in its datasheet)

Now press the RESET button, KC705 starts to eat the bits, after the transmission finished, it stopped at this position:

```
--============= Liftoff! ===============--
VexRiscv Machine Mode software built Sep 25 2019 10:40:12
--========== Booting Linux =============--
[    0.000000] No DT
```
I then did some experiments. At the end I found that if only copy emulator.bin in the prebuilt folder to overwrite the new one, then execute the "Load the Linux images over Serial" step again, it boots!

```sh
$ cp linux-on-litex-vexriscv-prebuilt/emulator/emulator.bin linux-on-litex-vexriscv/emulator/
```
I'm not sure if this is a bug or not. But it works for me.

enjoy-digital, 2020-02-28:
@atommann: thanks for the feedback, i've been trying to reproduce the issue without success.
Can you try with the `emulator.bin` attached? (i just tested it on a KC705) If it works, can you try to update LiteX/LiteX-boards and Linux-on-LiteX-Vexrisscv repositories and re-generate and `emulator.bin`? I'm also attaching the KC705 bitstream i used.

If it's working with the `emulator.bin` attached and not the one you generate, it's probably a  RISC-V toolchain issue. 

[emulator.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4266276/emulator.zip)
[kc705_bitstream.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4266283/kc705_bitstream.zip)

enjoy-digital, 2020-06-23:
Closing since no update.

closed 2020-06-23 by enjoy-digital

atommann, 2020-07-01:
Sorry for the late response.

Today I updated LiteX and related repositories, then everything works fine. I guess I made some mistakes before.

Thanks.

enjoy-digital, 2020-07-01:
@atommann: no problem, thanks for the feedback. That's also possible we fixed things since then.