litex-hub/linux-on-litex-vexriscv #89: Boot stalls on Genesys2 Board

genesys2, 2020-01-30:
Hi,

I've been trying to boot Linux on my Genesys2 board but it fails early in the boot sequence. This is the output I get: 

[LXTERM] Starting....

litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs Ltd

 BIOS built on Sep 25 2019 08:20:57
 BIOS CRC passed (afc9b4b4)

 Migen git sha1: f4fcd10
 LiteX git sha1: 1425a68d

--============ SoC info ================--
CPU:       VexRiscv @ 125MHz
ROM:       32KB
SRAM:      32KB
L2:        8KB
MAIN-RAM:  524288KB

--========= Peripherals init ===========--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |00000000000111111111111000| delay: 11
m1: |00000000000011111111111100| delay: 12
m2: |00000000011111111111100000| delay: 09
m3: |00000000011111111111100000| delay: 09
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00001111111100000000000000000000| delays: 08+-04
m0, b3: |00000000000000001111111110000000| delays: 20+-04
m0, b4: |00000000000000000000000000000011| delays: 31+-01
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b3 delays: 20+-04
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000111111100000000000000000000| delays: 08+-03
m1, b3: |00000000000000000011111100000000| delays: 21+-03
m1, b4: |00000000000000000000000000000001| delays: 31+-00
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b2 delays: 07+-03
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |11100000000000000000000000000000| delays: 01+-01
m2, b2: |00000011111111100000000000000000| delays: 10+-04
m2, b3: |00000000000000000000111111100000| delays: 23+-04
m2, b4: |00000000000000000000000000000000| delays: -
m2, b5: |00000000000000000000000000000000| delays: -
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |00000000000000000000000000000000| delays: -
best: m2, b2 delays: 10+-04
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |11111000000000000000000000000000| delays: 02+-02
m3, b2: |00000000111111111100000000000000| delays: 12+-04
m3, b3: |00000000000000000000001111111111| delays: 26+-04
m3, b4: |00000000000000000000000000000000| delays: -
m3, b5: |00000000000000000000000000000000| delays: -
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |00000000000000000000000000000000| delays: -
best: m3, b2 delays: 12+-04
SDRAM now under hardware control
Memtest OK

--========== Boot sequence =============--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0xc0000000 (5074340 bytes)...
[LXTERM] Upload complete (90.8KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0xc0800000 (4062720 bytes)...
[LXTERM] Upload complete (92.4KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0xc1000000 (1879 bytes)...
[LXTERM] Upload complete (99.4KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (10156 bytes)...
[LXTERM] Upload complete (94.7KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
KExecuting booted program at 0x20000000
--============= Liftoff! ===============--
VexRiscv Machine Mode software built Jan 30 2020 01:31:37
--========== Booting Linux =============--
[    0.000000] No DTB

I tried using the prebuild Linux as well as building it myself with the same result. There was a similar issue in this Git (#48 ) which might be related but I am not sure. 

Any help is appreciated. Thank you!

enjoy-digital, 2020-01-30:
Could it be similar to https://github.com/litex-hub/linux-on-litex-vexriscv/issues/88? Can you see if the workaround also works for you?

genesys2, 2020-01-30:
Thank you, didn't see that one. The workaround works for me.

Duplicate of #88

closed 2020-01-30 by genesys2