litex-hub/linux-on-litex-vexriscv #90: booting process

winner96, 2020-01-31:
Hi. I'm newbie.

I built KC705 bitstream and I will load it soon.

Before loading it on fpga can you help me to know more about this task?

As I know, general computers load bootloaders from BIOS and bootloaders are loading OS.

Can you teach me a little bit about how this project boot linux at start? ( for example, the address where bootloader or OS exists, or the works of core at the start)

Best regards.

winner96, 2020-01-31:
How can I know which bootloader is used? Thank :D

enjoy-digital, 2020-01-31:
Hi @winner96,

LiteX provides a BIOS that is used to load the Linux binaries to RAM and we then jump to it. That's what is described in the README at: https://github.com/litex-hub/linux-on-litex-vexriscv#load-the-linux-images-over-serial

Once you'll have Linux booting over serialboot, you could also test netboot that tries to get the binaries over TFTP from 192.168.1.100 by defaullt (but could be configured with --remote-ip parameter: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L265).

Some points to be aware of for the KC705:
- https://github.com/enjoy-digital/litex/issues/361
- https://github.com/litex-hub/linux-on-litex-vexriscv/issues/91
- https://github.com/litex-hub/linux-on-litex-vexriscv/issues/88

I'll try to revolve these soon.

enjoy-digital, 2020-02-14:
Closing since explanation has been provided and issues are open for specific KC705 points.

closed 2020-02-14 by enjoy-digital