litex-hub/linux-on-litex-vexriscv #93: Override dummy bytes spi flash?

pdp7, 2020-02-03:
Related to https://github.com/litex-hub/litex-boards/issues/35

The hadbadge is using 32MB SDRAM:
[Winbond W25Q128JVSIQ 128 Mbit NOR flash chip
](https://www.digikey.com/product-detail/en/W25Q128JVSIQ/W25Q128JVSIQ-ND/5803943/?itemSeq=316234062)

It requires 8 dummy clocks:
[image]

However, `SoCLinux` is set to use 11 dummy clocks, so I have to change it to 8:
```
diff --git a/soc_linux.py b/soc_linux.py
index 98f6d7e..8657cf4 100644
--- a/soc_linux.py
+++ b/soc_linux.py
@@ -66,7 +66,7 @@ def SoCLinux(soc_cls, **kwargs):
             spiflash_pads = self.platform.request("spiflash4x")
             self.submodules.spiflash = SpiFlash(
                 spiflash_pads,
-                dummy=11,
+                dummy=8,
                 div=2,
                 with_bitbang=True,
                 endianness=self.cpu.endianness)
```

Is there a way for me to change this just for hadbadge and not effect other boards?

pdp7, 2020-02-03:
fyi - @mithro suggested that an abstraction for SPI flash chips similar to LiteDRAM might be useful.

mithro, 2020-02-03:
I started something a while go;
 - https://docs.google.com/document/d/1JZ7pU7roOaJwsv_LlXDKpWrLMSp254-tuof5KIFcC-I/edit#
 - https://github.com/mithro/litex/tree/d7184b97494012a8c21ee93df231b758a61e9dec/litex/soc/cores/spi

enjoy-digital, 2020-02-03:
@mithro: sorry for not having look at it closer before. We really have now a strong need for that and i'll have a closer look at it in the next days/weeks.

mithro, 2020-02-03:
@enjoy-digital - I never got it to a stage where it actually worked, so I never sent a pull request. (I was also wondering if the spiflash stuff should be a separate `litespiflash` repository?)

I was also unsure how to deal with SPI parts which have dynamic dummy bit requirements -- maybe I should have just ignored them.

enjoy-digital, 2020-02-03:
The complexity here seems mostly related to the SPI Flash definitions, i'll try to think how complex things can becomes and evaluate if this make sense to have a separate repository, that's possible.
For the implementation, maybe we should start with something minimal for our current usecases (but while still preparing things for future features) and make it evolve as the features are needed.

mithro, 2020-02-03:
The module definition in that above repo looks pretty simple,
```python

class M25P05(SpiNorFlashModule):
    name = "m25p05"
    device_id = 0x00102020
    manufacturer_id   = SpiNorFlashManufacturerIDs.ST

    # OpCodes
    erase_opcode      = SpiNorFlashOpCodes.SE
    chip_erase_opcode = SpiNorFlashOpCodes.CHIP_ERASE

    # Sizes
    page_size   = 128
    sector_size = 0x8000
    total_pages = 0x10000
```

mithro, 2020-02-03:
```python
class N25Q00(SpiNorFlashModule):
    """Numonyx 1-gbit 3V, multiple IO, serial flash memory."""

    name = "n25q00"
    device_id = 0x00ba18
    #device_id = 0x001740ef
    manufacturer_id   = SpiNorFlashManufacturerIDs.MICRON

    dummy_bits = 10
    fmax = 108 * MHz

    # OpCodes
    erase_opcode      = SpiNorFlashOpCodes.SE
    chip_erase_opcode = SpiNorFlashOpCodes.CHIP_ERASE

    # Sizes
    page_size   = 256
    sector_size = 0x10000
    total_size  = bit2bytes(1 * Gbit)
```

enjoy-digital, 2020-02-03:
Yes, that seems nice and simple. If i was not clear, i wanted to say that the complexity comes from the characteristics of the SPI Flashes (that for most of them are handled by the software), not from the gateware core.

pdp7, 2020-02-11:
Is there anything I can do to help with the development or testing with regards to this?  thanks!

enjoy-digital, 2020-02-12:
@pdp7: thanks, i'll try to add minimal support soon from the initial work done by @mithro. Testing or adding new SPI Flash modules will be welcome.

enjoy-digital, 2020-02-14:
@pdp7: https://github.com/litex-hub/linux-on-litex-vexriscv/commit/da6eb79cbd44f77f629af14dc35668b98618f260 implements a temporary solution that should allow you to go further. Can you check https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L182-L183 are valid for you SPIFlash?

pdp7, 2020-02-14:
@enjoy-digital thanks for the update, I pulled the repos but am getting this error:
```
pdp7@x1:~/dev/litex-flash-test/linux-on-litex-vexriscv$ python3 make.py --board=hadbadge --build
Traceback (most recent call last):
  File "make.py", line 358, in <module>
    main()
  File "make.py", line 345, in main
    builder.build()
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/integration/builder.py", line 164, in build
    self.soc.finalize()
  File "/home/pdp7/dev/enjoy/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/integration/soc_sdram.py", line 126, in do_finalize
    SoCCore.do_finalize(self)
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/integration/soc_core.py", line 479, in do_finalize
    self._wb_slaves, register=True, timeout_cycles=self.wishbone_timeout_cycles)
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/interconnect/wishbone.py", line 167, in __init__
    self.submodules.decoder = Decoder(shared, slaves, register)
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/interconnect/wishbone.py", line 118, in __init__
    for i, (fun, bus) in enumerate(slaves)]
  File "/usr/local/lib/python3.7/dist-packages/litex-0.2.dev0-py3.7.egg/litex/soc/interconnect/wishbone.py", line 118, in <listcomp>
    for i, (fun, bus) in enumerate(slaves)]
TypeError: 'int' object is not callable
```
Any suggetsions?

pdp7, 2020-02-14:
Looks like I was just behind on the repo and have to git pull again.

It is now building OK:
```
 pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ git log --oneline
331ddce (HEAD -> master, origin/master, origin/HEAD) Merge pull request #103 from antmicro/sdram-timings
48e77a2 sim: add options to disable sdram timing checks and to enable verbose timing output
da6eb79 soc_linux/make: add spiflash dummy cycles support (temporary solution until better SPIFlash support is added to LiteX), enable SPIFlash on HadBadge

```
I'll test i out on the hardware next and see if i can read from the badge

pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ git log --oneline
331ddce (HEAD -> master, origin/master, origin/HEAD) Merge pull request #103 from antmicro/sdram-timings
48e77a2 sim: add options to disable sdram timing checks and to enable verbose timing output
da6eb79 soc_linux/make: add spiflash dummy cycles support (temporary solution until better SPIFlash support is added to LiteX), enable SPIFlash on HadBadge

pdp7, 2020-02-14:
@enjoy-digital yes, this does appear to work correctly.  I am able to read from flash:
```
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 14 2020 16:50:25
 BIOS CRC passed (9559e4d8)

 Migen git sha1: --------
 LiteX git sha1: --------

--=============== SoC ==================--
CPU:       VexRiscv @ 48MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 76Mbps Reads: 96Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Loading emulator.bin from flash...
Error: Invalid image length 0xffffffff
Booting from flash...
Error: Invalid image length 0xffffffff
No boot medium found

--============= Console ================--
litex> mr 0xd0000000 128
Memory dump:
0xd0000000  00 50 61 72 74 3a 20 4c 46 45 35 55 2d 34 35 46  .Part: LFE5U-45F
0xd0000010  2d 38 43 41 42 47 41 33 38 31 00 ff ff ff bd b3  -8CABGA381......
0xd0000020  ff ff ff ff 3b 00 00 00 e2 00 00 00 41 11 20 43  ....;.......A. C
0xd0000030  22 00 00 00 40 00 00 00 46 00 00 00 82 91 24 fe  "...@...F.....$.
0xd0000040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0xd0000050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0xd0000060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0xd0000070  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

pdp7, 2020-02-15:
@enjoy-digital this working ok for me.  Shall I close this issue?

enjoy-digital, 2020-02-16:
OK, thanks for the feedback.

closed 2020-02-16 by enjoy-digital