litex-hub/linux-on-litex-vexriscv #98: Error during 'sim.py' execution

natanvotre, 2020-02-09:
I am getting this error when I am trying to run `./sim.py`:

```
make: Entering directory '/home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
make: *** No rule to make target 'umodsi3.o', needed by 'libcompiler_rt.a'.  Stop.
make: Leaving directory '/home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
Traceback (most recent call last):
  File "./sim.py", line 224, in <module>
    main()
  File "./sim.py", line 215, in main
    trace_end   = int(args.trace_end))
  File "/home/natan/linux-on-litex-vexriscv/litex/soc/integration/builder.py", line 169, in build
    self._generate_software(not self.soc.integrated_rom_initialized)
  File "/home/natan/linux-on-litex-vexriscv/litex/soc/integration/builder.py", line 152, in _generate_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.5/subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt', '-f', '/home/natan/linux-on-litex-vexriscv/litex/soc/software/libcompiler_rt/Makefile']' returned non-zero exit status 2
```

I tried to run manually `make -C /home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt -f /home/natan/linux-on-litex-vexriscv/litex/soc/software/libcompiler_rt/Makefile` and I got this error

```
make: Entering directory '/home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
make: *** No rule to make target 'umodsi3.o', needed by 'libcompiler_rt.a'.  Stop.
make: Leaving directory '/home/natan/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
```

I think that riscv-gcc is not compiling properly, but I am a benniger on this platform and I couldn't solve.

Does someone can help me?
I tried to run using `Ubuntu 18.04`, docker and also a gcloud instace `Debian 9`

enjoy-digital, 2020-02-10:
it seems LiteX was not installed correctly and that compiler_rt submodule is missing. Can you tell us more how you installed LiteX? have you followed https://github.com/litex-hub/linux-on-litex-vexriscv#installing-litex to install LiteX? Can you check if compier_rt submodule has been correctly downloaded in `litex/soc/software/compiler_rt`?

natanvotre, 2020-02-10:
Yes, I followed these steps:

```
$ wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
$ chmod +x litex_setup.py
$ ./litex_setup.py init
$ sudo ./litex_setup.py install
```

During the `./litex_setup.py install` I got this report from `Litex` installation.

```
[installing litex]...
/usr/lib/python3/dist-packages/setuptools/dist.py:397: UserWarning: Normalizing '0.2.dev' to '0.2.dev0'
  normalized_version,
running develop
running egg_info
writing litex.egg-info/PKG-INFO
writing dependency_links to litex.egg-info/dependency_links.txt
writing entry points to litex.egg-info/entry_points.txt
writing requirements to litex.egg-info/requires.txt
writing top-level names to litex.egg-info/top_level.txt
reading manifest file 'litex.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'litex.egg-info/SOURCES.txt'
running build_ext
Creating /usr/local/lib/python3.6/dist-packages/litex.egg-link (link to .)
litex 0.2.dev0 is already the active version in easy-install.pth
Installing litex_read_verilog script to /usr/local/bin
Installing litex_server script to /usr/local/bin
Installing litex_sim script to /usr/local/bin
Installing litex_simple script to /usr/local/bin
Installing litex_term script to /usr/local/bin
Installing lxserver script to /usr/local/bin
Installing lxsim script to /usr/local/bin
Installing lxterm script to /usr/local/bin

Installed /opt/litex/litex
Processing dependencies for litex==0.2.dev0
Searching for pyserial==3.4
Best match: pyserial 3.4
Processing pyserial-3.4-py3.6.egg
pyserial 3.4 is already the active version in easy-install.pth
Installing miniterm.py script to /usr/local/bin

Using /usr/local/lib/python3.6/dist-packages/pyserial-3.4-py3.6.egg
Searching for migen==0.9.2
Best match: migen 0.9.2
migen 0.9.2 is already the active version in easy-install.pth

Using /opt/litex/migen
Searching for colorama==0.4.3
Best match: colorama 0.4.3
Processing colorama-0.4.3-py3.6.egg
colorama 0.4.3 is already the active version in easy-install.pth

Using /usr/local/lib/python3.6/dist-packages/colorama-0.4.3-py3.6.egg
Finished processing dependencies for litex==0.2.dev0
``` 

But, you are right. The folder `compiler_rt` is empty.

natanvotre, 2020-02-10:
I saw that `compiler_rt` was a submodule folder and something removed all the files inside.

I restored all the `compiler_rt`, installed litex again and I ran `.sim.py` again.

Now I got this error running manually `make -C /home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt -f /home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/libcompiler_rt/Makefile`:

```
make: Entering directory '/home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
 CC       umodsi3.o
In file included from /home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/compiler_rt/lib/builtins/int_lib.h:82,
                 from /home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/compiler_rt/lib/builtins/umodsi3.c:13:
/home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/compiler_rt/lib/builtins/int_types.h:70:1: error: unable to emulate 'TI'
 typedef int ti_int __attribute__((mode(TI)));
 ^~~~~~~
/home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/compiler_rt/lib/builtins/int_types.h:71:1: error: unable to emulate 'TI'
 typedef unsigned tu_int __attribute__((mode(TI)));
 ^~~~~~~
/home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/litex/litex/litex/soc/software/libcompiler_rt/Makefile:27: recipe for target 'umodsi3.o' failed
make: *** [umodsi3.o] Error 1
make: Leaving directory '/home/natan/Documents/Brickabode/FpgaResearch/riscV/spinalHDL/linux-on-litex-vexriscv/build/sim/software/libcompiler_rt'
```

enjoy-digital, 2020-02-12:
@natanvotre: i don't understand why compiler_rt was initially removed or empty. Can you try removing all repositories and doing a fresh install? If compiler_rt is still empty, can you post your install log?

natanvotre, 2020-02-12:
> @natanvotre: i don't understand why compiler_rt was initially removed or empty. Can you try removing all repositories and doing a fresh install? If compiler_rt is still empty, can you post your install log?

@enjoy-digital , I see, that was my mistake. For some reason I cleaned this repo with some command..
Anyway, I tried today with my other computer and I got this error

```
python3 -m litex.soc.software.mkmscimg bios.bin --little
/home/natan/anaconda3/bin/python3: Error while finding module specification for 'litex.soc.software.mkmscimg' (ModuleNotFoundError: No module named 'litex')
/home/natan/Documents/BrickAbode/FPGA/litex/litex/soc/software/bios/Makefile:18: recipe for target 'bios.bin' failed
make: *** [bios.bin] Error 1
make: Leaving directory '/home/natan/Documents/BrickAbode/FPGA/linux-on-litex-vexriscv/build/sim/software/bios'
Traceback (most recent call last):
  File "sim.py", line 224, in <module>
    main()
  File "sim.py", line 215, in main
    trace_end   = int(args.trace_end))
  File "/home/natan/Documents/BrickAbode/FPGA/litex/litex/soc/integration/builder.py", line 181, in build
    self._generate_software(not self.soc.integrated_rom_initialized)
  File "/home/natan/Documents/BrickAbode/FPGA/litex/litex/soc/integration/builder.py", line 164, in _generate_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/natan/Documents/BrickAbode/FPGA/linux-on-litex-vexriscv/build/sim/software/bios', '-f', '/home/natan/Documents/BrickAbode/FPGA/litex/litex/soc/software/bios/Makefile']' returned non-zero exit status 2.
```
and when I ran into the command line:
```
make: Entering directory '/home/natan/Documents/BrickAbode/FPGA/linux-on-litex-vexriscv/build/sim/software/bios'
 CC       isr.o
 CC       sdram.o
 CC       main.o
 CC       boot-helper-vexriscv.o
 CC       boot.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
/home/natan/anaconda3/bin/python3: Error while finding module specification for 'litex.soc.software.mkmscimg' (ModuleNotFoundError: No module named 'litex')
/home/natan/Documents/BrickAbode/FPGA/litex/litex/soc/software/bios/Makefile:18: recipe for target 'bios.bin' failed
make: *** [bios.bin] Error 1
make: Leaving directory '/home/natan/Documents/BrickAbode/FPGA/linux-on-litex-vexriscv/build/sim/software/bios'
```

After that, when I tried to mv the litex files and installed it again, the same error as the [first comment](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/98#issue-562222747) appeared. (probably was that my first error, I moved and accidentally removed somethings and got to this error.
After that, when I checkout to origin/master (commit 69445f095) appeared [this same](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/98#issuecomment-584127010) error.

natanvotre, 2020-02-13:
@enjoy-digital I solved the problem..
I have created a [Dockerfile](https://github.com/natanvotre/dockerfile-linux-on-litex-vexriscv) in order to set-up a machine for this repo, and it worked.

Probably the error was that I was using python3.6 and with some outdated libraries.
I need to confirm that, please don't close the issue yet.

enjoy-digital, 2020-02-19:
@natanvotre: have you been able to check what caused the issue?

freddy-, 2020-02-23:
I've had the same problem. The submodule compiler_rt was empty.
At first i thought that the submodule URL was offline or something, then I manually changed to `https://github.com/llvm-mirror/compiler-rt.git` and runned the command:
`git submodule update --init --recursive`
After that the SoC build worked.
Later I find that the original submodule URL was working correctly, I think it just doesn't open in the browser...

natanvotre, 2020-02-25:
> @natanvotre: have you been able to check what caused the issue?

It seems that it was just matter of environment.
With the docker, I could run either in pyhton3.6 or python3.7, That seemed to be the problem.

Anyway, thanks for helping me with this Issue and, either way, now we have a [Dockerfile](https://github.com/natanvotre/dockerfile-linux-on-litex-vexriscv) that can be used as reference!

closed 2020-02-25 by natanvotre