litex-hub/linux-on-litex-vexriscv #108: Arty board revisions have different flash

pdp7, 2020-02-21:
Issue: older Arty board flash needs dummy cycles of 11 while newer Arty boards need dummy cycles of 7.

@enjoy-digital wrote in https://github.com/litex-hub/litex-boards/issues/42#issuecomment-587122124:
> Note that i have an old Arty, and it seems the Flash on new versions of Arty is different (FL128SAIF00) so you'll probably have to change the dummy_cycles (it should be 11+-4).

Based on https://github.com/litex-hub/litex-boards/issues/42#issuecomment-589052025:
> Thanks! The Arty does work now with dummy bytes of 7. Dummy bytes of 8 was invalid.
> How should a variant of a board by handled?
>
> I seems like old Arty needs 11 but newer, like mine, needs SPIFLASH_DUMMY_CYCLES = 7

enjoy-digital, 2020-02-27:
@pdp7: support has been added with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/f1644ae9857ac9f1d75b6dac1a69ab1e9ffc72ec. Let's call the original Arty board Arty and the new one Arty A7. (that's how Digilent calls it https://reference.digilentinc.com/reference/programmable-logic/arty/start). This is now possible to build for it with: `./make.py --board=arty_a7 --build.`

closed 2020-02-27 by enjoy-digital

pdp7, 2020-02-28:
@enjoy-digital thank you!