litex-hub/linux-on-litex-vexriscv #109: SDRAM is too slow for framebuffer

sajattack, 2020-02-22:
I'm not sure if this should go in litevideo or here, but anyways, @gregdavill and I tried to get framebuffer to work on two different boards with sdram, and I measured that hsync ended up 2.3-2.4x slower than it should be during pixel output, but not during blanking. Presumably, fetching the pixels from sdram was slowing down hsync.  https://github.com/enjoy-digital/litevideo/tree/master/litevideo/terminal did work however. I'm not sure what action to suggest here, but documenting it is a good start.

enjoy-digital, 2020-02-22:
@sajattack: can you give more informations about your setup: SDRAM module and data width, system clock frequency and the resolution/refresh rate you were testing?

sajattack, 2020-02-22:
AS4C16M16 sdram module, 16bit data width, 50Mhz system clock, I tried multiple resolutions including 1280x720@60Hz, 640x480@75Hz, 640x480@60Hz and 320x240@60Hz.

sajattack, 2020-02-22:
I think it is fair to say the speeds required are outside the limits of most SDRAM modules though.

enjoy-digital, 2020-02-26:
With a 16-bit SDRAM at 50MHz, we can expect 16*50e6*0.8(80% efficiency)=640Mbps.
For 640x480@60Hz, we need 640*480*60*32=590Mbps.
Since the SDRAM is also shared with the CPU, we will indeed be very close to the SDRAM maximal bandwidth. The solution could be to run the SDRAM at 100MHz with a CPU still running at 50MHz.

sajattack, 2020-02-27:
It didn't even work at 320x240 (147Mbps) so I'm not sure if that's going to help.

enjoy-digital, 2020-07-20:
It's now possible to use a HalfRate PHY with SDRAM, see:
- https://github.com/enjoy-digital/litedram/pull/206
- https://github.com/litex-hub/litex-boards/pull/91
Which should provide ~2X the SDRAM bandwidth, so will help on this.

enjoy-digital, 2021-03-30:
This is now possible with the simplified Video Framebuffer and has been tested on Minispartan6 and ULX3S (SDRAM) in 1:1 rate mode: https://twitter.com/enjoy_digital/status/1376519927767650311 For more margin, we could enable the HalfRate PHY.

closed 2021-03-30 by enjoy-digital