litex-hub/linux-on-litex-vexriscv #112: Linux Boot Fails with modified fetch unit

genesys2, 2020-02-25:
Hi,

I've been trying to make changes to the fetch unit of Vex Riscv in a way, that no instruction is fetched until we know for certain, that it will be executed. So basically, we wait for the decode of the predecessor instruction or - in case it is a branch - for the branch target. The changes I made work fine with the GenFull and GenSmallest testbench of VexRiscv (the ones in the regression folder). However, if I try to boot linux with sim.py the boot fails relatively late with a kernel panic. This is the output of sim.py: 
````
--=============== SoC ==================--
CPU:       VexRiscv @ 1MHz
ROM:       32KB
SRAM:      4KB
MAIN-RAM:  32768KB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x20000000

--============= Liftoff! ===============--
VexRiscv Machine Mode software built Feb 25 2020 13:29:56
--========== Booting Linux =============--
[    0.000000] No DTB passed to the kernel
[    0.000000] Linux version 5.0.13 (florent@lab) (gcc version 8.3.0 (Buildroot 2019.08-rc2-00011-gad9efda578)) #1 Thu Sep 12 14:20:26 CEST 2019
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000c0000000-0x00000000c1ffffff]
[    0.000000] elf_hwcap is 0x1101
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M@0xc0000000 rootwait console=liteuart earlycon=sbi root=/dev/ram0 init=/sbin/init swiotlb=32
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Sorting __ex_table...
[    0.000000] Memory: 19812K/32768K available (3415K kernel code, 148K rwdata, 509K rodata, 140K init, 216K bss, 12956K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
[    0.000455] sched_clock: 64 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
[    0.006596] Console: colour dummy device 80x25
[    0.009637] Calibrating delay loop (skipped), value calculated using timer frequency.. 100.00 BogoMIPS (lpj=200000)
[    0.012566] pid_max: default: 32768 minimum: 301
[    0.017008] Unable to handle kernel paging request at virtual address 83f6b50c
[    0.018582] Oops [#1]
[    0.019601] CPU: 0 PID: 0 Comm: swapper Not tainted 5.0.13 #1
[    0.021078] sepc: c0181270 ra : c0178cf8 sp : c03fbe40
[    0.022425]  gp : c041d5b8 tp : c04032d0 t0 : 00000001
[    0.023769]  t1 : 000001f1 t2 : 0000020f s0 : c03fbe50
[    0.025109]  s1 : 00000068 a0 : c1fb5a80 a1 : 00000406
[    0.026425]  a2 : 00000069 a3 : 00000003 a4 : 00000008
[    0.027772]  a5 : 83f6b50c a6 : c1fb5580 a7 : 00000000
[    0.029109]  s2 : 00000406 s3 : c1fb5a80 s4 : 00000001
[    0.030438]  s5 : 00000000 s6 : fffffffe s7 : 00000003
[    0.031781]  s8 : 0000006b s9 : 006000c0 s10: c1fb5e80
[    0.033126]  s11: c03debdc t3 : 00000004 t4 : fffffffe
[    0.034399]  t5 : 00000400 t6 : 000fffff
[    0.035567] sstatus: 00000100 sbadaddr: 83f6b50c scause: 0000000d
[    0.037840] random: get_random_bytes called from print_oops_end_marker+0x34/0x74 with crng_init=0
[    0.039908] ---[ end trace 1ba7c24ce56d389e ]---
[    0.041143] Kernel panic - not syncing: Attempted to kill the idle task!
[    0.042767] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

````

I build the core using `sbt compile "runMain vexriscv.GenCoreDefault --csrPluginConfig linux-minimal --outputFile VexRiscv_Linux"`.

Do you have any idea where it could go wrong? Since the boot process gets so far, I assume that the modifications are correct in general but something goes wrong during the linux boot. 

I changed the fetchPC area in the fetcher plugin as follows:
```` val fetchPc = new Area{
      //PC calculation without Jump
      val output = Stream(UInt(32 bits))
      val pcReg = Reg(UInt(32 bits)) init(if(resetVector != null) resetVector else externalResetVector) addAttribute(Verilator.public)
      val corrected = False
      val pcRegPropagate = False
      val booted = RegNext(True) init (False)
      val branchTakenInfo = branchTakenInterface.getTakenInfo()
      val inc = RegInit(False) clearWhen(pcRegPropagate) setWhen(decode.arbitration.isFiring && decode.output(BRANCH_CTRL) === BranchCtrlEnum.INC || branchTakenInfo.branchNotTaken
      inc.clearWhen(inc)
    
      val pc = pcReg + (inc ## B"00").asUInt
      val requested = Reg(False) setWhen(booted  && (inc || corrected) || !booted)
      requested.clearWhen(output.fire)
      when(jump.pcLoad.valid) {
        corrected := True
        pc := jump.pcLoad.payload
      }
      when(booted && (corrected || inc)){
        pcReg := pc
      }

      pc(0) := False
      if(!pipeline(RVC_GEN)) pc(1) := False

      //output.valid := !fetcherHalt && booted
      output.valid := requested && booted
      output.payload := pc
    }
````

branchTakenInfo simply shows whether the branch is resolved but not taken. 

If you have any idea where to look, I would appreciate your help. So far, I tried looking at a lot of traces (which seem to be okay) and special cases like interrupts and context switches (which also seem to be okay).

Thanks a lot!

Dolu1990, 2020-02-25:
@genesys2 Hi,
Hmm such issue would be better on the VexRiscv repository, as i could miss things here, but no worries, let's continue here ^^

So, testing with only GenFull and GenSmallest isn't suffisent, basicaly in the VexRiscv repo, i have a special tester which generate random configs and run the regression on them. It has the advantage of exploring the design space which can show up bugs way faster.

Basicaly, to run it, you can just do a    "sbt test" in the SpinalHDL repo, it will take a while but should provide a better coverate.

Can you give a try ?
I should also document how to use the test flow efficently.

Then about the fetch stuff, i'm currently reworking it in the rework_fetch branch, That's currently working, but not on all configs ( 1950/2000 configs pass all test ).
As on some config it significantly improve performances, see 
https://github.com/SpinalHDL/VexRiscv/commit/999a868c1412796c0a230bc2802449055ece990a#diff-04c6e90faac2675aa89e2176d2eec7d8R103
i will merge it in dev as soon as all tests are ok, but it might have some conflicts with your implementation.

Do you have a repo with your changes ?

genesys2, 2020-02-26:
Hi, 

first, thank you very much for your help! I tried running sbt test and my modified version does fail more tests than the unmodified version (120 fails vs 83 fails in the original). This is the final result:

````
[info] - final_report
[info] Run completed in 1 hour, 24 minutes, 57 seconds.
[info] Total number of tests run: 222
[info] Suites: completed 2, aborted 0
[info] Tests: succeeded 102, failed 120, canceled 0, ignored 0, pending 0
[info] *** 120 TESTS FAILED ***
[error] Failed tests:
[error]         vexriscv.DhrystoneBench
[error]         vexriscv.TestIndividualFeatures
[error] (Test / test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 5099 s, completed Feb 26, 2020, 11:22:22 AM
````

What is the best method to further identify and investigate relevant tests that failed? It would be nice to have traces for the tests that failed my modified version but passed the original one. 

I tried building the core versions that behave differently, unfortunately I don't know the parameters for sbt and the makefile parameters to run the regression tests. 

Interoperability with the git version is for now not a major concern since I just need a demonstrator to run some performance benchmarks and show that the core works fine with the modifications (which is why booting Linux would be very nice). For now, I am working on a cloned version of the repo.

Dolu1990, 2020-02-26:
Hmmm did you clonned the repo recusively ? it look like it doesn't find the linux regression binaries (looking at the original failure ratio.

I have to add a clean way to rerun a specific test, and document it, i should be able to add that this weekend, i will update you on that asap, i'm just currently relocating in a new city and leaving a job, so quite busy right now :)

Also, likely you will break RVC with your changes, so you might want to disable RVC in the tests, i will also document about that. 

Just ping me if you see i forgot about it

genesys2, 2020-02-27:
Ah, good point, I didn't see that there was a linked repo so I did not check it out recurisvely. I am now re-running the test.

Thank you and no pressure :-)

Dolu1990, 2020-03-01:
@genesys2 I improved the test infrastructure and redocumented it :
https://github.com/SpinalHDL/VexRiscv/blob/dev/README.md#regression-tests

Now you should have everything you need to test your new feature, including disabling RVC via environnement variables.

In the future i will have to  multithread the testing of the different configuration, right now they are tested sequancialy.

genesys2, 2020-03-02:
Thanks a lot! :-)

Dolu1990, 2020-03-20:
@genesys2 I merged all my changes related to the fetch unit in dev/master

genesys2, 2020-03-23:
@Dolu1990 With a few changes my modification also works with the new version. Thanks again for your help!  :-)

closed 2020-03-23 by genesys2