litex-hub/linux-on-litex-vexriscv #113: porting kc705 not using commands

winner96, 2020-02-26:
Hi. I'm working on remote server ubuntu, and my fpga is connected to my local server PC.

As my FPGA is not directly connected, it makes me hard to implement with commands written on tutorials.

So here's my question. I wants to work it manually through remote server Vivado and my host hardware server.

1. I generated top.bit for kc705 on remote server. If I loaded only 'top.bit' via Program Device, does that mean vexriscv is on my KC705 board?
(I'm not sure if there is additional binaries like bios.bin for loading on)

2. Is it enough to load 'Image' in /buildroot for porting Linux?

3. I wants to see results on my monitor using UART, how can I check the baudrate?

It was Newbie's question thanks.

enjoy-digital, 2020-02-26:
Hi @winner96,

when the  top.bit is loaded, your FPGA is programmed with the Linux SoC (Vexriscv CPU + DRAM controller + base peripherals) and the bios is executed from an integrated ROM. 

The binaries that needs to be loaded to run Linux are the ones in https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/images.json.

For the UART, you need to configure the baudrate to the same baudrate used when generating the bitstream. On KC705 it's lowered to 500Kbauds: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L304

winner96, 2020-02-27:
Thanks for your detailed explanation :D

winner96, 2020-02-27:
@enjoy-digital 
Can you help me if there is problem with this?

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 27 2020 11:11:12
 BIOS CRC passed (7fbb537f)

 Migen git sha1: d11565a
 LiteX git sha1: f3f9808

--=============== SoC ==================--
CPU:       VexRiscv @ 125MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  262144KB

--========== Initialization ============--
Ethernet init...
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |10000000000000111111111111| delay: 14
m1: |00000000000000111111111111| delay: 14
m2: |11100000000000000111111111| delay: 17
m3: |11110000000000000111111111| delay: 17
m4: |11111111000000000000011111| delay: 00
m5: |11111111100000000000000111| delay: 00
m6: |11111111111000000000000011| delay: 00
m7: |11111111111000000000000001| delay: 00
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |11111000000000000000000000000000| delays: 02+-02
m0, b3: |00000000111111111100000000000000| delays: 12+-05
m0, b4: |00000000000000000000111111111100| delays: 25+-05
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b3 delays: 12+-04
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |11110000000000000000000000000000| delays: 02+-02
m1, b3: |00000001111111111000000000000000| delays: 12+-05
m1, b4: |00000000000000000001111111111100| delays: 24+-05
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b4 delays: 24+-05
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |11000000000000000000000000000000| delays: 01+-01
m2, b3: |00000111111111100000000000000000| delays: 10+-05
m2, b4: |00000000000000000011111111100000| delays: 22+-04
m2, b5: |00000000000000000000000000000001| delays: 31+-00
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |00000000000000000000000000000000| delays: -
best: m2, b3 delays: 10+-05
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |11000000000000000000000000000000| delays: 01+-01
m3, b3: |00000111111111100000000000000000| delays: 10+-05
m3, b4: |00000000000000000011111111100000| delays: 22+-04
m3, b5: |00000000000000000000000000000001| delays: 31+-00
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |00000000000000000000000000000000| delays: -
best: m3, b3 delays: 10+-05
m4, b0: |00000000000000000000000000000000| delays: -
m4, b1: |00000000000000000000000000000000| delays: -
m4, b2: |00000000000000000000000000000000| delays: -
m4, b3: |00000000000000000000000000000000| delays: -
m4, b4: |00000000000000000000000000000000| delays: -
m4, b5: |00000000000000000000000000000000| delays: -
m4, b6: |00000000000000000000000000000000| delays: -
m4, b7: |00000000000000000000000000000000| delays: -
best: m4, b0 delays: -
m5, b0: |00000000000000000000000000000000| delays: -
m5, b1: |00000000000000000000000000000000| delays: -
m5, b2: |00000000000000000000000000000000| delays: -
m5, b3: |00000000000000000000000000000000| delays: -
m5, b4: |00000000000000000000000000000000| delays: -
m5, b5: |00000000000000000000000000000000| delays: -
m5, b6: |00000000000000000000000000000000| delays: -
m5, b7: |00000000000000000000000000000000| delays: -
best: m5, b0 delays: -
m6, b0: |00000000000000000000000000000000| delays: -
m6, b1: |00000000000000000000000000000000| delays: -
m6, b2: |00000000000000000000000000000000| delays: -
m6, b3: |00000000000000000000000000000000| delays: -
m6, b4: |00000000000000000000000000000000| delays: -
m6, b5: |00000000000000000000000000000000| delays: -
m6, b6: |00000000000000000000000000000000| delays: -
m6, b7: |00000000000000000000000000000000| delays: -
best: m6, b0 delays: -
m7, b0: |00000000000000000000000000000000| delays: -
m7, b1: |00000000000000000000000000000000| delays: -
m7, b2: |00000000000000000000000000000000| delays: -
m7, b3: |00000000000000000000000000000000| delays: -
m7, b4: |00000000000000000000000000000000| delays: -
m7, b5: |00000000000000000000000000000000| delays: -
m7, b6: |00000000000000000000000000000000| delays: -
m7, b7: |00000000000000000000000000000000| delays: -
best: m7, b0 delays: -
SDRAM now under hardware control
Memtest bus failed: 128/256 errors
Memtest data failed: 262144/524288 errors
Memtest addr failed: 4096/8192 errors
Memory initialization failed

--============= Console ================--

//////////////////////////////////////////////////////////////////////////////

With your help, I can print out upper ones.

What's the problem with Memtest errors :(

Furthermore, referring other issues, I wondered why size of SRAM, L2 are different with others.
 I wants to know how can I generate other bitstream changing specs of SRAM and L2.

enjoy-digital, 2020-02-27:
For now with the KC705, you have to use this workaround: https://github.com/enjoy-digital/litex/issues/361. It will be fixed soon.

enjoy-digital, 2020-02-27:
@winner96: enjoy-digital/litex#361 has been fixed, you can now avoid the workaround.

winner96, 2020-02-28:
@enjoy-digital  :

Thanks :D

If you don't mind, can you let me know a little bit about  reason of problem? (Everything is new to me and it triggers my curiosity)

winner96, 2020-02-28:
@enjoy-digital 👍 

With your help, I can port my kc705 with serialboot.

At first trial it works with no problem. But rebooting makes error just like #15.

For my workaround, I removed files in images.json and copied new files. And it worked again.

Is this trivial problem..? Thanks

enjoy-digital, 2020-02-28:
@winner96: for the SDRAM fix, it's just that the software need a bit of help to calibrate correctly, we adjust the CLK/CMD delay vs DQ/DQS delay to center the write leveling window:

Before (delay = 0 taps):
````
Write leveling:
m0: |11000000000000011111111111| delay: 15
m1: |00000000000000111111111111| delay: 14
m2: |11110000000000000111111111| delay: 17
m3: |11110000000000000011111111| delay: 18
m4: |11111111110000000000000111| delay: 00
m5: |11111111110000000000000111| delay: 00
m6: |11111111111000000000000001| delay: 00
m7: |11111111111000000000000011| delay: 00
````

After (delay = 12 taps):
````
Write leveling:
m0: |11111111111111000000000000| delay: 00
m1: |11111111111100000000000001| delay: 00
m2: |00011111111111110000000000| delay: 03
m3: |00011111111111110000000000| delay: 03
m4: |00000000111111111111110000| delay: 08
m5: |00000000111111111111110000| delay: 08
m6: |00000000001111111111111000| delay: 10
m7: |00000000001111111111111000| delay: 10
````

in the future, we'll probably make this fully automated, but for now it's still required.

I'm not sure to understand the issue, how are you rebooting? Can you post the logs of the startup on reboot?

enjoy-digital, 2020-02-28:
Is is similar to https://github.com/litex-hub/linux-on-litex-vexriscv/issues/88?

winner96, 2020-02-29:
@enjoy-digital 
I can post logs of the startup on monday.

Issue is similar to #15 . At first trial everything is fine. And pushing reset button on KC705 start Vexriscv and serial boot again.

When all files are loaded, there is no progress after 'LIFT OFF!' like #15 , while #88 is showing
 --========== Booting Linux =============--
[    0.000000] No DT

Thanks

winner96, 2020-03-02:
sh@sh-VirtualBox:~/linux-on-litex-vexriscv$ sudo lxterm --images=images.json /dev/ttyUSB0 --speed=500000 --no-crc
[sudo] password for sh: 
[LXTERM] Starting....

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 28 2020 11:32:52
 BIOS CRC passed (78496e9f)

 Migen git sha1: e2e6c72
 LiteX git sha1: f3f9808

--=============== SoC ==================--
CPU:       VexRiscv @ 125MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  262144KB

--========== Initialization ============--
Ethernet init...
Initializing SDRAM...
Setting clk/cmd delay to 13 taps
SDRAM now under software control
Write leveling:
m0: |01111111111111000000000000| delay: 01
m1: |01111111111111000000000000| delay: 01
m2: |00011111111111110000000000| delay: 03
m3: |00001111111111111000000000| delay: 04
m4: |00000000011111111111100000| delay: 09
m5: |00000000011111111111111000| delay: 09
m6: |00000000000111111111111100| delay: 11
m7: |00000000000111111111111110| delay: 11
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |11111000000000000000000000000000| delays: 02+-02
m0, b4: |00000001111111111000000000000000| delays: 12+-05
m0, b5: |00000000000000000000111111111100| delays: 25+-05
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b4 delays: 12+-05
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |11110000000000000000000000000000| delays: 02+-02
m1, b4: |00000001111111110000000000000000| delays: 11+-04
m1, b5: |00000000000000000000111111111100| delays: 25+-05
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b5 delays: 25+-05
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |00000000000000000000000000000000| delays: -
m2, b3: |11000000000000000000000000000000| delays: 01+-01
m2, b4: |00000111111111000000000000000000| delays: 09+-04
m2, b5: |00000000000000000011111111100000| delays: 22+-04
m2, b6: |00000000000000000000000000000001| delays: 31+-00
m2, b7: |00000000000000000000000000000000| delays: -
best: m2, b4 delays: 09+-04
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |00000000000000000000000000000000| delays: -
m3, b3: |11000000000000000000000000000000| delays: 01+-01
m3, b4: |00000111111111000000000000000000| delays: 09+-04
m3, b5: |00000000000000000011111111110000| delays: 23+-05
m3, b6: |00000000000000000000000000000001| delays: 31+-00
m3, b7: |00000000000000000000000000000000| delays: -
best: m3, b5 delays: 23+-05
m4, b0: |00000000000000000000000000000000| delays: -
m4, b1: |00000000000000000000000000000000| delays: -
m4, b2: |00000000000000000000000000000000| delays: -
m4, b3: |00000000000000000000000000000000| delays: -
m4, b4: |11111111100000000000000000000000| delays: 04+-04
m4, b5: |00000000000111111111110000000000| delays: 17+-05
m4, b6: |00000000000000000000000001111111| delays: 28+-03
m4, b7: |00000000000000000000000000000000| delays: -
best: m4, b5 delays: 16+-05
m5, b0: |00000000000000000000000000000000| delays: -
m5, b1: |00000000000000000000000000000000| delays: -
m5, b2: |00000000000000000000000000000000| delays: -
m5, b3: |00000000000000000000000000000000| delays: -
m5, b4: |11111111000000000000000000000000| delays: 04+-04
m5, b5: |00000000000111111111100000000000| delays: 16+-05
m5, b6: |00000000000000000000000011111111| delays: 28+-04
m5, b7: |00000000000000000000000000000000| delays: -
best: m5, b5 delays: 16+-05
m6, b0: |00000000000000000000000000000000| delays: -
m6, b1: |00000000000000000000000000000000| delays: -
m6, b2: |00000000000000000000000000000000| delays: -
m6, b3: |00000000000000000000000000000000| delays: -
m6, b4: |11111110000000000000000000000000| delays: 03+-03
m6, b5: |00000000011111111111000000000000| delays: 14+-05
m6, b6: |00000000000000000000000111111111| delays: 27+-04
m6, b7: |00000000000000000000000000000000| delays: -
best: m6, b5 delays: 14+-05
m7, b0: |00000000000000000000000000000000| delays: -
m7, b1: |00000000000000000000000000000000| delays: -
m7, b2: |00000000000000000000000000000000| delays: -
m7, b3: |00000000000000000000000000000000| delays: -
m7, b4: |11111100000000000000000000000000| delays: 03+-03
m7, b5: |00000000011111111110000000000000| delays: 14+-05
m7, b6: |00000000000000000000011111111110| delays: 26+-05
m7, b7: |00000000000000000000000000000000| delays: -
best: m7, b5 delays: 14+-05
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 360Mbps Reads: 384Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/rootfs.cpio to 0xc0800000 (8029184 bytes)...
[LXTERM] Upload complete (41.1KB/s).
[LXTERM] Uploading buildroot/Image to 0xc0000000 (4545524 bytes)...
[LXTERM] Upload complete (41.0KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (9700 bytes)...
[LXTERM] Upload complete (37.0KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0xc1000000 (1441 bytes)...
[LXTERM] Upload complete (43.1KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
KExecuting booted program at 0xc1000000

--============= Liftoff! ===============--

This does not make any progress!
Removing 'emulator.bin, rootfs.cpio, rv32.dtb, Image' files from directory and copying new files makes solution.

I don't know why this happens.

Thanks for your works.

enjoy-digital, 2020-06-23:
While i was able to reproduce the issue when posted, the issue is no longer present with upstream LiteX. Boot has been tested several time over TFTP and Serial. This has probably been fixed with the recent improvements/fixes made to LiteX.

closed 2020-06-23 by enjoy-digital