litex-hub/linux-on-litex-vexriscv #116: On ./make.py --flash Error: couldn't open emulator/emulator.bin.fbi

bradley-evans, 2020-03-03:
I am having trouble flashing the Linux image to the Arty A7 35T.

`./make.py --board=arty --build` seems to run without a hitch. Warnings, but no showstoppers.

Relevant logs:
https://gist.github.com/8feaf8d59dc36c1f23d21a3431a23e71
https://gist.github.com/7c4559f913627f477dffa3dce10e99a0

When I run `./make.py --board=arty --load` I see the board reacting but nothing appears in `screen /dev/ttyUSB2` or `ttyUSB1`. This includes when hitting reset on the board.

Relevant logs from load:
https://gist.github.com/baffced4dc643c21c95df696c991b514
https://gist.github.com/593c216ab5907cf805888fb4aa8cb963

When I run `./make.py --board=arty --flash`, with the device connected, the device reacts (lights, blink, etc) but the flash process ends with `Error: couldn't open emulator/emulator.bin.fbi`.

Logs here are more useful.
Log where error actually occurs: https://gist.github.com/8ba4b365655c3bec2a2c666bf8b16891
https://gist.github.com/884eca3214c297f3def8de26ab8fa472

bradley-evans, 2020-03-03:
Whoops. Solution was adding `--fbi` to command.

enjoy-digital, 2020-03-05:
Thanks for the feedback and test. The Flash instructions were not up to date and is has been updated with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/25d8acafdfa2365382df483f4242788f9b26a90a.

closed 2020-03-05 by enjoy-digital