litex-hub/linux-on-litex-vexriscv #119: Linux doesn't boot for Mimas A7

feliks-montez, 2020-03-18:
I'm attempting to boot Linux on the Mimas A7 board. I've modified `make.py` and `litex` to add support for the Mimas A7 and have managed to get Litex BIOS booting properly. When I load the linux image via
```bash
lxterm --images=images.json /dev/ttyUSB0 --speed=1e6 --no-crc
```
then reset the board, it boots, copies the files over, then fails with `Too many consecutive errors, aborting`. Here's the full output:
```[LXTERM] Starting....

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital

 BIOS built on Mar 17 2020 20:08:38
 BIOS CRC passed (9943e62d)

 Migen git sha1: 0d0e17a
 LiteX git sha1: 862e784e

--=============== SoC ==================--
CPU:       VexRiscv @ 100MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  262144KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11111111111000000000000000000000| delays: 05+-05
m0, b1: |00000000000000111111111111100000| delays: 20+-06
m0, b2: |00000000000000000000000000000001| delays: 31+-00
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 20+-06
m1, b0: |11111111111000000000000000000000| delays: 05+-05
m1, b1: |00000000000001111111111111000000| delays: 19+-06
m1, b2: |00000000000000000000000000000111| delays: 30+-01
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b1 delays: 19+-06
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 213Mbps Reads: 240Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0xc0000000 (4545524 bytes)...
[LXTERM] Upload complete (1834.7KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0xc0800000 (8029184 bytes)...
[LXTERM] Upload complete (1834.1KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0xc1000000 (1707 bytes)...
[LXTERM] Upload complete (1582.6KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (10156 bytes)...
[LXTERM] Upload complete (1780.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
CCCCToo many consecutive errors, abortingNo boot medium found

--============= Console ================--
litex>
```
followed by a bunch of garbley gook.

Any idea what's wrong?

enjoy-digital, 2020-03-18:
Can you do a test without the `--no-crc`, it will be slower but should check that the binaries are copied correctly. You can also try to lower the baudrate as done on the KC705: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L325-L326.

feliks-montez, 2020-03-18:
Tried without `--no-crc` and it hangs when trying to upload the image:
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0xc0000000 (4545524 bytes)...
|>                    | 0%
```
Perhaps the images aren't even getting loaded and `--no-crc` overlooks that.

My `uart_name` is set to `usb_fifo`, as in [this](https://github.com/enjoy-digital/litex/issues/334) issue. My computer apparently received the firmware download request successfully, but could `uart_name` be somehow interfering with the upload?

enjoy-digital, 2020-03-19:
Ah ok, the issue is probably related to the `usb_fifo`, i remember trying to load the linux images with it without success. I could have a look, can you try to use a regular uart to confirm it's the issue?

feliks-montez, 2020-03-19:
Switched over to RS232 and after uploading the files, it started booting, but hung here:
```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0xc0000000 (4545524 bytes)...
[LXTERM] Upload complete (94.7KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0xc0800000 (8029184 bytes)...
[LXTERM] Upload complete (94.6KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0xc1000000 (1707 bytes)...
[LXTERM] Upload complete (95.2KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (10156 bytes)...
[LXTERM] Upload complete (94.2KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
KExecuting booted program at 0x20000000

--============= Liftoff! ===============--
VexRiscv Machine Mode software built Mar 19 2020 11:01:47
--========== Booting Linux =============--
[    0.000000] No DTB passe
```
Looks like the `usb_fifo` is indeed causing a problem, but maybe not the only issue.
For the sake of learning (and for fun) I'd love to help resolve this. What are your suspicions? Could it be an issue with protocol in `usb_fifo.py`?

enjoy-digital, 2020-03-19:
Thanks for the test, there is probably another issue here, but this indeed confirms that uploading with `usb_fifo` does not seem to work correctly. `usb_fifo` is a lot faster than the traditional uart and the issue could be related to that. I don't think the issue is related to the gateware, but more probably to the software running on the CPU: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/boot.c#L114-L233 or the script used for the upload: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_term.py#L198-L231. The first thing would probably be to instrument the script to see what is returned by the CPU software. To instrument the CPU software, it will difficult to use the UART since this will break the transfer, so you could use a GPIO (similar to this for example) and make the sotfware write to it to output debug infos.

feliks-montez, 2020-03-20:
I don't have a logic analyzer, so making sense of lots of GPIO output would be difficult. The BIOS does let me read the RAM though. I could dump the memory at the locations of interest to see how it compares between RS232 mode and `usb_fifo` mode.
According to the VexRiscv project, the "linux balanced" CPU can handle about 180MHz. If it is a timing issue and `usb_fifo` is just too fast, I could also see if increasing the clock speed a bit changes what gets loaded into RAM.
The strange thing is that running with `usb_fifo` mode allows me to both send commands and receive data from the CPU just fine. Maybe that means it does have something to do with the `serialboot` command.
I'll play with some of these ideas and see what I can come up with.

enjoy-digital, 2020-03-20:
Vexriscv is able to run at 180MHz on high end FPGA, on an Artix7 and with the full SoC, i don't think it will pass timing with a `sys_clk_freq` >= 125MHz. To simplify the test, you could reduced the size of the binary (for example just load emulator.bin) and try to see if the CRC correct (https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/boot.c#L155-L168) and if not, try to understand why. I will also try to help on that.

Dolu1990, 2020-03-20:
@feliks-montez It seem that the speed grade of the FPGA on the dev kit is a slow one (XC7A50T-1FGG484C)

So, in such case, you have to expect about 30 % slower speed, as the 180 Mhz reported on the VexRiscv repo are for the fastest speed grades.

I just updated the VexRiscv readme to be clear on the speed grade used.

feliks-montez, 2020-03-22:
After poking around a bit, I've determined that `serialboot` hangs in the payload-receiving loop (seemingly frame.payload[136] is the last byte received). Somehow, `uart_read()` must be hanging.
Here's my annotated breakdown of payload:
[image]
After the 4 address bytes, there are the first correct 119 bytes of `emulator.bin`. I'm not sure what that other stuff is. My guess is that's where something went wrong in transmission.

I'll dig into lxterm's source code next to see if I can find anything wrong on that end.

enjoy-digital, 2020-03-22:
Thanks @feliks-montez, can you share the debug code you used? Otherwise, that would probably be interesting to see if playing with the payload length (https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_term.py#L55) or `usb_fifo` depth (https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/usb_fifo.py#L150) as an effect on this.

feliks-montez, 2020-03-23:
The debug code I used was just placing various `printf` statements throughout `serialboot` to see where it was hanging and what the values of things were. Changing [these lines](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_term.py#L192-L193) in litex to print out any debug info is also helpful.

lxterm looks good, but the default serial library doesn't support FTDI's closed-source protocol, D2XX, which FIFO 245 uses. Not sure how important that is because we can evidently still use `pyserial` to communicate with a D2XX-mode FTDI chip (I don't completely understand why yet). Regardless, I'm having trouble finding a stable and well-supported python library that wraps D2XX. I might experiment with one of the "smaller" ones just to see what happens.

feliks-montez, 2020-03-23:
Got Linux booting! There was some packet misalignment in litex (which surprises me because this problem isn't happening for other boards using UART) and I improved `serialboot`'s handling of bad packets. Let me clean up my code a bit and verify a few things, then I'll create a PR.

feliks-montez, 2020-03-24:
Draft PR is for litex: https://github.com/enjoy-digital/litex/pull/437

enjoy-digital, 2020-03-25:
Thanks @feliks-montez, with your changes in https://github.com/enjoy-digital/litex/pull/437 + https://github.com/enjoy-digital/litex/commit/ffe83ef0f3d6846f0674d8085f49bcaf0f4b130c i'm able to boot Linux on the Minispartan6+ with `usb_fifo`:

````
$lxterm /dev/ttyUSB1 --images=images.json --no-crc

[LXTERM] Starting....
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 25 2020 08:49:26
 BIOS CRC passed (a8b13fed)

 Migen git sha1: 3f9809b
 LiteX git sha1: c154d8d2

--=============== SoC ==================--
CPU:       VexRiscv @ 80MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 164Mbps Reads: 235Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0x40000000 (4545524 bytes)...
[LXTERM] Upload complete (597.5KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0x40800000 (8029184 bytes)...
[LXTERM] Upload complete (594.6KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0x41000000 (1703 bytes)...
[LXTERM] Upload complete (589.5KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x41100000 (9600 bytes)...
[LXTERM] Upload complete (604.7KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
KExecuting booted program at 0x41100000

--============= Liftoff! ===============--
VexRiscv Machine Mode software built Mar 24 2020 12:26:58
--========== Booting Linux =============--
[    0.000000] No DTB passed to the kernel
[    0.000000] Linux version 5.0.13 (florent@lab) (gcc version 8.3.0 (Buildroot 2019.08-rc2-00011-gad9efda578)) #1 Thu Sep 12 14:20:26 CEST 2019
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] Initial ramdisk at: 0x(ptrval) (8388608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x0000000041ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000041ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x0000000041ffffff]
[    0.000000] elf_hwcap is 0x1101
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M@0x40000000 rootwait console=liteuart earlycon=sbi root=/dev/ram0 init=/sbin/init swiotlb=32
````

Can you do the test on the Mimas A7? I think it really slow when not using the `--no-crc` because of the read/write bus turnaround with the `usb_fifo` that is maybe not optimal. I could investigate, but  even if improved, the next bottleneck when loading the images will be the CRC32 computed by the CPU, so we will probably still use `--no-crc` with Linux-on-LiteX-Vexriscv.

feliks-montez, 2020-03-26:
I can verify that Linux boots on the Mimas A7 with these changes _using_ crc. With `--no-crc` I still get garbley-gook. Just to confirm, it works with `--no-crc` on the Minispartan6+? If so, there might be much deeper stuff to untangle here.

enjoy-digital, 2020-03-26:
@feliks-montez: yes it works on the Minispartan6+ with `--no-crc`. Since i still don't explain why this delay is needed: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_term.py#L227, could you try increasing it and see if it's better with the Mimas A7? If so, we will just need to understand why this delay is needed and implement the proper fix.

feliks-montez, 2020-03-27:
I tried with a delay of `1e-5` and it didn't work, but `1e-4` did! Perhaps it means that the FIFO is getting filled up and overwritten because litex tries to write to it too quickly? That's my hunch because when I was analyzing the bad frames, it sure looked like the header for the next frame slipped into the rear of the first frame. But I still wouldn't expect that significant of a timing difference if our CPUs are both 100MHz.

enjoy-digital, 2020-04-09:
The default delay has been reduced to `1e-4`and i created a specific issue in LiteX for this: https://github.com/enjoy-digital/litex/issues/460. I think we can close this since we now have a workaround to load Linux on the Mimas A7 using the usb_fifo and i'll continue the investigation in https://github.com/enjoy-digital/litex/issues/460.

closed 2020-04-09 by enjoy-digital