litex-hub/linux-on-litex-vexriscv #120: Linux does not boot on de10nano

rob-ng15, 2020-03-24:
de10nano no longer boots after emulator was moved to sdram.

For lxterm images.json seems to have a conflicting memory map, having main memory at 0xc000000 not 0x4000000

spisdcardboot seems to load the images correctly, as evidenced by...

`litex> spisdcardboot
SD Card via SPI Initialising
Reading MBR
Partition 1 Information: Active=0x00, Type=0x06, LBAStart=0x00000800
Partition 1 is FAT16

Read FAT16 Boot Sector
  Jump Code:              0xeb 0x3c 0x90
  OEM Code:               [mkfs.fat]
  Sector Size:            512
  Sectors Per Cluster:    128
  Reserved Sectors:       128
  Number of Fats:         2
  Root Dir Entries:       2048
  Total Sectors Short:    0
  Media Descriptor:       0xf8
  Fat Size Sectors:       128
  Sectors Per Track:      32
  Number of Heads:        64
  Hidden Sectors:         2048
  Total Sectors Long:     4194304
  Drive Number:           0x80
  Current Head:           0x01
  Boot Signature:         0x29
  Volume ID:              0x0b69faab
  Volume Label:           [NO NAME    ]
  Volume Label:           [FAT16   ]
  Boot Sector Signature:  0xaa55

Reclocking from 400KHz to 12500KHz

sdCardFatTable = 0x41ff0000  Reading Fat16 Table (128 Sectors Long)

sdCardFat16RootDir = 0x41fe0000  Reading Root Directory (128 Sectors Long)

Root Directory
  File 0 [AI m a g. e ] @ Cluster 0 for -1 bytes
  File 1 [IMAGE   .   ] @ Cluster 207 for 4545524 bytes
  File 2 [Ar o o t. f ] @ Cluster 0 for -65536 bytes
  File 3 [ROOTFS~1.CPI] @ Cluster 277 for 8029184 bytes
  File 4 [Ar v 3 2. . ] @ Cluster 0 for -1 bytes
  File 5 [RV32    .DTB] @ Cluster 400 for 1441 bytes
  File 6 [Ae m u l. a ] @ Cluster 0 for 110 bytes
  File 7 [EMULATOR.BIN] @ Cluster 3 for 9600 bytes
  File 8 [ p a r t.   ] @ Cluster 0 for -1 bytes
  File 9 [ e m u l. a ] @ Cluster 0 for 3014766 bytes
  File 10 [ MULAT~1.PAR] @ Cluster 3 for 9600 bytes
  File 11 [ r o o t. f ] @ Cluster 0 for 7340078 bytes
  File 12 [ OOTFS~1.PAR] @ Cluster 79 for 8265728 bytes

Reading File [IMAGE.] into 0x40000000 : File starts at Cluster 207 length 4545524
Clusters: 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 

Reading File [ROOTFS~1.CPI] into 0x40800000 : File starts at Cluster 277 length 8029184
Clusters: 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 

Reading File [RV32.DTB] into 0x41000000 : File starts at Cluster 400 length 1441
Clusters: 400 

Reading File [EMULATOR.BIN] into 0x41100000 : File starts at Cluster 3 length 9600
Clusters: 3 

litex> mr 0x40000000 64
Memory dump:
0x40000000  73 10 40 10 97 d1 41 00 93 81 41 5b b7 62 00 00  s.@...A...A[.b..
0x40000010  73 b0 02 10 93 86 41 22 13 06 10 00 af a6 c6 00  s.....A"........
0x40000020  63 9e 06 0c 97 f6 41 00 93 86 c6 fd 17 57 45 00  c.....A......WE.
0x40000030  13 07 c7 19 63 d8 e6 00 23 a0 06 00 93 86 46 00  ....c...#.....F.
litex> mr 0x40800000 64
Memory dump:
0x40800000  30 37 30 37 30 31 30 30 30 34 41 46 46 32 30 30  0707010004AFF200
0x40800010  30 30 34 31 45 44 30 30 30 30 30 30 30 30 30 30  0041ED0000000000
0x40800020  30 30 30 30 30 30 30 30 30 30 30 30 31 31 35 44  000000000000115D
0x40800030  38 31 43 33 33 32 30 30 30 30 30 30 30 30 30 30  81C3320000000000
litex> mr 0x41000000 64
Memory dump:
0x41000000  d0 0d fe ed 00 00 05 a1 00 00 00 38 00 00 04 50  ...........8...P
0x41000010  00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00  ...(............
0x41000020  00 00 01 51 00 00 04 18 00 00 00 00 00 00 00 00  ...Q............
0x41000030  00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00  ................
litex> mr 0x41100000 64
Memory dump:
0x41100000  6f 00 00 0b 13 00 00 00 13 00 00 00 13 00 00 00  o...............
0x41100010  13 00 00 00 13 00 00 00 13 00 00 00 13 00 00 00  ................
0x41100020  23 2e 11 fe 23 2c 51 fe 23 2a 61 fe 23 28 71 fe  #...#,Q.#*a.#(q.
0x41100030  23 26 a1 fe 23 24 b1 fe 23 22 c1 fe 23 20 d1 fe  #&..#$..#"..# ..
litex> `

(I'd taken the actual jump to memory out to check that loading was correct)

This results in a hang a Liftoff

rob-ng15, 2020-03-24:
With boot command added, showing hang at Liftoff

`        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 24 2020 13:48:44
 BIOS CRC passed (62b4f59a)

 Migen git sha1: 3f9809b
 LiteX git sha1: c69ef818

--=============== SoC ==================--
CPU:       VexRiscv @ 50MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 110Mbps Reads: 147Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--
litex> spisdcardboot
SD Card via SPI Initialising
Reading MBR
Partition 1 Information: Active=0x00, Type=0x06, LBAStart=0x00000800
Partition 1 is FAT16

Read FAT16 Boot Sector
  Jump Code:              0xeb 0x3c 0x90
  OEM Code:               [mkfs.fat]
  Sector Size:            512
  Sectors Per Cluster:    128
  Reserved Sectors:       128
  Number of Fats:         2
  Root Dir Entries:       2048
  Total Sectors Short:    0
  Media Descriptor:       0xf8
  Fat Size Sectors:       128
  Sectors Per Track:      32
  Number of Heads:        64
  Hidden Sectors:         2048
  Total Sectors Long:     4194304
  Drive Number:           0x80
  Current Head:           0x01
  Boot Signature:         0x29
  Volume ID:              0x0b69faab
  Volume Label:           [NO NAME    ]
  Volume Label:           [FAT16   ]
  Boot Sector Signature:  0xaa55

Reclocking from 400KHz to 12500KHz

sdCardFatTable = 0x41ff0000  Reading Fat16 Table (128 Sectors Long)

sdCardFat16RootDir = 0x41fe0000  Reading Root Directory (128 Sectors Long)

Root Directory
  File 0 [AI m a g. e ] @ Cluster 0 for -1 bytes
  File 1 [IMAGE   .   ] @ Cluster 207 for 4545524 bytes
  File 2 [Ar o o t. f ] @ Cluster 0 for -65536 bytes
  File 3 [ROOTFS~1.CPI] @ Cluster 277 for 8029184 bytes
  File 4 [Ar v 3 2. . ] @ Cluster 0 for -1 bytes
  File 5 [RV32    .DTB] @ Cluster 400 for 1441 bytes
  File 6 [Ae m u l. a ] @ Cluster 0 for 110 bytes
  File 7 [EMULATOR.BIN] @ Cluster 3 for 9600 bytes
  File 8 [ p a r t.   ] @ Cluster 0 for -1 bytes
  File 9 [ e m u l. a ] @ Cluster 0 for 3014766 bytes
  File 10 [ MULAT~1.PAR] @ Cluster 3 for 9600 bytes
  File 11 [ r o o t. f ] @ Cluster 0 for 7340078 bytes
  File 12 [ OOTFS~1.PAR] @ Cluster 79 for 8265728 bytes

Reading File [IMAGE.] into 0x40000000 : File starts at Cluster 207 length 4545524
Clusters: 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 

Reading File [ROOTFS~1.CPI] into 0x40800000 : File starts at Cluster 277 length 8029184
Clusters: 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 

Reading File [RV32.DTB] into 0x41000000 : File starts at Cluster 400 length 1441
Clusters: 400 

Reading File [EMULATOR.BIN] into 0x41100000 : File starts at Cluster 3 length 9600
Clusters: 3 

Executing booted program at 0x41100000

--============= Liftoff! ===============--

`

rob-ng15, 2020-03-24:
And from lxterm...

`[LXTERM] Starting....
�
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 24 2020 13:48:44
 BIOS CRC passed (62b4f59a)

 Migen git sha1: 3f9809b
 LiteX git sha1: c69ef818

--=============== SoC ==================--
CPU:       VexRiscv @ 50MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  32768KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 110Mbps Reads: 147Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading buildroot/Image to 0xc0000000 (4545524 bytes)...
[LXTERM] Upload complete (82.4KB/s).
[LXTERM] Uploading buildroot/rootfs.cpio to 0xc0800000 (8029184 bytes)...
[LXTERM] Upload complete (82.6KB/s).
[LXTERM] Uploading buildroot/rv32.dtb to 0xc1000000 (1703 bytes)...
[LXTERM] Upload complete (82.7KB/s).
[LXTERM] Uploading emulator/emulator.bin to 0x20000000 (9700 bytes)...
[LXTERM] Upload complete (81.7KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
CCCCToo many consecutive errors, abortingNo boot medium found

--============= Console ================--
litex>  �$A)�)�*��*A+g��
Command not found
litex> @��▒o���%�       ��
litex>  �Tl�▒���        ��
litex> ��T�_��  p�o�����
                        �G��    ��o�_��#$�#&#"��$����#$����#�$�#��#$��#$� �#$����#$�� �#$�� �$�#$�� ���#$�� �$��#$�� �$�$�#$�� �$��A�#$�� �$��$�#$�� �$��$A�#$�� �$��$A�#$�� �$��$A�#$�� �$��$Ag�#$�� �$��$A��#$�� �$��$Ag�#$�� �$��$Ag��#$�� �$��$Ag��#$�� �$��$Ag���#$�� �$��$Ag�#�#$�� �$��$Ag��$�#$�� �$��$Ag��#��#$�� �$��$Ag��#$�#$�� �$��$Ag��#$�#�#$�� �$��$Ag��#$�&�#$�� �$��$Ag��#$�#�#$�� �$��$Ag��#$�#&�#$�� �$��$Ag��#$�#&#�#$�� �$��$Ag��#$�#&"�#$�� �$��$Ag��#$�#&#��#$�� �$��$Ag��#$�#&#"�#$�� �$��$Ag��#$�#&#"�#�#$�� �$��$Ag��#$�#&#"� �#$�� �$��$Ag��#$�#&#"�#!�#$�� �$��$Ag��#$�#&#"�# �#$�� �$��$Ag��#$�#&#"�# !�#$�� �$��$Ag��#$�#&#"�# !�#$�� �$��$Ag��#$�#&#"�# !�#$�� �$��$Ag��#$�#&#"�# !�#$�� �$��$Ag��#$�#&#"�# !�#$�� �$��$Ag��#$�#&#"�# !(�#$�� �$��$Ag��#$�#&#"�# !E�#$�� �$��$Ag��#$�#&#"�# !(�#$�� �$��$Ag��#$�#&#"�# !(E��#$�� �$��$Ag��#$�#&#"�# !(E'�#$�� �$��$Ag��#$�#&#"�# !(E���#$�� �$��$Ag��#$�#&#"�# !(E�'c�#$�� �$��$Ag��#$�#&#"�# !(E�'Ȁ�#$�� �$��$Ag��#$�#&#"�# !(E�'�c�#$�� �$��$Ag��#$�#&#"�# !(E�'�c���#$�� �$��$Ag��#$�#&#"�# !(E�'�c���#$�� �$��$Ag��#$�#&#"�# !(E�'�c���#$�� �$��$Ag��#$�#&#"�# !(E�'�c�� �#$�� �$��$Ag��#$�#&#"�# !(E�'�c�瀃��#$�� �$��$Ag��#$�#&#"�# !(E�'�c�瀃 �#$�� �$��$Ag��#$�#&#"�# !(E�'�c�瀃 ��#$�� �$��$Ag��#$�#&#"�# !(E�'�c�瀃 �$�#$�� �$��$Ag��#$�#&#"�# `

With more garbage being emitted

rob-ng15, 2020-03-24:
sim.py also fails

`
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 24 2020 13:56:07
 BIOS CRC passed (e0e71feb)

 Migen git sha1: 3f9809b
 LiteX git sha1: c69ef818

--=============== SoC ==================--
CPU:       VexRiscv @ 1MHz
ROM:       32KB
SRAM:      4KB
MAIN-RAM:  32768KB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x20000000

--============= Liftoff! ===============--
VexRiscv Machine Mode software built Mar 24 2020 13:56:01
--========== Booting Linux =============--
- /home/rob/Development/Quartus/Projects/linux-on-litex-vexriscv/build/sim/gateware/dut.v:1199: Verilog $finish`

rob-ng15, 2020-03-24:
Any advice on how to help with debugging this issue appreciated

enjoy-digital, 2020-03-24:
Can you check that emulator.bin/rv32.dfb have been re-generated correctly and re-written to SDCard?

rob-ng15, 2020-03-24:
There were no error messages during recompiling. I have updated the sdcard, still no boot. I'll try lxterm with the new images.json later.

enjoy-digital, 2020-03-24:
Ok thanks. I'll also look at that a bit later.

enjoy-digital, 2020-03-25:
That's strange, i don't have the issue. Can you test with these binaries? Also have you re-generated the FPGA bitstream?
[de10nano_binaries.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4379919/de10nano_binaries.zip)

rob-ng15, 2020-03-25:
Many thanks.

I had to delete my build/de10nano directory and rebuild from scratch. I guess something did not recompile. Many thanks for your assistance. Closing this.

closed 2020-03-25 by rob-ng15

enjoy-digital, 2020-03-25:
Ok good :)