litex-hub/linux-on-litex-vexriscv #122: Making soc_controller (and all kernel drivers) work on all widths and alignments

gsomlo, 2020-03-24:
Right now, the soc_controller in linux-on-litex-vexriscv is hardcoded for 32bit cpu, 8bit csr-data-width:
https://github.com/gsomlo/linux/blob/litex-devel-2020-03-23-01/include/linux/litex.h#L9

I'd like to eventually see this in proper upstream Linux, but before that we should probably ensure it works with 64bit CPUs and/or 32bit csr-data-width as well (there's at least one working 64bit linux-on-litex project that I know of, http://www.contrib.andrew.cmu.edu/~somlo/BTCP -- shameless self promotion notwithstanding :) -- and it would be nice if everyone ended up being taken care of when things eventually go upstream).

To that end, I'd like to work on making the soc_controller and the rest of the Linux kernel drivers (well, at *least* liteeth and spi, but really all of them) work independently of whether the CPU is 32-vs-64 bit, or whether csr-data-width is 8-vs-32 bit (just the way LiteX bios itself works ATM).

I've assembled all patches into a branch against the current upstream tree here:
https://github.com/gsomlo/linux/tree/litex-devel-2020-03-23-01
(it feels much easier to work against something like that than against a dump of disparate patches -- for one thing, I can rebase it against newer upstream masters very easily as time goes by, and switch to a new branch if I have to change anything).

The very first thing I'd like your help deciding (@mateusz-holenko @enjoy-digital, anyone else?) is whether to allow for the csr-data-width to be dynamically read in via the litex_soc DT node, or should it rather be a Kconfig compile-time parameter?

- With the DT option, the same kernel could work on LiteX systems of different csr-data-width (but still only of the same cpu data width), at the expense of checking a variable at run-time when accessing CSR subregisters.

- With the hardcoded Kconfig option, we can't move a kernel across LiteX systems of different csr-data-width, but compile-time constants will likely allow for more optimized accessors.

Any thoughts, suggestions, opinions welcome!

Thanks much,
--Gabriel

mateusz-holenko, 2020-03-25:
Would it be possible to have both? Kconfig option could allow to set a fixed value (like 8 or 32) OR a special value meaning DYNAMIC (that would require information at runtime in DT).

This would probably require some ifdefing and additional checks (what if the kernel is compiled with a hardcoded csr-data-width, but DT sets a different value?), but would allow everyone to decide what's more important - portability or performance.

gsomlo, 2020-03-25:
The idea of a hard-coded kernel that *can't* honor an advertised DT
setting scares the living daylights out of me (and, I worry, would
get us scorn from the upstream kernel devs if we tried to upstream it) :)

The more I think about it, maybe starting with hard-coded Kconfig
values might be the path of least resistance -- and allows the option
to (fully) switch to DT-configured dynamic behavior, if needed, in the
future, when LiteX becomes widespread and popular enough that geenric
kernels would need to be deployed across diverse builds. :)

I'll start working on that assumption, but it should be easy to change
course based on further discussions here...

mithro, 2020-03-25:
@gsomlo It's quite common to have stuff inside the devicetree that your kernel might not support (don't have the driver, trying to run an i686 binary on i586 device, etc). A kernel which can't match the given device tree should just kernel panic.

I suggest we start with statically configured values. Then in the future we can add support for dynamic configuration. Doing the dynamic configuration without adversely impacting performance is going to require things like start time patching (similar to how SSE version is detected and patched).

gsomlo, 2020-03-31:
I've grabbed all the Linux patches available in https://github.com/litex-hub/linux-on-litex-vexriscv/tree/master/buildroot/board/litex_vexriscv/patches/linux and stuck them in a branch of my upstream Linus-tree fork: https://github.com/gsomlo/linux/tree/litex-devel-2020-03-23-01 (you can click on "compare" to see the 16 current commits).

I then added a bunch of commits to make the LiteX soc driver 64-bit CPU and 32-bit csr-data-width compatible, here: https://github.com/gsomlo/linux/tree/litex-devel-2020-03-31-01 (it's *19* commits ahead of master, i.e. three additional commits on top of the original 16).

@mateusz-holenko @enjoy-digital can you please try these out and verify that they still work exactly the same on vexriscv? (they should).

This brings me to my next thought: I'd like to keep rebasing and reformatting these commits, particularly as I'll work on switching the liteeth driver over to using litex_soc methods. I'll publish those further changes in a new branch (still against upstream master). I'm thinking this may be an easier way to do kernel development for LiteX specifically as compared to publishing a "dump-of-patches" like the current state of https://github.com/litex-hub/linux-on-litex-vexriscv/tree/master/buildroot/board/litex_vexriscv/patches/linux -- any thoughts?

Thanks, as usual,
--G

gsomlo, 2020-04-19:
@mateusz-holenko @fkokosinski @enjoy-digital I now have working soc-controller, liteeth, and litespi drivers on (64-bit) Linux: https://github.com/gsomlo/linux/tree/litex-devel-next (check out the [video](http://www.contrib.andrew.cmu.edu/~somlo/BTCP/litex-spi-mmc.mp4) of sdcard over mmc_spi).

I'm putting the final touches on the updated spi-litespi.c kernel driver, and have a few questions about the original design:
  - at https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/board/litex_vexriscv/patches/linux/0007-drivers-spi-add-LiteSPI-driver.patch#L117, what possible values can we have for `t->bits_per_word`, assuming we issued sane values for `litespi,max-bpw` in DTS (8, 16, or 32)? Why would there ever be `bits_per_word` values that are *NOT* multiples of 8?
  - at https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/board/litex_vexriscv/patches/linux/0007-drivers-spi-add-LiteSPI-driver.patch#L121 -- is using memcpy endian-safe? We know that writing LiteX subregisters happens in cpu-native endianness, but wouldn't `memcpy()` place things at the *wrong* end of a larger int type on the wrong-endian CPU? @florent, in a `>=1` bpw scenario, which byte gets transmitted (or received) first? LSB or MSB? If I knew that, I could make sure multi-bpw cases are handled correctly :)
  - finally, what's with the `bus_mutex` at https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/board/litex_vexriscv/patches/linux/0007-drivers-spi-add-LiteSPI-driver.patch#L96? What sort of race condition are we protecting against? And, if we absolutely have to use mutual exclusion, couldn't (shouldn't) we use `spi_bus_lock(master)` and `spi_bus_unlock(master)` instead of rolling our own?

Please advise, and I'll update my patch set accordingly :)

Also, @florent, would it make sense to add a https://github.com/litex-hub/linux clone of https://github.com/torvalds/linux with a few branches carrying these patches for both 32 and 64 bit systems moving forward?

Thanks, as always,
--G

enjoy-digital, 2020-04-19:
@gsomlo: i agree we should change the way manage the Linux changes. I initially used patches on Linux-on-LiteX-Vexriscv to keep things simple (and since only a few peripherals were supported), but we should indeed move to a proper fork with branches. @gsomlo: could you push your changes to a branch on https://github.com/litex-hub/linux? @mateusz-holenko: could you do the same for the Linux-on-LiteX-Vexriscv changes? (on a different branch) I could then adapt Linux-on-LiteX-Vexriscv to use the branch instead of the patches and in the future we could see if we still need two separate 32/64-bit branches or if we could keep a single one.

For the SPI driver/gateware, i think we can consider we will only transfer multiple of 8-bits and the bits are always serialized MSB first. (if data_width=32, bit-31 is transfered first).

mateusz-holenko, 2020-04-21:
@enjoy-digital Sure, I'll push the LKML patchsets on a branch to litex-hub/linux.

@gsomlo I'm gathering responses, I will get back to you soon.

gsomlo, 2020-04-21:
@mateusz-holenko (and @enjoy-digital) -- please check out https://github.com/litex-hub/linux/tree/litex-vexriscv-rebase -- it contains all patches currently in https://github.com/litex-hub/linux-on-litex-vexriscv/tree/master/buildroot/board/litex_vexriscv/patches/linux as of a few days ago.

Hopefully this should save you time :) I noticed some more recent changes (there were only 16 patches there as of a few days ago). Maybe rebasing and force-pushing litex-hub/linux/tree/litex-vexriscv-rebase is less work than doring the whole thing from scratch ?

I've also pushed a 64-bit branch: https://github.com/litex-hub/linux/tree/litex-rocket-rebase containing the 64-bit fixes for the soc, liteeth, and litespi (the latter is still a work in progress :)

The idea is to hopefully decide that the 64-bit updates are still working OK on 32-bit vexriscv, and to maybe start a new, common branch of things that we're going to eventually send upstream (perhaps after collapsing (squashing) all the different commits touching individual drivers, to make it easier to review by upstream).

Please take a look, and let me know what you think.

mateusz-holenko, 2020-04-22:
@gsomlo 

> I'm putting the final touches on the updated spi-litespi.c kernel driver, and have a few questions about the original design:
>
> * at [...]

We based our implementation on other SPI drivers (e.g., https://github.com/torvalds/linux/blob/master/drivers/spi/spi.c#L3404) and they were written to support any value in range 1-32. 

I also looks that litespi is **not** limited by the desing to sending just multiples of 8-bytes. If it's of any use is a completely different story.

@enjoy-digital said that we can assume only multiples of 8-bytes so I guess the code could be simplified.

> * at https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/board/litex_vexriscv/patches/linux/0007-drivers-spi-add-LiteSPI-driver.patch#L121 -- is using memcpy endian-safe? We [...]

It indeed looks like a bug.

> * finally, what's with the `bus_mutex` at https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/buildroot/board/litex_vexriscv/patches/linux/0007-drivers-spi-add-LiteSPI-driver.patch#L96? [...]

Again, we were inspired by other drivers that use mutexes directly, but I totally agree that `spi_bus_lock/unlock()` looks fine and could be used instead.

We wanted to protect against the situation when `litespi_setup` changes `BPW` during an ongoing transfer.

gsomlo, 2020-04-22:
@mateusz-holenko -- Thanks much for the helpful comments! I think I understand `spi-litespi.c` much better now (except for the bus locking thing, I tried using `spi_bus_[lock|unlock]()` instead of the mutex, and apparently I need to try harder, but that's not really an immediate concern right now :)

I've pushed my current version to https://github.com/litex-hub/linux/blob/gls-litespi-devel/drivers/spi/spi-litespi.c

Turns out, the spi harness already insists that transfers be multiples of power-of-two length (bits-per-word) transfers: https://github.com/torvalds/linux/blob/master/drivers/spi/spi.c#L3466

The current version of my driver (in the gls-litespi-devel branch) works fine on 8-bit data-width SPIMaster gateware. Before @enjoy-digital added an "aligned" mode, the gateware in its default "raw" mode couldn't use the same code-path for 8-bit transfers if it had been built for wider (16 or 32 bit max-bpw)! IOW, instead of:

```
  ...
  if (t->bits_per_word <= 8) {
    const u8 *tx = t->tx_buf;
    u8 *rx = t->rx_buf;

    /* word size is 1 byte */
    for (i = 0; i < t->len; i++) {
      if (tx)
	litex_reg_writeb(hw->base + LITESPI_OFF_MOSI, *tx++);

      litespi_ctrl_start(hw);

      if (rx)
	*rx++ = litex_reg_readb(hw->base + LITESPI_OFF_MISO);
    }
  } else if (t->bits_per_word <= 16) { ...
```
I would have had to do:
```
  ...
  if (t->bits_per_word <= 8) {
    const u8 *tx = t->tx_buf;
    u8 *rx = t->rx_buf;

    /* word size is 1 byte */
    for (i = 0; i < t->len; i++) {
      if (tx)
	litex_reg_writew(hw->base + LITESPI_OFF_MOSI, (*tx++) << 8);

      litespi_ctrl_start(hw);

      if (rx)
	*rx++ = litex_reg_readw(hw->base + LITESPI_OFF_MISO) >> 8;
    }
  } else if (t->bits_per_word <= 16) { ...
```
to support 8-bit transfers on gateware compiled for 16-bit data width.

With the new "aligned" mode added by commit https://github.com/enjoy-digital/litex/commit/0b3c4b50fa99cd92ba83ea64a1386b94195a35d4 the 8-bit mode works the same on all gateware max-data-width options.

My current problem in testing whether the updated driver works in wider `bits_per_word` modes is that `mmc_spi.c` hardcodes transfer length to 8 bpw regardless of the underlying hardware capabilities: https://github.com/torvalds/linux/blob/master/drivers/mmc/host/mmc_spi.c#L1305
and I don't have any other hardware that would use wider transfers to test. Forcing `mmc_spi` to use 16 bits also "needs more investigation" at this point (i.e., doesn't "just work" out of the box :)

Anyhow, I wanted to have all this information out there, so maybe @mateusz-holenko and others can review the driver, chime in on how we can test "aligned" mode gateware on `> 8bit` bpw, etc.

gsomlo, 2020-04-22:
@mateusz-holenko check out https://github.com/litex-hub/linux/commit/871e9006234f6b2166abb435e20ffcbd2562d435 -- I think I figured out how to get rid of locking altogether (by setting bpw on the gateware on a *per_transfer* basis, so that `setup()` doesn't get to stomp all over a transfer that's in progress at the time.

enjoy-digital, 2021-01-08:
@gsomlo: I think this is now covered by your recent changes. Is there still some work to do on this? If so could you give an update? Otherwise we could probably close this issue.

gsomlo, 2021-01-08:
There's a proposed patch series for upstream linux currently awaiting review that would handle 32-bit CSR data width (in addition to the currently supported 8-bit width): https://lore.kernel.org/lkml/20201227161320.2194830-1-gsomlo@gmail.com
@mateusz-holenko if you have a chance to review or ack that at some point, as the official litex subsystem maintainer, that'd be much appreciated!

The series (along with fixed-up drivers) is also available in https://github.com/litex-hub/linux/tree/litex-rebase (and I tested ethernet and [spi-]sdcard on both 8- and 32-bit CSR subregisters, using rocket -- but it "SHOULD" work with vexriscv as well).

Finally, we've standardized on 32-bit alignment everywhere, so that's no longer a variable AFAICT.

I think we can close this, from where I'm looking at it.

enjoy-digital, 2021-01-14:
Closing then, thanks @gsomlo.

closed 2021-01-14 by enjoy-digital