litex-hub/linux-on-litex-vexriscv #123: Test SDCard boot with OrangeCrab

enjoy-digital, 2020-03-25:
I just added SDCard SPI-Mode support to the OrangeCrab with: https://github.com/litex-hub/litex-boards/commit/6e6b6dac5524e2b9ebef0d3d1be316674a7bd012 and 
https://github.com/litex-hub/linux-on-litex-vexriscv/commit/dc61f0acf95506b08e8486e33590ff9ddf8b3095 but don't have the harware to test it.

@pdp7, @gregdavill interested in testing? :)

If so, you can use the attached bitstream and copy the attached images to a FAT16 formated SDCard. 

Then just try to boot with `spisdcardboot` command. (that's possible you'll have to type it twice since the SDCard does not always seem to be initialized correctly the first time).

[orangecrab_linux_images.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4380013/orangecrab_linux_images.zip)
[orangecrab_bitstream_with_spisdcard.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4380016/orangecrab_bitstream_with_spisdcard.zip)

pdp7, 2020-03-25:
@gregdavill is it still possible to drop to the LiteX BIOS with the current USB CDC setup?

enjoy-digital, 2020-03-25:
@pdp7: sorry i miss the fact the OrangeCrab was using USB CDC.

With: https://github.com/enjoy-digital/litex/commit/b509df8bb6ecb7d9fb81592c6f12f251ecb24afc, https://github.com/litex-hub/litex-boards/commit/4053c02d7ef3b2db6607af8c5a23e68423acda3b and https://github.com/litex-hub/linux-on-litex-vexriscv/commit/2cf330dc38442c53f39c44105e6b02e35d4dc522 it should now be possible.

Here is the bitstream if you want to test: [orangecrab_usb_cdc_spisdcard.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4382870/orangecrab_usb_cdc_spisdcard.zip) :)

gregdavill, 2020-03-25:
Nice I'll check this out soon.

On ECP5 I've previously used one of the open cores SD gateware: https://github.com/gregdavill/bosonFrameGrabber/tree/master/fw/bosonECP5/source/sdc_dma/verilog

SPI is a good first step, I've seen there is some recent work on implementing a PHY layer for the ECP5 in liteSDCard

pdp7, 2020-03-25:
I put the files on an SD Card but maybe my problem is `W95 FAT32 (LBA)`?
```
pdp7@x1:~$ sudo fdisk /dev/sdb
[sudo] password for pdp7: 
Sorry, try again.
[sudo] password for pdp7: 

Welcome to fdisk (util-linux 2.34).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/sdb: 14.84 GiB, 15917383680 bytes, 31088640 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x150bc457

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdb1        2048 31088639 31086592 14.8G  c W95 FAT32 (LBA)
```
The files:
```
pdp7@x1:~$ ls -ltar /media/pdp7/26DD-66E2/
total 2100
drwxr-xr-x  2 pdp7 pdp7    8192 Jan  1  1970 .
-rw-r--r--  1 pdp7 pdp7 1442857 Mar 25 19:38 top.svf
-rw-r--r--  1 pdp7 pdp7  682044 Mar 25 19:38 top.bit
drwxr-x---+ 3 root root    4096 Mar 26 00:31 ..
```

pdp7, 2020-03-26:
@gregdavill  is there a way for me to stop the crab from booting vexriscv and keep at the litex bios prompt?

when I open serial port /dev/ttyUSB0, it immediately boots

pdp7, 2020-03-26:
@gregdavill ah, I just needed to be fast and hit ESC right away
[image]

pdp7, 2020-03-26:
@enjoy-digital I don't see `spisdcardboot` as an option?
```
litex> help
LiteX BIOS, available commands:
mr         - read address space
mw         - write address space
mc         - copy address space
fe         - erase whole flash
fw         - write to flash

crc        - compute CRC32 of a part of the address space
ident      - display identifier

reboot     - reset processor
serialboot - boot via SFL
flashboot  - boot from flash

memtest    - run a memory test
```

pdp7, 2020-03-26:
Ah ok, I was misunderstanding.

I have run:
```
pdp7@x1:~/dev/enjoy/linux-on-litex-hadbadge/orangecrab$ dfu-util -d 1209:5bf0 -D ~/Downloads/top.bit 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1209:5bf0
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 4096
Copying data from PC to DFU device
Download	[=========================] 100%       682044 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
```

pdp7, 2020-03-26:
I did it manual with fdisk and mkfs.vfat
```
Disk /dev/sdb: 14.84 GiB, 15917383680 bytes, 31088640 sectors
Disk model: MassStorageClass
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x150bc457

Device     Boot Start    End Sectors  Size Id Type
/dev/sdb1        2048 206847  204800  100M  6 FAT16
```

pdp7, 2020-03-26:
[image]

enjoy-digital, 2020-03-26:
@pdp7: thanks for the test, that's already good the USB CDC is working. For the SDCard issue, i suspect the reclocking. Can you do a test with the attached bitstream with reclocking disabled? It will be slower since the SDCard is only running at 400KHz but it will allow us to know if that's the issue.

[orangecrab_usb_cdc_spisdcard_no_reclocking.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4385459/orangecrab_usb_cdc_spisdcard_no_reclocking.zip)

rob-ng15, 2020-03-26:
Just a thought about reclocking. I aimed for 16MHZ, although on the
DE10nano this, as I was being conservative, actually goes to 12.5MHZ. I
also have a class 10 card.

Is it worth aiming for a slower speed? Is it worth actually checking the
speed class of the card, if that is possible, and acting accordingly?

Rob.

enjoy-digital, 2020-03-26:
@rob-ng15: i would say we could wait the result at 400KHz to see if that's really the issue here. If this is the issue and checking the class speed is easy to do, that's something we should probably indeed add.

rob-ng15, 2020-03-26:
What is the processor speed of the orange crab? It may be that the minimum
divider needs increasing, that may be a good step, say from 2 to 4.

I only have class 10 cards, so I picked a default that works for that. My
cards are all SanDisk too, my cards all work at 25mhz, but I went for less
than 20mhz as a safe option.

I'd suggest if we have to go slower aiming for 8mhz or less.

Rob.

pdp7, 2020-03-26:
> [orangecrab_usb_cdc_spisdcard_no_reclocking.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4385459/orangecrab_usb_cdc_spisdcard_no_reclocking.zip)
I tried this but no output after Liftoff:
```
5 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 

Reading File [RV32.DTB] into 0x41000000 : File starts at Cluster 6149 length 1707
Clusters: 6149 

Reading File [EMULATOR.BIN] into 0x41100000 : File starts at Cluster 3 length 9600
Clusters: 3 0 111 69 449 

Executing booted program at 0x41100000

--============= Liftoff! ===============--
```
@gregdavill @enjoy-digital is there anyway I can debug that further?

rob-ng15, 2020-03-26:
That means that the file allocation table didn't read correctly, or was
corrupted.

rob-ng15, 2020-03-26:
I had this issue when developing the code. I ended up repartitioning the
card and rewriting the files to it.

To debug, remove the boot(...) from boot.c and use the mr command to see
what data has been read into memory.

Rob.

rob-ng15, 2020-03-26:
I'll try reducing the reclock to 8MHZ or at least divide 4 on the reclock.

I think I have found documentation to read the card speed. What class/make of card do you have?

pdp7, 2020-03-26:
@enjoy-digital is if there is are way for me to rebuild bitsream for testing different clock speeds with the SD card?

@rob-ng15 it is a generic 16GB MicroSD from a computer store (MicroCenter). it does not list the class.  I could try another brand if you think that might help.

pdp7, 2020-03-26:
I just tried a SanDisk Class 10 card.

I should note that this is what happens at first:
```
litex> spisdcardboot
SD Card via SPI Initialising
SD Card Timeout
```
then if I run the command again:
```
SD Card via SPI Initialising
Reading MBR
Partition 1 Information: Active=0x00, Type=0x06, LBAStart=0x00000800
Partition 1 is FAT16

Read FAT16 Boot Sector
  Jump Code:              0xeb 0x3c 0x90
  OEM Code:               [mkfs.fat]
  Sector Size:            512
  Sectors Per Cluster:    4
  Reserved Sectors:       4
  Number of Fats:         2
  Root Dir Entries:       512
  Total Sectors Short:    0
  Media Descriptor:       0xf8
  Fat Size Sectors:       200
  Sectors Per Track:      32
  Number of Heads:        64
  Hidden Sectors:         2048
  Total Sectors Long:     204800
  Drive Number:           0x80
  Current Head:           0x01
  Boot Signature:         0x29
  Volume ID:              0xcac4def6
  Volume Label:           [NO NAME    ]
  Volume Label:           [FAT16   ]
  Boot Sector Signature:  0xaa55

sdCardFat16RootDir = 0x47fe3000  Reading Root Directory (32 Sectors Long)

Root Directory
  File 0 [Ae m u l. a ] @ Cluster 0 for 110 bytes
  File 1 [EMULATOR.BIN] @ Cluster 3 for 9600 bytes
  File 2 [AI m a g. e ] @ Cluster 0 for -1 bytes
  File 3 [IMAGE   .   ] @ Cluster 8 for 4545524 bytes
  File 4 [Ar v 3 2. . ] @ Cluster 0 for -1 bytes
  File 5 [RV32    .DTB] @ Cluster 2228 for 1707 bytes

Reading File [IMAGE.] into 0x40000000 : File starts at Cluster 8 length 4545524
Clusters: 8 19 65105 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535
```

pdp7, 2020-03-27:
Here are the results with new SD card (SanDisk class 10) and bitstream from @enjoy-digital 
orangecrab_usb_cdc_spisdcard_no_reclocking.zip
```
sdcardboot

SD Card via SPI Initialising
Reading MBR
Partition 1 Information: Active=0x00, Type=0x06, LBAStart=0x00000800
Partition 1 is FAT16

Read FAT16 Boot Sector
  Jump Code:              0xeb 0x3c 0x90
  OEM Code:               [mkfs.fat]
  Sector Size:            512
  Sectors Per Cluster:    4
  Reserved Sectors:       4
  Number of Fats:         2
  Root Dir Entries:       512
  Total Sectors Short:    0
  Media Descriptor:       0xf8
  Fat Size Sectors:       200
  Sectors Per Track:      32
  Number of Heads:        64
  Hidden Sectors:         2048
  Total Sectors Long:     204800
  Drive Number:           0x80
  Current Head:           0x01
  Boot Signature:         0x29
  Volume ID:              0xcac4def6
  Volume Label:           [NO NAME    ]
  Volume Label:           [FAT16   ]
  Boot Sector Signature:  0xaa55

sdCardFat16RootDir = 0x47fe3000  Reading Root Directory (32 Sectors Long)

Root Directory

  File 0 [Ae m u l. a ] @ Cluster 0 for 110 bytes
  File 1 [EMULATOR.BIN] @ Cluster 4455 for 9600 bytes
  File 2 [AI m a g. e ] @ Cluster 0 for -1 bytes
  File 3 [IMAGE   .   ] @ Cluster 4460 for 4545524 bytes
  File 4 [Ar v 3 2. . ] @ Cluster 0 for -1 bytes
  File 5 [RV32    .DTB] @ Cluster 4454 for 1707 bytes
  File 6 [Ar o o t. f ] @ Cluster 0 for -65536 bytes
  File 7 [ROOTFS~1.CPI] @ Cluster 6680 for 8029184 bytes

Reading File [IMAGE.] into 0x40000000 : File starts at Cluster 4460 length 4545524

Clusters: 4460 11299 12469 0 111 69 449 58591 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 [SNIP]
65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 

Reading File [RV32.DTB] into 0x41000000 : File starts at Cluster 4454 length 1707
Clusters: 4454 
Reading File [EMULATOR.BIN] into 0x41100000 : File starts at Cluster 4455 length 9600

Clusters: 4455 23 65137 65535 65535 
Executing booted program at 0x41100000
--============= [1mLiftoff![0m ===============--
```

rob-ng15, 2020-03-27:
Just been reading a little bit more. The code I wrote will only work on
sdhc cards, not sdxc, etc due to the block sizes.

From what I can see, all cards should work at 8mhz.

It does look like the file allocation table is not being read/stored
properly, looking at the list of clusters being read; there should never be
a repeat.

Data is being read correctly from the card at some point, the MBR and root
directory, otherwise there'd be an error earlier in the process.

pdp7, 2020-03-27:
@rob-ng15 all the cards I have are 16GB microSD SDHC class 10 (sandisk and microcenter brands).  Should that be ok?

I could try to buy older 8GB class 4 card.

@enjoy-digital @gregdavill have you tried any SD cards?

pdp7, 2020-03-28:
@gsomlo thatnks for the advice in irc.  what hardware are you testing with?

pdp7, 2020-03-28:
@enjoy-digital @gregdavill is there a branch I can use to built the bitstream myself?

enjoy-digital, 2020-03-28:
@pdp7: you should be able to build it with upstream repos. ValentyUSB is automatically cloned when building the OrangeCrab bitstream: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L221.

pdp7, 2020-03-28:
@enjoy-digital thanks I'll try it out

gsomlo, 2020-03-29:
On Sat, Mar 28, 2020 at 10:24:49AM -0700, Drew Fustini wrote:
> @gsomlo thatnks for the advice in irc. what hardware are you testing with?

nexys4ddr and trellisboard

enjoy-digital, 2020-06-20:
The code has been refactored (to use FatFs and allow FAT16/FAT32 partitions), performance has been improved (>1MB/s transfers) and has been tested recently on the OrangeCrab: https://twitter.com/enjoy_digital/status/1271128930385362947

A new wiki page that explains the different boot options is also available at: https://github.com/enjoy-digital/litex/wiki/Load-Application-Code-To-CPU

closed 2020-06-20 by enjoy-digital