litex-hub/linux-on-litex-vexriscv #124: emulator:

frantony, 2020-04-03:
```
--- a/emulator/main.c
+++ b/emulator/main.c
@@ -39,7 +39,7 @@ static void litex_putchar(char c){
     full = 1;
     while ((full & 0x1) != 0)
         full = uart_txfull_read();
-    if (c != '\r')
+//    if (c != '\r')
         uart_rxtx_write(c);
 }

```

enjoy-digital, 2020-12-16:
@frantony sorry for the delay, I don't remember the exact  reason for this, but this is not present in OpenSBI port (https://github.com/litex-hub/opensbi/blob/0.8-linux-on-litex-vexriscv/platform/litex/vexriscv/litex.c#L55-L59) that will be used with #165.

enjoy-digital, 2020-12-16:
https://github.com/litex-hub/linux-on-litex-vexriscv/pull/165 has been merged, we can now close this.

closed 2020-12-16 by enjoy-digital