litex-hub/linux-on-litex-vexriscv #126: Memory system to handle multicore with good performances

Dolu1990, 2020-04-12:
Hi,

The integration of a multicore system should normaly come soon. So to anticipate things a bit there is a few questions : 

L2 :
- What is the memory bus between L2 and the DDR controller, and the masters ?
- Does L2 block all requests while refilling (blocking L2) ?
- Does L2 support multiple line refill at the same time ?

Else likely we would either need a new L2 design, or just not having any, as this would reduce performance (which is fine too, as likely, because of the small amount of RAM in FPGA, with a multicore system, the amount of ram left would not allow a big enough L2 to realy make a difference). 

There is also a few other things to think about but let's not mix everything at once XD

enjoy-digital, 2020-04-12:
Hi @Dolu1990,

the current L2 cache is blocking during refill and not supporting multiple line refill :( It's using a wishbone interface with the DDRAM controller with configurable width allow bursting on the DDRAM.

But on the DDRAM side, we could use the native interface (simplified AXI with 3 channels: cmd / write data / read data) or AXI and rework the L2 to improve or just use the caches from VexRiscv and provide an efficient interface between the caches and DDRAM (the native interface should be very similar to your BMB and we could probably bridge between BMB and the DRAM very easily while keeping the burst support), so maybe you could just expose your BMB bus?

Dolu1990, 2020-04-12:
> the current L2 cache is blocking during refill and not supporting multiple line refill :(

No worries, To me it totaly make sense for quite a large spectrum of applications

> simplified AXI with 3 channels: cmd / write data / read data
> the native interface should be very similar to your BMB

Is the ordering between read and write preserved ? (for a given ID)
Is the DDR controller do some cross ID reordering ?

(the memory ordering can be problematic)

> just use the caches from VexRiscv and provide an efficient interface between the caches and DDRAM

If we don't put L2, then i will at least add some sort of write aggregators, to compact sequancial CPU writes to a single burst. (CPU cache is write-through)
Unless the DDR controller implement any form of access aggregation ?

Thanks ^^

Dolu1990, 2020-04-12:
Another one which is important to know is : 

Does the DDR controller (when no bank conflict) support 1 command per cycle if the access is of the size of the DDR burst length ? (ex 16 bits DDR3 => 128 bits bursts)

enjoy-digital, 2020-04-12:
The read/write ordered is preserved yes and the controller is not currently doing reordering. The native width for a quarter rate controller and 16-bits DDR3 is indeed 128 bits and it will accept one command/cycle when possible. One interesting things is that we can simulate the CPU + DRAM controller with the LiteX simulator, so this should simplify this work a lot.

Dolu1990, 2020-04-12:
> ordered is preserved  /  it will accept one command/cycle when possible

Cool, should work well then.

>  we can simulate the CPU + DRAM controller with the LiteX simulator,

Nice to have, will have to squeze out all little juice of performance here ^^

enjoy-digital, 2020-05-04:
A first version is experimented here https://github.com/enjoy-digital/litex_vexriscv_smp with the VexRiscv SMP Cluster converting the internal BMB interfaces to LiteDRAMNative ports. This version is working in simulation (SpinalSim, LiteXSim) and on hardware and will allow identifying the different bottlenecks.

closed 2020-08-13 by Dolu1990