litex-hub/linux-on-litex-vexriscv #128: Add support for new board

bitglicher, 2020-04-23:
Hello, I want to add support for a Cyclone IV board.
The board features :
- Ethernet 
- HDMI
- SD card slot
- DAC
- 1GB DDR2 memory

 I think that this board is a great candidate to run linux on. The issue is that I haven't found any documentation on how to add a new board.

Thanks in advance.

enjoy-digital, 2020-04-23:
Hello @bitglicher,

you will first need to add board support to your board in litex-boards: 
https://github.com/litex-hub/litex-boards

And then integrate it on the Linux-on-LiteX-Vexriscv `make.py.`

Here is for example a board support that has been recently added:
- https://github.com/litex-hub/litex-boards/commit/efb13bc1189d1af39b8bd8e35d8f70f1e2b919d1
- https://github.com/litex-hub/litex-boards/commit/6fe4c4ea6229ab0bb4aaffd2f779d3172dee0dae
- https://github.com/litex-hub/litex-boards/commit/865b01ec751b8b8e1acb5833dae99f779d9233ac

And the Linux-on-LiteX-Vexriscv support:
- https://github.com/litex-hub/linux-on-litex-vexriscv/commit/3b5cffb4404e8dff3c5416c3af32e6268f1eede9

bitglicher, 2020-04-23:
Thanks for the help.

Any recommendations for what PHY interface to use for the DDR2 memory?

Im trying to interface with a SOODIM module, but in the schematics of my fpga board, I don't see a dqs_n pin, I just have the sqs_p.

I tried to just declare the dqs_p signal, but the make.py file gives me the following errors:
`
Traceback (most recent call last):
  File "./make.py", line 475, in <module>
    main()
  File "./make.py", line 410, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/user/Documents/linux-on-litex-vexriscv/soc_linux.py", line 268, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/user/Documents/linux-on-litex-vexriscv/soc_linux.py", line 94, in __init__
    soc_cls.__init__(self,
  File "/home/user/Documents/litex-boards/litex_boards/targets/a_e115fb.py", line 59, in __init__
    self.submodules.ddrphy = s7ddrphy.A7DDRPHY(platform.request("ddram"),
  File "/home/user/Documents/litedram/litedram/phy/s7ddrphy.py", line 604, in __init__
    S7DDRPHY.__init__(self, pads, with_odelay=False, **kwargs)
  File "/home/user/Documents/litedram/litedram/phy/s7ddrphy.py", line 404, in __init__
    io_IOB = pads.dqs_n[i],
  File "/home/user/Documents/litedram/litedram/common.py", line 81, in __getattr__
    for i in range(len(getattr(self.groups[0], name)))
AttributeError: 'Record' object has no attribute 'dqs_n'
`

enjoy-digital, 2020-04-24:
Unfortunately, LiteDRAM does not currently have a Cyclone IV DDR PHY available. If you want to use the DDR2 of your board, you will probably have to 1) use the Altera Wizard to create the DDR2 controller and 2) integrate it in LiteX or create a LiteDRAM PHY for Cyclone IV. 1) will probably be easier than 2) and could be used as a basis to create a LiteDRAM PHY in the future.

You could start by creating a more simple LiteX SoC with CPU/SRAM/ROM/Ethernet and add DDR2 when the simple SoC is already working.

bitglicher, 2020-04-25:
Seems that im out of luck, I can't use the DDR2 on the board, Qsys doesn't support DDR2 for Cyclone IV E.

closed 2020-04-27 by bitglicher