litex-hub/linux-on-litex-vexriscv #129: Switch to OpenSBI

enjoy-digital, 2020-05-04:
To prepare for SMP, we should replace the emulator with OpenSBI: https://github.com/riscv/opensbi

Changes used for https://github.com/enjoy-digital/litex_vexriscv_smp are here: https://github.com/SpinalHDL/opensbi/tree/litex/platform/litex/vexriscv.

enjoy-digital, 2020-05-04:
Since OpenSBI will live in DRAM, we also need to make sure it's defined in a reserved memory region: https://github.com/enjoy-digital/litex_vexriscv_smp/blob/master/images/dts#L82-L90.

enjoy-digital, 2020-05-04:
Useful commands to compile OpenSBI:
````
export CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf-
export PLATFORM_RISCV_XLEN=32
make PLATFORM=spinal/vexriscv/sim/smp clean
make PLATFORM=spinal/vexriscv/sim/smp -j24
````

Dolu1990, 2020-05-04:
Just tested, it is fine to compile openSBI with VEX_HART_COUNT set to more hart than what you realy have. Only the linux dts realy need to be updated.

enjoy-digital, 2020-05-04:
Thanks @Dolu1990, so now the question is: how many harts are you planning to support :)

Dolu1990, 2020-05-04:
@enjoy-digital 4 for sure will be used in practice, so far, 8 seem fesable, as the data bandwidth do not seem realy high, 16 for fun and science would be the very max i think.

enjoy-digital, 2020-12-15:
This has been done and will be part of https://github.com/litex-hub/linux-on-litex-vexriscv/pull/165. OpenSBI fork is located at https://github.com/litex-hub/opensbi  on branch `0.8-linux-on-litex-vexriscv`.

enjoy-digital, 2020-12-16:
https://github.com/litex-hub/linux-on-litex-vexriscv/pull/165 has been merged, we can now close this.

closed 2020-12-16 by enjoy-digital