litex-hub/linux-on-litex-vexriscv #132: SMP should not be enabled for single core systems

kgugala, 2020-05-13:
This commit https://github.com/litex-hub/linux-on-litex-vexriscv/commit/3fc6ef1b9580acc4d4338e09ca7897d7d49ade6c enabled SMP in the kernel. 

Unfortunately it causes single core systems to hang in user space (in init). I think we should have a separate config for SMP systems.

Dolu1990, 2020-05-13:
That's weird that it hang, maybe it does some unsupported ecall to the emulator.bin ? Or the VexRiscv.v used is a old version which miss some fixes ?

I already got single core to boot with SMP enabled, but that was with opensbi.

Else i don't realy know how much overhead enabling SMP does.
I should do some mesurements about it.

enjoy-digital, 2020-05-13:
Thanks for reporting @kgugala. We are going to switch to OpenSBI (https://github.com/litex-hub/linux-on-litex-vexriscv/issues/129) so for now i'll revert the SMP change and will do more tests with OpenSBI.

rob-ng15, 2020-05-13:
Can confirm on de10nano with emulator.bin.

Had been trying to build a more 'fun' linux set with buildroot, but wasn't
booting. After changing to smp=n got a running system again.

Thanks for finding this.

enjoy-digital, 2020-05-13:
This has been disabled with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/b5afdeb534236f51d6c2e2f9a89914cb04aa14c6.

closed 2020-05-13 by enjoy-digital