litex-hub/linux-on-litex-vexriscv #135: ./sim.py: UART_EV_RX undeclared

dayjaby, 2020-05-19:
On a fresh installation on Ubuntu 18.04 after following the install commands and running ./sim.py I get:

```
main.c: In function 'litex_getchar':
main.c:52:31: error: 'UART_EV_RX' undeclared (first use in this function)
         uart_ev_pending_write(UART_EV_RX);
                               ^~~~~~~~~~
main.c:52:31: note: each undeclared identifier is reported only once for each function it appears in
Makefile:30: recipe for target 'main.o' failed
make: *** [main.o] Error 1
Traceback (most recent call last):
  File "./sim.py", line 231, in <module>
    main()
  File "./sim.py", line 206, in main
    with_ethernet         = args.with_ethernet)
  File "./sim.py", line 101, in __init__
    }, "little")
  File "/home/david/Development/litex/litex/litex/soc/integration/common.py", line 36, in get_mem_data
    data_size = max(int(base, 16) + os.path.getsize(filename), data_size)
  File "/usr/lib/python3.6/genericpath.py", line 50, in getsize
    return os.stat(filename).st_size
FileNotFoundError: [Errno 2] No such file or directory: 'buildroot/Image'
```

enjoy-digital, 2020-05-19:
@dayjaby: thanks for the feedback, are you using upstream LiteX/Linux-on-LiteX-Vexriscv? (i did some changes on that today).

dayjaby, 2020-05-19:
After performing `./litex_setup.py update`, the error is gone, but I got a new one:

```
make: *** No rule to make target '/home/david/Development/litex/litex/litex/soc/software/bios/commands/cmd_bios.c', needed by 'cmd_bios.o'.  Stop.
make: Leaving directory '/home/david/Development/linux-on-litex-vexriscv/build/sim/software/bios'
Traceback (most recent call last):
  File "./sim.py", line 231, in <module>
    main()
  File "./sim.py", line 222, in main
    trace_end   = int(args.trace_end))
  File "/home/david/Development/litex/litex/litex/soc/integration/builder.py", line 208, in build
    self._generate_rom_software(not self.soc.integrated_rom_initialized)
  File "/home/david/Development/litex/litex/litex/soc/integration/builder.py", line 188, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/david/Development/linux-on-litex-vexriscv/build/sim/software/bios', '-f', '/home/david/Development/litex/litex/litex/soc/software/bios/Makefile']' returned non-zero exit status 2
```

dayjaby, 2020-05-20:
After copying the files Image and rootfs.cpio from linux-on-litex-vexriscv-prebuilt/buildroot, it runs now.

closed 2020-05-20 by dayjaby

enjoy-digital, 2020-05-20:
Thanks for feedback, great you got it working.