litex-hub/linux-on-litex-vexriscv #138: change wishbone bus to axi bus

strokkk, 2020-06-21:
I want to use axi bus, not wishbone bus. I want to use the system you provided, can you help me with a little?

We have confirmed that interface2 and interface5 are associated with the sdram port. I used wishbone to axi ip ​​to use vivado's MIG. However, this method does not pass memtest at boot time.

enjoy-digital, 2020-06-21:
The SoC is currently built using a Wishbone bus as the main bus. It's not yet possible to use a AXI bus as the main bus (but we are working on this), but it's already possible to create a bridge between the Wishbone bus and an AXI Bus.

It's not clear to me what interface2 and interface5 are and why you are using the MIG instead of LiteDRAM already integrated in the project. Can you give a more general description of your use case?

strokkk, 2020-06-22:
I am glad that you are working it.

My vivado project was created using the KCU105 board.

In order to analyze the configuration of the project,

 I made a new module by gathering only by variables related to each other in over 20,000 lines of top.v.

The soclinux_soclinux_cpu_dbus(ibus) from vexriscv arrives at interface_bank_bus via comb_rhs_array_muxed36, shared_adr, soclinux_soclinux_wishbone.
These are interface0 through interface8.
Each interface plays a different role. (soccontroller, led, storage, etc.)

[top.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4810951/top.txt)

strokkk, 2020-06-22:
and.. the project I'm working on in the lab includes MIG and uses the AXI bus. I want to integrate my system with your system.

enjoy-digital, 2020-06-22:
The verilog code generated by LiteX is not intended to me manually modified. If you want reuse the Linux SoC and replace the LiteDRAM controller with the MIG from Xilinx, the best is to modify the `add_sdram` method (https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1114-L1147) and instead of connecting the Wishbone bus to LiteDRAM, convert it to AXI with `Wishbone2AXILite`  (https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/axi.py#L415) and connect this AXI bus to the MIG directly in LiteX or export it as in interface and connect it to the MIG in your traditional flow.

Once we'll be supporting AXI, we could imagine creating a generator for the Linux SoC with AXI interfaces, but  for now there are no generator and the project is still tighly coupled to LiteX, so for such use case, the source code need to be modified.

strokkk, 2020-06-23:
Thanks for the help. I haven't solved it yet, but I think I have the right direction.
Thank you! enjoy-digital!

enjoy-digital, 2020-06-23:
Thanks @strokkk, i'm closing since i think the main question is answered but feel free to create others(s) issue with more specific questions if any.

closed 2020-06-23 by enjoy-digital

Beauxrel, 2023-07-17:
Does this generator exist yet @enjoy-digital? I need to do the same.

enjoy-digital, 2023-07-17:
Hi @Beauxrel,

this hasn't been created yet no. A generator has however been created for LiteX SoC: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_soc_gen.py with demo here: https://github.com/enjoy-digital/litex_soc_gen_test. This allow generating a standalone core for LiteX SoCs, but for VexRiscv-SMP, there is still the memory interfaces to look at.

Beauxrel, 2023-07-17:
For future reference. I believe this is what you were explaining. I'm going to be testing this today, but I was able to generate the SoC without litedram doing this.

                     #-------------------AXI INTERFACE DEVELOPMENT-------------
            # Wishbone Slave <--> LiteDRAM bridge.
            from litex.soc.interconnect.axi import Wishbone2AXI
            from litex.soc.interconnect.axi import AXIInterface

            axi         = AXIInterface(data_width=512, address_width=36)

            ddraxiinterface = Wishbone2AXI(
                wishbone        = litedram_wb,
                axi             = axi,
                base_address    = self.bus.regions["main_ram"].origin
            )
            self.submodules += ddraxiinterface
            #-------------------AXI INTERFACE DEVELOPMENT-------------

Beauxrel, 2023-07-18:
@enjoy-digital can you assist me, I'm having trouble exporting the signals for use. what will the signals be named? I don't think what I put there is actually correct.

Beauxrel, 2023-07-18:
I think I need to downconvert the l2 master side (512/36) to (32/32) slave and then I need to export, but I dont know the command for that.