litex-hub/linux-on-litex-vexriscv #139: Kernel panic when enabling spisdcard

enjoy-digital, 2020-06-30:
With the current kernel/dts, enabling spisdcard causes a kernel panic. This is probably a configuration issue since this is working on other Linux/LiteX projects.

oskirby, 2020-07-11:
I think I have been able to trace the cause of the panic down to a bug in the spi-litespi.c driver. In particular the `litespi_rxtx()` function is causing memory corruption by mutating the tx_buf and rx_buf pointers. I'm guessing that something in the mmc-spi driver is expecting those pointers to remain constant.

After rebuilding with a fix, I no longer get a kernel panic, but there is still no life out of the SD card.

enjoy-digital, 2020-07-11:
Thanks for looking at this, on my side i haven't still investigated. This branch has been tested by @gsomlo with Rocket CPU: https://github.com/litex-hub/linux/tree/litex-rocket-rebase Could you have a look to see if there is a difference on the spi-litespi.c driver and try to use it if it's the case?

gsomlo, 2020-07-11:
> /linux/tree/litex-rocket-rebase Could you have a look to see if there is a

I haven't tested it, but the rocket kernel *should* also build and
work ok on vexriscv (i.e., I tried *extending* select parts to 64-bit
support, rather than hacking it with *only* 64-bit operation in mind :)

oskirby, 2020-07-11:
I just gave it a try, and the https://github.com/litex-hub/linux/tree/litex-rocket-rebase branch seems to boot just fine on a vexriscv CPU (and it also fixes the kernel panic), so it seems like you have done a great job with your 64-bit branch.

However, neither the litex-rocket-rebase or nor my fixed https://github.com/oskirby/linux-litex-kernel/tree/litex-spisdcard-panic branch seem to have any luck detecting micro-SD cards, while the Litex BIOS does detect and boot from the micro-SD card.

gsomlo, 2020-07-11:
On Sat, Jul 11, 2020 at 03:32:28PM -0700, oskirby wrote:
> However, neither the litex-rocket-rebase or nor my fixed https://github.com/
> oskirby/linux-litex-kernel/tree/litex-spisdcard-panic branch seem to have any
> luck detecting micro-SD cards, while the Litex BIOS does detect and boot from
> the micro-SD card.

You'll need an entry in your DT that looks something like this (modify
"reg" according to whatever the MMIO address is on your gateware):

https://github.com/gsomlo/riscv-pk/blob/gls-litex-devel/machine/litex_rocket.dts#L110

enjoy-digital, 2020-12-16:
We are now using https://github.com/litex-hub/linux/tree/litex-rocket-rebase branch as created and maintained by @gsomlo. This has just been tested on the OrangeCrab by using `spisdcard` instead of `sdcard`:

```
[    6.138604] mmc0: new SDHC card on SPI
[    6.172418] mmcblk0: mmc0:0000 SD32G 29.9 GiB 
```

closed 2020-12-16 by enjoy-digital