litex-hub/linux-on-litex-vexriscv #140: DRAM initialization fails on Genesys2

shingarov, 2020-07-01:
Building from commit 7d2bea successfully produces a `top.bit`, but booting it produces:
```lxterm /dev/ttyUSB0 --speed=1e6 --no-crc       
[LXTERM] Starting....
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!
 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs
 BIOS built on Jun 28 2020 09:58:32
 BIOS CRC passed (1c2757d3)
 Migen git sha1: --------
 LiteX git sha1: 68d3804c
--=============== SoC ==================--
CPU:       VexRiscv @ 125MHz
BUS:       WISHBONE 32-bit @ 4GiB
CSR:       8-bit data
ROM:       64KiB
SRAM:      4KiB
L2:        8KiB
MAIN-RAM:  1048576KiB
--========== Initialization ============--
Ethernet init...
Initializing SDRAM...
SDRAM now under software control
SDRAM now under software control
Write leveling:
Command/Clk scan:
|11111111000000000001111111111111| best: 21
Data scan:
m0: |00000111111111111000000000| delay: 05
m1: |00000011111111111110000000| delay: 06
m2: |00011111111111100000000000| delay: 03
m3: |00011111111111110000000000| delay: 03
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |00000000000000000000000000000000| delays: -
m0, b09: |00000000000000000000000000000000| delays: -
m0, b10: |00000000000000000000000000000000| delays: -
m0, b11: |11111111100000000000000000000000| delays: 04+-04
m0, b12: |00000000000111111111110000000000| delays: 16+-06
m0, b13: |00000000000000000000000011111111| delays: 28+-04
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b12 delays: 16+-06
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |00000000000000000000000000000000| delays: -
m1, b09: |00000000000000000000000000000000| delays: -
m1, b10: |00000000000000000000000000000000| delays: -
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b00 delays: -
m2, b00: |00000000000000000000000000000000| delays: -
m2, b01: |00000000000000000000000000000000| delays: -
m2, b02: |00000000000000000000000000000000| delays: -
m2, b03: |00000000000000000000000000000000| delays: -
m2, b04: |00000000000000000000000000000000| delays: -
m2, b05: |00000000000000000000000000000000| delays: -
m2, b06: |00000000000000000000000000000000| delays: -
m2, b07: |00000000000000000000000000000000| delays: -
m2, b08: |00000000000000000000000000000000| delays: -
m2, b09: |00000000000000000000000000000000| delays: -
m2, b10: |00000000000000000000000000000000| delays: -
m2, b11: |11111111111000000000000000000000| delays: 05+-05
m2, b12: |00000000000001111111111100000000| delays: 18+-05
m2, b13: |00000000000000000000000000011111| delays: 29+-03
m2, b14: |00000000000000000000000000000000| delays: -
m2, b15: |00000000000000000000000000000000| delays: -
best: m2, b11 delays: 05+-05
m3, b00: |00000000000000000000000000000000| delays: -
m3, b01: |00000000000000000000000000000000| delays: -
m3, b02: |00000000000000000000000000000000| delays: -
m3, b03: |00000000000000000000000000000000| delays: -
m3, b04: |00000000000000000000000000000000| delays: -
m3, b05: |00000000000000000000000000000000| delays: -
m3, b06: |00000000000000000000000000000000| delays: -
m3, b07: |00000000000000000000000000000000| delays: -
m3, b08: |00000000000000000000000000000000| delays: -
m3, b09: |00000000000000000000000000000000| delays: -
m3, b10: |00000000000000000000000000000000| delays: -
m3, b11: |11111111111100000000000000000000| delays: 06+-06
m3, b12: |00000000000001111111111110000000| delays: 19+-06
m3, b13: |00000000000000000000000000011111| delays: 29+-02
m3, b14: |00000000000000000000000000000000| delays: -
m3, b15: |00000000000000000000000000000000| delays: -
best: m3, b11 delays: 06+-06
SDRAM now under hardware control
Memtest bus failed: 256/256 errors
Memtest data failed: 522277/524288 errors
Memtest addr failed: 8192/8192 errors
Memory initialization failed
--============= Console ================--
litex>
```

Just to exclude physical problems I verified with the prebuilt `top.bit` and it boots fine.

enjoy-digital, 2020-07-01:
Thanks, could you also provide the generated verilog + bitstream do also test it on hardware?

shingarov, 2020-07-01:
[top.bit.gz](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4859111/top.bit.gz)
[top.v.gz](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4859113/top.v.gz)

shingarov, 2020-07-15:
Just FYI (although this isn't about "litex-vexriscv") I've been having some further fun today with Microwatt with litedram, after pulling the last LiteX, getting this dram init result:
```
Welcome to Microwatt !

 Soc signature: f00daa5500010001
  Soc features: UART DRAM SPIFLASH 
          DRAM: 1024 MB
     DRAM INIT: 16 KB
           CLK: 100 MHz
  SPI FLASH ID: 010219 Cypress/Spansion (CF1=02) [quad IO mode]
 SPI FLASH OFF: 0xa00000 bytes

LiteDRAM built from Migen 4fea1bd and LiteX 83d24d08
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Write leveling:
Command/Clk scan:
|11111110000000000000000011111111| best: 26
Data scan:
m0: |000001111111111111110000| delay: 05
m1: |000000111111111111111000| delay: 06
m2: |000111111111111111000000| delay: 03
m3: |000111111111111111000000| delay: 03
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |00000000000000000000000000000000| delays: -
m0, b09: |00000000000000000000000000000000| delays: -
m0, b10: |00000000000000000000000000000000| delays: -
m0, b11: |00000000000000000000000000000000| delays: -
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b00 delays: -
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |00000000000000000000000000000000| delays: -
m1, b09: |00000000000000000000000000000000| delays: -
m1, b10: |00000000000000000000000000000000| delays: -
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b00 delays: -
m2, b00: |00000000000000000000000000000000| delays: -
m2, b01: |00000000000000000000000000000000| delays: -
m2, b02: |00000000000000000000000000000000| delays: -
m2, b03: |00000000000000000000000000000000| delays: -
m2, b04: |00000000000000000000000000000000| delays: -
m2, b05: |00000000000000000000000000000000| delays: -
m2, b06: |00000000000000000000000000000000| delays: -
m2, b07: |00000000000000000000000000000000| delays: -
m2, b08: |00000000000000000000000000000000| delays: -
m2, b09: |00000000000000000000000000000000| delays: -
m2, b10: |00000000000000000000000000000000| delays: -
m2, b11: |00000000000000000000000000000000| delays: -
m2, b12: |00000000000000000000000000000000| delays: -
m2, b13: |00000000000000000000000000000000| delays: -
m2, b14: |00000000000000000000000000000000| delays: -
m2, b15: |00000000000000000000000000000000| delays: -
best: m2, b00 delays: -
m3, b00: |00000000000000000000000000000000| delays: -
m3, b01: |00000000000000000000000000000000| delays: -
m3, b02: |00000000000000000000000000000000| delays: -
m3, b03: |00000000000000000000000000000000| delays: -
m3, b04: |00000000000000000000000000000000| delays: -
m3, b05: |00000000000000000000000000000000| delays: -
m3, b06: |00000000000000000000000000000000| delays: -
m3, b07: |00000000000000000000000000000000| delays: -
m3, b08: |00000000000000000000000000000000| delays: -
m3, b09: |00000000000000000000000000000000| delays: -
m3, b10: |00000000000000000000000000000000| delays: -
m3, b11: |00000000000000000000000000000000| delays: -
m3, b12: |00000000000000000000000000000000| delays: -
m3, b13: |00000000000000000000000000000000| delays: -
m3, b14: |00000000000000000000000000000000| delays: -
m3, b15: |00000000000000000000000000000000| delays: -
best: m3, b00 delays: -
SDRAM now under hardware control
Memtest at 0x0x40000000...

- bus errors:  105/256
- addr errors: 8066/8192
- data errors: 516224/524288
Memtest KO
```
What would be the best approach to isolate litedram problems?

enjoy-digital, 2020-07-16:
@shingarov: sorry, i haven't yet been able to look at it, but will do, probably tomorrow.

enjoy-digital, 2020-08-03:
@shingarov: i just tested it on the Genesys2 with upstream repositories. It's possible https://github.com/litex-hub/litex-boards/commit/2ce24df76dda20cff9ac40c334300d5dc1311d60 helped on this. Here is the attached bitstream: genesys2_2020_08_03.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5014681/genesys2_2020_08_03.zip) 

Could you check you are able to reproduce these results on your side? For the Microwatt integration could you also make sure internal_vref is configured on the DDR3 bank?

shingarov, 2020-08-04:
>Could you check you are able to reproduce these results on your side?
Yes, the problem was indeed the `INTERNAL_VREF 0.750`.  I just pulled today's LX and now it works!

enjoy-digital, 2020-08-04:
Nice! thanks for the feedback and sorry for the delay.

shingarov, 2020-08-04:
>For the Microwatt integration could you also make sure internal_vref is configured on the DDR3 bank?

Strangely, for Microwatt it doesn't make any difference.  Not sure what else may be wrong.

enjoy-digital, 2020-08-04:
Would you mind sharing your project for microwatt? LiteDRAM confg file + tarball of the project to be able rebuild it? (you could send it to me by mail if you don't want to share it here).

shingarov, 2020-08-04:
I just pushed https://github.com/shingarov/microwatt/tree/genesys2c

I am re-running the vivado implementation from that and will tar the fusesoc workdir for you as soon as that finishes (in case you want to look at vivado's log).

shingarov, 2020-08-04:
tarball here:  www.shingarov.com/mw.tar.gz

enjoy-digital, 2020-08-06:
Thanks for the files, looking at the .yml you were using, i realized the .yml in LiteDRAM for the Genesys2 was not using the cmd_latency value we are using in LiteX-Boards. Would you mind doing a test with `cmd_latency` set to `1` (similar to https://github.com/enjoy-digital/litedram/commit/4e62d28af6bc98c4c25fb3f7840e166f0eb89486)? I just tested a LiteDRAM controller on the Genesys2 with this configuration.

shingarov, 2020-08-06:
Yes!  Microwatt works now!  Thanks!

closed 2020-08-06 by shingarov

enjoy-digital, 2020-08-06:
Great, thanks for the feedback.