litex-hub/linux-on-litex-vexriscv #143: build fail for ULX3S

wuxx, 2020-07-25:
when execute
$./make.py --board=ulx3s --build
it report a yosys-abc failure,  but if I execute the command separate, it is succ, really strange, can somebody help figure out the problem. detail log in attached file
-----------------
ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ ./make.py --board=ulx3s --build
........
4.47.18.5. Executing ABC9.
[ulx3s.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/4975005/ulx3s.txt)

Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-ONKnaB/abc.script 2>&1" failed: return code -1.
Traceback (most recent call last):
  File "./make.py", line 486, in <module>
    main()
  File "./make.py", line 468, in main
    builder.build(build_name="top", run=args.build)
  File "/home/ubuntu/oss/ulx3s/litex/litex/soc/integration/builder.py", line 205, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/ubuntu/oss/ulx3s/litex/litex/soc/integration/soc.py", line 1017, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/platform.py", line 34, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/trellis.py", line 229, in build
    _run_script(script)
  File "/home/ubuntu/oss/ulx3s/litex/litex/build/lattice/trellis.py", line 160, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed

-----------------
ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ /usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-ONKnaB/abc.script 2>&1
ABC command line: "source /tmp/yosys-abc-ONKnaB/abc.script".

+ read_lut /tmp/yosys-abc-ONKnaB/input.lut
+ read_box /tmp/yosys-abc-ONKnaB/input.box
+ &read /tmp/yosys-abc-ONKnaB/input.xaig
+ &ps
/tmp/yosys-abc-ONKnaB/input : i/o =   4562/   5511  and =   20511  lev =   43 (7.14)  mem = 0.48 MB  box = 518  bb = 120
+ &scorr
Warning: The network is combinational.
+ &sweep
+ &dc2
+ &dch -f
+ &ps
/tmp/yosys-abc-ONKnaB/input : i/o =   4562/   5511  and =   28653  lev =   35 (4.15)  mem = 0.57 MB  ch = 3686  box = 512  bb = 114
+ &if -W 300 -v
K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
Node =   28653.  Ch =  3142.  Total mem =    6.60 MB. Peak cut mem =    0.50 MB.
P:  Del = 5041.00.  Ar =   26111.0.  Edge =    32699.  Cut =   318202.  T =     0.07 sec
P:  Del = 4920.00.  Ar =   25448.0.  Edge =    31858.  Cut =   316355.  T =     0.07 sec
P:  Del = 4920.00.  Ar =   10437.0.  Edge =    24062.  Cut =   679900.  T =     0.13 sec
F:  Del = 4917.00.  Ar =    7484.0.  Edge =    21303.  Cut =   582797.  T =     0.13 sec
A:  Del = 4917.00.  Ar =    6933.0.  Edge =    19344.  Cut =   570621.  T =     0.19 sec
A:  Del = 4917.00.  Ar =    6853.0.  Edge =    19107.  Cut =   593313.  T =     0.20 sec
Total time =     0.79 sec
+ &write -n /tmp/yosys-abc-ONKnaB/output.aig
+ &mfs
+ &ps -l
/tmp/yosys-abc-ONKnaB/input : i/o =   4562/   5511  and =   18843  lev =   27 (4.48)  mem = 0.46 MB  box = 509  bb = 114
Mapping (K=7)  :  lut =   5169  edge =   18760  lev =   10 (2.34)  levB =   33  mem = 0.24 MB
LUT = 5169 : 2=836 16.2 %  3=1401 27.1 %  4=2017 39.0 %  5=755 14.6 %  6=78 1.5 %  7=82 1.6 %  Ave = 3.63
+ &write -n /tmp/yosys-abc-ONKnaB/output.aig
+ time
elapse: 5.22 seconds, total: 5.22 seconds
ubuntu@ubuntu:~/oss/ulx3s/linux-on-litex-vexriscv$ echo $?
0

daveshah1, 2020-07-25:
This is doubtless a Yosys issue not a LiteX one.

wuxx, 2020-07-25:
ok, I will open it on Yosys, thanks.

closed 2020-07-25 by wuxx