litex-hub/linux-on-litex-vexriscv #144: Unable to Build Successfully for DE0nano

teezzan, 2020-07-27:
Hello,
I am trying to build the image for my Altera Cyclone IV dev board. I thought it would be okay if I based my work on the existing Altera/Intel board  supported. In my bid to successfuly build an image for de0nano, I have been encountering an error.
The error occurs during the compilation by Quartus Toolchain. The error part of the build log is as follows 
```bash
Warning (332056): PLL cross checking found inconsistent PLL clock settings:
    Warning (332056): Node: ALTPLL|auto_generated|pll1|clk[0] was found missing 1 generated clock that corresponds to a base clock with a period of: 20.000
    Warning (332056): Node: ALTPLL|auto_generated|pll1|clk[1] was found missing 1 generated clock that corresponds to a base clock with a period of: 20.000
Critical Warning (332168): The following clock transfers have no clock uncertainty assignment. For more accurate results, apply clock uncertainty assignments or use the derive_clock_uncertainty command.
    Critical Warning (332169): From clk50 (Rise) to clk50 (Rise) (setup and hold)
Info (332129): Detected timing requirements -- optimizing circuit to achieve only the specified requirements
Info (332111): Found 1 clocks
    Info (332111):   Period   Clock Name
    Info (332111): ======== ============
    Info (332111):   20.000        clk50
Info (176233): Starting register packing
Info (176235): Finished register packing
    Extra Info (176218): Packed 32 registers into blocks of type Block RAM
    Extra Info (176218): Packed 132 registers into blocks of type Embedded multiplier output
Info (171121): Fitter preparation operations ending: elapsed time is 00:00:27
Info (14896): Fitter has disabled Advanced Physical Optimization because it is not supported for the current family.
Info (170189): Fitter placement preparation operations beginning
Error (170040): Can't place all RAM cells in design
    Info (170034): Selected device has 66 memory locations of type M9K. The current design requires 85 memory locations of type M9K to successfully fit.
    Info (170033): Memory usage required for the design in the current device: 129% M9K memory block locations required
Info (170190): Fitter placement preparation operations ending: elapsed time is 00:00:01
Info (11888): Total time spent on timing analysis during the Fitter is 0.63 seconds.
Error (171000): Can't fit design in device
Warning (169177): 18 pins must meet Intel FPGA requirements for 3.3-, 3.0-, and 2.5-V interfaces. For more information, refer to AN 447: Interfacing Cyclone IV E Devices with 3.3/3.0/2.5-V LVTTL/LVCMOS I/O Systems.
    Info (169178): Pin sdram_dq[0] uses I/O standard 3.3-V LVTTL at G2 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[1] uses I/O standard 3.3-V LVTTL at G1 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[2] uses I/O standard 3.3-V LVTTL at L8 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[3] uses I/O standard 3.3-V LVTTL at K5 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[4] uses I/O standard 3.3-V LVTTL at K2 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[5] uses I/O standard 3.3-V LVTTL at J2 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[6] uses I/O standard 3.3-V LVTTL at J1 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[7] uses I/O standard 3.3-V LVTTL at R7 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[8] uses I/O standard 3.3-V LVTTL at T4 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[9] uses I/O standard 3.3-V LVTTL at T2 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[10] uses I/O standard 3.3-V LVTTL at T3 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[11] uses I/O standard 3.3-V LVTTL at R3 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[12] uses I/O standard 3.3-V LVTTL at R5 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[13] uses I/O standard 3.3-V LVTTL at P3 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[14] uses I/O standard 3.3-V LVTTL at N3 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin sdram_dq[15] uses I/O standard 3.3-V LVTTL at K1 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 16
    Info (169178): Pin clk50 uses I/O standard 3.3-V LVTTL at R8 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 7
    Info (169178): Pin serial_rx uses I/O standard 3.3-V LVTTL at B4 File: /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.v Line: 6
Info (144001): Generated suppressed messages file /home/gal3li0/Documents/Litex_linux/test/linux-on-litex-vexriscv/build/de0nano/gateware/top.fit.smsg
Error: Quartus Prime Fitter was unsuccessful. 2 errors, 9 warnings
    Error: Peak virtual memory: 1366 megabytes
    Error: Processing ended: Mon Jul 27 02:49:15 2020
    Error: Elapsed time: 00:01:06
    Error: Total CPU time (on all processors): 00:01:02
Info: *******************************************************************
Info: Running Quartus Prime Assembler
    Info: Version 19.1.0 Build 670 09/22/2019 SJ Lite Edition
    Info: Copyright (C) 2019  Intel Corporation. All rights reserved.
    Info: Your use of Intel Corporation's design tools, logic functions 
    Info: and other software and tools, and any partner logic 
    Info: functions, and any output files from any of the foregoing 
    Info: (including device programming or simulation files), and any 
    Info: associated documentation or information are expressly subject 
    Info: to the terms and conditions of the Intel Program License 
    Info: Subscription Agreement, the Intel Quartus Prime License Agreement,
    Info: the Intel FPGA IP License Agreement, or other applicable license
    Info: agreement, including, without limitation, that your use is for
    Info: the sole purpose of programming logic devices manufactured by
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Mon Jul 27 02:49:18 2020
[... 21 lines cut ...]
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Mon Jul 27 02:49:21 2020
Info: Command: quartus_sta top -c top
Info: qsta_default_script.tcl version: #1
Error: Can't run the Timing Analyzer (quartus_sta) -- Fitter (quartus_fit) failed or was not run. Run the Fitter (quartus_fit) successfully before running the Timing Analyzer (create_timing_netlist).
Error: Quartus Prime Timing Analyzer was unsuccessful. 1 error, 0 warnings
    Error: Peak virtual memory: 695 megabytes
    Error: Processing ended: Mon Jul 27 02:49:22 2020
    Error: Elapsed time: 00:00:01
    Error: Total CPU time (on all processors): 00:00:01
buildroot/rv32.dtb: Warning (simple_bus_reg): Node /soc/interrupt-controller missing or empty reg/ranges property
 CC       main.o
 LD       emulator.elf
chmod -x emulator.elf
 OBJCOPY  emulator.bin
chmod -x emulator.bin
```

What do you propose that I do? Again, my goal is to build an image successfully for my unnamed  Cyclone IV dev board.
Thank you very much.

Best Regards,
Taiwo

teezzan, 2020-07-27:
Hi, 
Looking at the error and looking through the code, I noticed that it was due to the available Block Ram. The code block for the de0nano looked like this.
```python
class De0Nano(Board):
    soc_kwargs = {"l2_size": 2048} # Reduce l2_size (Not enough blockrams).
    def __init__(self):
        from litex_boards.targets import de0nano
        Board.__init__(self, de0nano.BaseSoC, soc_capabilities={
            # Communication
            "serial",
        }, bitstream_ext=".sof")
```

I reduced that to 132 and it didn't work. The blockRam reduced to 120%  compared to the one shown in the first error log.
Setting it to 0 did not help. It stayed at 120%.  What do you propose that I do?

Best Regards,
Taiwo

enjoy-digital, 2020-08-03:
Sorry, there was a configuration error on the De0Nano, that has been with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/e0cca5b8b892e523aa8c5e93374439524d252fbb. Please find attached the tested bitstream.
[de0nano_2020_08_03.zip](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5014665/de0nano_2020_08_03.zip)

closed 2020-08-03 by enjoy-digital